earlephilhower/arduino-pico #775: [Feature Request] Port these libraries lwIP_w5100, lwIP_w5500 and lwIP_enc28j60 from ESP826

khoih-prog, 2022-08-20:
Hi @earlephilhower 

As you've done so great work for lwIP_CYW43, lwIP_Ethenet and cyw43 WiFi libraries, could you please, when having time, port these  libraries : 

1. [**lwIP_w5100**](https://github.com/esp8266/Arduino/tree/master/libraries/lwIP_w5100)
2. [**lwIP_w5500**](https://github.com/esp8266/Arduino/tree/master/libraries/lwIP_w5500)
3. [**lwIP_enc28j60**](https://github.com/esp8266/Arduino/tree/master/libraries/lwIP_enc28j60)

from ESP826 to this powerful and popular core.

Those LwIP-based libraries would be used in newly-created Async-related libraries, similar to those for RP2040W, as in [**Libraries created / updated to support RASPBERRY_PI_PICO_W using CYW43439 WiF**i #757](https://github.com/earlephilhower/arduino-pico/discussions/757)

earlephilhower, 2022-08-20:
I have an ENC28 and a W5500S(thanks, WizNet!), but not the other two.  Do you have a W5100 or W5500 to test on?  It's in the queue, but only so many hours in the day.

When they get ported everything `WiFi` can do, they will be able to do (i.e. SSL connections, UDP, servers, NTP, etc.).  You'll still need to use a `WiFiClient` or `WiFiServer` object, but they are really `LWIPClient`and `LWIPServer` under the hood and have no `WiFi` constraints (or at least, they shouldn't!)

khoih-prog, 2022-08-20:
Good news the tasks are in the queue.

> It's in the queue, but only so many hours in the day.

Totally understand. I get the same constraint, even if I'd like to help port them.

I have W5100 and W5500 to test.

---

BTW, is that possible to add **DNSServer** library or code into the core or WiFi lib ?

I'm trying to port **DNSServer** library, but having some issue with UDP packet receiving.

Everything will be done much faster by you, as previous maintainer of ESP8266 core.

earlephilhower, 2022-08-20:
@khoih-prog Try #779.  It just built fine with trivial fixes (WDT reset ignore, remove obsolete BIT() macro).

khoih-prog, 2022-08-20:
Thanks for the DNSServer.

Still have the same issue in `WiFiUDP::parsePacket()` : _ctx->next is always NULL.
Some missing config in LwIP stack relating to DNSServer ??

https://github.com/earlephilhower/arduino-pico/blob/7be472932b7ac1c179b79a10ae882c3b88f00e42/libraries/WiFi/src/WiFiUdp.cpp#L182-L184

---

#### Test code

```
#define USE_WIFI_NINA         false
#define USE_WIFI101           false
#define USE_WIFI_CUSTOM       false

#include <WiFi.h>
#include <DNSServer.h>
#include <WiFiWebServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 42, 1);

DNSServer dnsServer;
WiFiWebServer webServer(80);

String responseHTML = ""
                      "<!DOCTYPE html><html lang='en'><head>"
                      "<meta name='viewport' content='width=device-width'>"
                      "<title>CaptivePortal</title></head><body>"
                      "<h1>Hello World!</h1><p>This is a captive portal example."
                      " All requests will be redirected here.</p></body></html>";

void setup()
{
  // Debug console
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  delay(1000);

  Serial.print("\nStart CaptivePortal on "); Serial.println(BOARD_NAME);
  
  WiFi.mode(WIFI_AP);

  WiFi.beginAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  bool dnsServer_OK = false;
  
  dnsServer_OK = dnsServer.start(DNS_PORT, "*", apIP);

  if (dnsServer_OK)
    Serial.println("dnsServer starting OK");
  else
    Serial.println("dnsServer starting failure");  

  // replay to all requests with same HTML
  webServer.onNotFound([]()
  {
    webServer.send(200, "text/html", responseHTML);
  });

  webServer.begin();
}

void loop()
{
  dnsServer.processNextRequest();
  webServer.handleClient();
}
```

khoih-prog, 2022-08-20:
Strange, `WiFiUDP::parsePacket()` is OK when I'm running [**UdpNTPClient**](https://github.com/khoih-prog/WiFiWebServer/tree/master/examples/UdpNTPClient)

No problem with `_ctx->next` is always NULL.

earlephilhower, 2022-08-20:
Yeah, I just went back and checked the WiFiUDP example to make sure things were good.

At this point I think it may be "routing" related since the UDP code doesn't know or care if in AP or STA mode.  This looks to take some digging.

An easy verification would to make the WiFiUDP example run over an AP not STA and verify failure.  Plus that would be a simpler test on a known good code block for any fix (i.e. I did not dig into the DNS server guts so maybe there are 8266-specific assumptions somewhere).

earlephilhower, 2022-08-20:
Thinking out loud, LWIP UDP while in AP mode can't be completely borked.  The DHCP server does give addresses out therefore it is able to receive and send UDP packets.  So something in this core is looking at a received UDP packet and saying, "nope...not for me."

earlephilhower, 2022-08-20:
@khoih-prog well, that was obvious.  DHCP server gave 8.8.8.8 hardcoded as DNS server, not the GW IP.  Check the latest DNSServer PR for the update.  I now see DNS requests being processed by the AP.

khoih-prog, 2022-08-20:
@earlephilhower 

Wow. Amazingly quick and the bug is fixed.

Now **Captive Portal for DNSServer** is working now when entering any non-existing site, such as `example.ca`.

[image]

Would you like me to make a PR for the examples for DNSServer

---

### Captive Portal Code used for the test

```
#define USE_WIFI_NINA         false
#define USE_WIFI101           false
#define USE_WIFI_CUSTOM       false

#include <WiFi.h>
#include <DNSServer.h>
#include <WiFiWebServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 42, 1);

DNSServer dnsServer;
WiFiWebServer webServer(80);

String responseHTML = ""
                      "<!DOCTYPE html><html lang='en'><head>"
                      "<meta name='viewport' content='width=device-width'>"
                      "<title>CaptivePortal</title></head><body>"
                      "<h1>Hello World!</h1><p>This is a captive portal example."
                      " All requests will be redirected here.</p></body></html>";

void setup()
{
  // Debug console
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  delay(1000);

  Serial.print("\nStart CaptivePortal on "); Serial.println(BOARD_NAME);
  
  WiFi.mode(WIFI_AP);

  WiFi.beginAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  bool dnsServer_OK = false;
  
  dnsServer_OK = dnsServer.start(DNS_PORT, "*", apIP);

  if (dnsServer_OK)
    Serial.println("dnsServer starting OK");
  else
    Serial.println("dnsServer starting failure");  

  // replay to all requests with same HTML
  webServer.onNotFound([]()
  {
    webServer.send(200, "text/html", responseHTML);
  });

  webServer.begin();
}

void loop()
{
  dnsServer.processNextRequest();
  webServer.handleClient();
}
```

khoih-prog, 2022-08-21:
@earlephilhower 

Is that possible to add **WebServer** library, similar to [**ESP8266WebServer**](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer) into the core ?

It's required for certain WebServer apps, with simpler way to use, such as

```
server.on("/", handleRoot);
```

I can help if you'd like.

earlephilhower, 2022-08-21:
It's in the queue.  First I'm looking into HTTPClient (and want to fix the odd requirements we have on the 8266/32 which catch folks by surprise) then the web server.  Lots of stuff in this here queue. :)

khoih-prog, 2022-08-21:
Wow. Glad to know.

closed 2023-09-15 by earlephilhower

thechordmaster, 2025-03-15:
Sorry to open this thread again.. I have a w55rp2040 ebb pico board . I am not able to get the ethernet port working with any of the examples available for the evb w5500… are the lpinouts different for the evb 2040?

commit e1274dca (2023-09-09): Allow non-CYW43 network interfaces (W5500, etc.)

WIP - Builds and can get a DHCP address using W5500.

Lots of infrastructure work still required:
[ ] Separate CYW43 stuff and LWIP stuff (mutex, etc.)
[ ] Add async worker for pumping frames
[ ] Real LWIP-only mutex
[ ] Should WiFiClient be available as TCPClient?
etc.

Fixes #775
PR #1703 (2023-09-15): Support wired network interfaces (W5500, W5100, ENC28J60)
Allows wired networking support for all the WiFi-related classes (WebServer, WiFiClient, WiFiServer, etc.)

Supports WizNet W5500, WizNet 5100, and ENC28J60 modules.

Fixes #775
commit 1f3d5011 (2023-09-15): Support wired network interfaces (W5500, W5100, ENC28J60) (#1703)

Enable use of wired Ethernet modules as first-class LWIP citizens.  All
networking classes like MDNS, WebServer, HTTPClient, WiFiClient, and OTA
can use a wired Ethernet adapter just like built-in WiFi.

Two examples updated to show proper use.

Uses the Async Context support built into the Pico SDK.  When running on the
Pico  it will use the CYW43 async instance.

Uses modified wired Ethernet drivers, thanks Nicholas Humfrey!

Note, the classic, non-LWIP integrated `Ethernet` and related libraries
should still work fine (but n [... cut]