earlephilhower/arduino-pico #776: Servo sweep only giving 90 degrees of movement and not 180ish

johnmholmes, 2022-08-20:
Hi I have been trying the Arduino pic board manager and have found I am only getting about 90 Degrees of movement and not 180 ish. I tried it first on the Cryton maker pi and then bought a original Raspberry Pi Pico  board. Both boards give the same results.

The servo has been tested on an Arduino Nano and works with full sweep movement. I also tried the RP2040_ISR_servo and this gives me full sweep on both Pico boards.

I have also tested if with both versions of the Arduino Ide. There is no compile errors showing do you have any ideas what I may need to change to allow for closer to 180 movement.

earlephilhower, 2022-08-20:
The defaults in the Servo library here are a little conservative here to avoid damage in the case of over-driving.  The pulse widths individual servos, especially the no-name or clones, need are sometimes variable.

You can set the min and max servo pulse width in the `attach` command.  See https://www.arduino.cc/reference/en/libraries/servo/attach/

You'd do something like `attach(pin, 1000, 3000)` for example, to double the max pulse length.

closed 2022-08-20 by earlephilhower

johnmholmes, 2022-08-20:
Thank you Just given that a try and used what I had used before for Arduino Nano values   "attach(15, 600, 2400) " and its work spot on.

Thanks for all the good work you have done and for a fast response.