earlephilhower/arduino-pico #778: EEPROM.update is missing

arturo182, 2022-08-20:
I have some Arduino code that I want to make work on multiple cores, including this one, the code uses `EEPROM.update`, but this function is not available in this core.

From my understanding `EEPROM.put` and `EEPROM.update` differ only in the way that `put` always writes the value to EEPROM, and `update` only does it if the previous value doesn't match the new one. Looking at the code for `EEPROM.put` in this core, that's what `put` it already does anyway. So I would like to propose adding an `update` function that just calls `put`, and the issue will be resolved.

I can make a PR for it, just wanted to consult first if this is the way you want it handled, or maybe you had something else in mind.

Thanks!

earlephilhower, 2022-08-20:
@arturo182 the call was omitted on purpose.  There is no real EEPROM, so it is impossible to write updates to EEPROM on byte-wide basis.  Basically, the EEPROM library need to do your edits on a in-RAM copy and then erase and write a changed block to flash on `EEPROM.commit()`.  

The reason it was done this way is the flash probably has 1,000-10,000 write cycles before it fails, whereas EEPROM can probably go 100,000-1,000,000 cycles before an issue happens.  So it is important to only `commit()` when you need to.  See the examples for the proper EEPROM cycle here.

Another alternative is to drop the fake EEPROM and just use a `File`.  We have a real filesystem with directories and wear-leveling, and you can commit partial writes (will write to a new flash block and mark old one invalid).  That's a much better solution if you can put in a little work.

arturo182, 2022-08-20:
Ah, I see, I didn't look into the implementation too much. Of course, it makes sense to not clobber the flash all the time :)

But, I'm not sure how the implementation lead to the decision to skip `update`, from the Arduino docs it does seem `put` and `update` perform the same operation in the same way, except for `update` first checking the existing value, so adding an `update` as an alias for `put` shouldn't affect or break anything except making more existing code compatible with this core.

earlephilhower, 2022-08-20:
Fair enough.  Would you like to make a new method `EEPROM.update()` which just calls `EEPROM.put()`?  The user would still need to worry about the `commit()` addition, but there's not much that can be done about it at compile time.

arturo182, 2022-08-21:
Yes, the user still has to call `begin` and `commit` :+1:

closed 2022-08-21 by earlephilhower