earlephilhower/arduino-pico #792: Error: unaligned opcodes detected in executable segment with flag -Os

victpork, 2022-08-23:
Referring to previous issue raised in project lvgl:
https://github.com/lvgl/lvgl/issues/3603

When trying to compile to code with standard `-Os` in Arduino IDE, it raises an error
```
C:\Users\user\AppData\Local\Temp\ccamc2cB.s: Assembler messages:
C:\Users\user\AppData\Local\Temp\ccamc2cB.s: Error: unaligned opcodes detected in executable segment

exit status 1
```
and fails to compile. No errors after switching to other optimization flag. 
May I know if this is a compiler issue (as lvgl dev points out)?

I'm using version 2.4.0. Will try out 2.4.1 when it is available in the IDE board manager.

earlephilhower, 2022-08-23:
Can you attach a full build output log?    This could be one of two things as I see it:  Either an error in the linker file or a `gcc` bug.  

The linker file, I can fix.  A `gcc` internal error,  not so much and you'll need to get on the gcc mailing list.

earlephilhower, 2022-08-23:
Also I need a full set of files to build (lv_conf.h?  others?).  Installing `lvgl` from the IDE Library manager and the `lvgl-examples` I still get failures with some hardcoded paths:
````
In file included from /home/earle/Arduino/libraries/lvgl/src/../src/misc/lv_log.h:16,

                 from /home/earle/Arduino/libraries/lvgl/src/../lvgl.h:25,
                 from /home/earle/Arduino/libraries/lvgl/src/lvgl.h:17,
                 from /home/earle/Arduino/sketch_aug22b/sketch_aug22b.ino:1:
/home/earle/Arduino/libraries/lvgl/src/../src/misc/../lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
   41 |         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
````

victpork, 2022-08-23:
Please find the build output log attached:
[compile.log](https://github.com/earlephilhower/arduino-pico/files/9399621/compile.log)

>Also I need a full set of files to build (lv_conf.h? others?). Installing lvgl from the IDE Library manager and the lvgl-examples I still get failures with some hardcoded paths:

As for the lvgl library it needs some config along with the TFT-eSPI library installed and configured. I'll attached the configured one here for convenience.
[lvgl.zip](https://github.com/earlephilhower/arduino-pico/files/9399641/lvgl.zip)
[TFT_eSPI.zip](https://github.com/earlephilhower/arduino-pico/files/9399644/TFT_eSPI.zip)

As for the sample code, please use the one below instead. The demo need some extra config (moving some header files to the root directory... I didn't bother)
```c
#include <lvgl.h>
#include <TFT_eSPI.h>

/*Change to your screen resolution*/
static const uint16_t screenWidth  = 480;
static const uint16_t screenHeight = 320;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[ screenWidth * 10 ];

TFT_eSPI tft = TFT_eSPI(screenWidth, screenHeight); /* TFT instance */

/* Display flushing */
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
    uint32_t w = ( area->x2 - area->x1 + 1 );
    uint32_t h = ( area->y2 - area->y1 + 1 );

    tft.startWrite();
    tft.setAddrWindow( area->x1, area->y1, w, h );
    tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
    tft.endWrite();

    lv_disp_flush_ready( disp );
}

/*Read the touchpad*/
void my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
    uint16_t touchX, touchY;

    bool touched = tft.getTouch( &touchX, &touchY, 600 );

    if( !touched )
    {
        data->state = LV_INDEV_STATE_REL;
    }
    else
    {
        data->state = LV_INDEV_STATE_PR;

        /*Set the coordinates*/
        data->point.x = touchX;
        data->point.y = touchY;

        Serial.print( "Data x " );
        Serial.println( touchX );

        Serial.print( "Data y " );
        Serial.println( touchY );
    }
}

void setup()
{
    Serial.begin( 115200 ); /* prepare for possible serial debug */

    lv_init();

    tft.begin();          /* TFT init */
    tft.setRotation( 3 ); /* Landscape orientation, flipped */

    /*Set the touchscreen calibration data,
     the actual data for your display can be acquired using
     the Generic -> Touch_calibrate example from the TFT_eSPI library*/
    uint16_t calData[5] = { 275, 3620, 264, 3532, 1 };
    tft.setTouch( calData );

    lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * 10 );

    /*Initialize the display*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init( &disp_drv );
    /*Change the following line to your display resolution*/
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register( &disp_drv );

    /*Initialize the (dummy) input device driver*/
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init( &indev_drv );
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register( &indev_drv );

    /* Begin code here */
    
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_font(&style, &lv_font_montserrat_48);
    lv_style_set_width(&style, 250);

    lv_obj_t * label1 = lv_label_create(lv_scr_act());
    lv_label_set_text(label1, LV_SYMBOL_OK);
    lv_obj_add_style(label1, &style, 0); 
    lv_obj_center(label1);
}

void loop()
{
    lv_timer_handler(); /* let the GUI do its work */
    delay( 5 );
}
```

earlephilhower, 2022-08-23:
Sorry, I think you need to include an lv_conf.h file in your ZIP.  I just deleted my `~Arduino/libraries` copies of your 2 zip dirs, unzipped them, and tried building the pasted example.  Same problem (with unconfigured I guess):
````
ResolveLibrary(lvgl.h)
  -> candidates: [lvgl@8.3.1]
/home/earle/Arduino/hardware/pico/rp2040/system/arm-none-eabi/bin/arm-none-eabi-g++ -I /tmp/arduino_build_971008/core -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0xf00a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipicow\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefix/home/earle/Arduino/hardware/pico/rp2040/ @/home/earle/Arduino/hardware/pico/rp2040/lib/platform_inc.txt -I/home/earle/Arduino/hardware/pico/rp2040/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0xf00a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10816 -DARDUINO_RASPBERRY_PI_PICO_W "-DBOARD_NAME=\"RASPBERRY_PI_PICO_W\"" -DARDUINO_ARCH_RP2040 "-DSTASSID=\"NOBABIES\"" "-DSTAPSK=\"ElephantsAreGreat\"" -DDEBUG_RP2040_PORT=Serial -DDEBUG_RP2040_WIRE -DDEBUG_RP2040_SPI -DDEBUG_RP2040_CORE -Os -I/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040 -I/home/earle/Arduino/hardware/pico/rp2040/variants/rpipicow -I/home/earle/Arduino/libraries/lvgl/src /tmp/arduino_build_971008/sketch/sketch_aug23a.ino.cpp -o /dev/null
Alternatives for ../../lv_conf.h: []
ResolveLibrary(../../lv_conf.h)
  -> candidates: []
In file included from /home/earle/Arduino/libraries/lvgl/src/../src/misc/lv_log.h:16,
                 from /home/earle/Arduino/libraries/lvgl/src/../lvgl.h:25,
                 from /home/earle/Arduino/libraries/lvgl/src/lvgl.h:17,
                 from /home/earle/Arduino/sketch_aug23a/sketch_aug23a.ino:1:
/home/earle/Arduino/libraries/lvgl/src/../src/misc/../lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
   41 |         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
Using library lvgl at version 8.3.1 in folder: /home/earle/Arduino/libraries/lvgl 
exit status 1
Error compiling for board Raspberry Pi Pico W.
````

victpork, 2022-08-23:
[lv_conf.zip](https://github.com/earlephilhower/arduino-pico/files/9404503/lv_conf.zip)
Here you are, thanks

earlephilhower, 2022-08-23:
This is a GCC error and not something we can help here.  I suggest going to the GCC mailing list and see if anyone else has hit this.

The compile line that fails is `lv_event.c`:
````
earle@amd:~/Arduino/libraries/lvgl$ /home/earle/Arduino/hardware/pico/rp2040/system/arm-none-eabi/bin/arm-none-eabi-gcc -c -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0xf00a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipicow\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -MMD -iprefix/home/earle/Arduino/hardware/pico/rp2040/ @/home/earle/Arduino/hardware/pico/rp2040/lib/platform_inc.txt -I/home/earle/Arduino/hardware/pico/rp2040/include -std=gnu17 -g -DSERIALUSB_PID=0xf00a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10816 -DARDUINO_RASPBERRY_PI_PICO_W "-DBOARD_NAME=\"RASPBERRY_PI_PICO_W\"" -DARDUINO_ARCH_RP2040 -DDEBUG_RP2040_PORT=Serial -DDEBUG_RP2040_WIRE -DDEBUG_RP2040_SPI -DDEBUG_RP2040_CORE -Os -I/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040 -I/home/earle/Arduino/hardware/pico/rp2040/variants/rpipicow -I/home/earle/Arduino/libraries/lvgl/src -I/home/earle/Arduino/libraries/TFT_eSPI -I/home/earle/Arduino/hardware/pico/rp2040/libraries/SPI/src -I/home/earle/Arduino/hardware/pico/rp2040/libraries/LittleFS/src /home/earle/Arduino/libraries/lvgl/src/core/lv_event.c -o /tmp/arduino_build_971008/libraries/lvgl/core/lv_event.c.o
/tmp/ccEi3heN.s: Assembler messages:
/tmp/ccEi3heN.s: Error: unaligned opcodes detected in executable segment
````

Looking into the file, there is a fcn
````
static bool event_is_bubbled(lv_event_t * e)
{
    if(e->stop_bubbling) return false;

    /*Event codes that always bubble*/
    switch(e->code) {
        case LV_EVENT_CHILD_CREATED:
        case LV_EVENT_CHILD_DELETED:
            return true;
        default:
            break;
    }

    /*Check other codes only if bubbling is enabled*/
    if(lv_obj_has_flag(e->current_target, LV_OBJ_FLAG_EVENT_BUBBLE) == false) return false;
    switch(e->code) {
        case LV_EVENT_HIT_TEST:
        case LV_EVENT_COVER_CHECK:
        case LV_EVENT_REFR_EXT_DRAW_SIZE:
        case LV_EVENT_DRAW_MAIN_BEGIN:
        case LV_EVENT_DRAW_MAIN:
        case LV_EVENT_DRAW_MAIN_END:
        case LV_EVENT_DRAW_POST_BEGIN:
        case LV_EVENT_DRAW_POST:
        case LV_EVENT_DRAW_POST_END:
        case LV_EVENT_DRAW_PART_BEGIN:
        case LV_EVENT_DRAW_PART_END:
        case LV_EVENT_REFRESH:
        case LV_EVENT_DELETE:
        case LV_EVENT_CHILD_CREATED:
        case LV_EVENT_CHILD_DELETED:
        case LV_EVENT_CHILD_CHANGED:
        case LV_EVENT_SIZE_CHANGED:
        case LV_EVENT_STYLE_CHANGED:
        case LV_EVENT_GET_SELF_SIZE:
            return false;
        default:
            return true;
    }
}
````

If I make that case have less than 16 entries, it builds fine at `-Os`:
````
    switch(e->code) {
#if 0
        case LV_EVENT_HIT_TEST:
        case LV_EVENT_COVER_CHECK:
        case LV_EVENT_REFR_EXT_DRAW_SIZE:
        case LV_EVENT_DRAW_MAIN_BEGIN:
#endif
        case LV_EVENT_DRAW_MAIN:
        case LV_EVENT_DRAW_MAIN_END:
        case LV_EVENT_DRAW_POST_BEGIN:
        case LV_EVENT_DRAW_POST:
        case LV_EVENT_DRAW_POST_END:
        case LV_EVENT_DRAW_PART_BEGIN:
        case LV_EVENT_DRAW_PART_END:
        case LV_EVENT_REFRESH:
        case LV_EVENT_DELETE:
        case LV_EVENT_CHILD_CREATED:
        case LV_EVENT_CHILD_DELETED:
        case LV_EVENT_CHILD_CHANGED:
        case LV_EVENT_SIZE_CHANGED:
        case LV_EVENT_STYLE_CHANGED:
        case LV_EVENT_GET_SELF_SIZE:
            return false;
        default:
            return true;
    }
````

At 16 entries or more, it fails in the assembler stage.  So that points to an error in handling of case-16-entry-or-more optimizer (my name, but that's the general problem area I believe).

As a workaround, use `-O2` which seems to work fine. You/they might be able to isolate the optimization bug by manually setting the `-Oxxxx` flags (i.e. cse-elimination, etc., not the simple `-O0/1/2/3/s`) and seeing which optimization flag causes it to hit.

victpork, 2022-08-26:
Thanks, for that I won't bother for now as there are alternatives, thanks your your investigation works anyway :)