earlephilhower/arduino-pico #794: Error opening the serial port 'COM12'.(Port busy) Arduino IDE 1.8.19 after sketch uploads

johnmholmes, 2022-08-23:
Hi Earle 
I am trying to use the serial monitor to read inputs in my sketch. The sketch uploads fine on the Cytro maker pi rp2040. But when I try to open the serial monitor it gives me the error message

-----------

Arduino: 1.8.19 (Windows 10), Board: "Cytron Maker Pi RP2040, 2MB (no FS), 133 MHz, Small (-Os) (standard), Disabled, Disabled, Disabled, Disabled, None, Pico SDK, IPv4 Only"

Sketch uses 53484 bytes (2%) of program storage space. Maximum is 2093056 bytes.

Global variables use 7304 bytes (2%) of dynamic memory, leaving 254840 bytes for local variables. Maximum is 262144 bytes.

Resetting COM12

Converting to uf2, output size: 137216, start address: 0x2000

Flashing E: (RPI-RP2)

Wrote 137216 bytes to E:/NEW.UF2

Error opening serial port 'COM12'. (Port busy)

----------

Do you have any idea what is stopping this from opening

earlephilhower, 2022-08-23:
The serial port is SW driven in the Pico, so the act of uploading a file causes the port to disappear, just as if the USB stick was removed quickly.  It then re-appears after upload and restart.  The OS can either keep the same `COM:` or `/dev/tty` mapping, or give it a new address (i.e. `COM13:` instead of `COM12:`).  It's really out of our hands and all on the OS.

I think your error message means that Windows was in the process of destroying COM12 when the IDE tried to re-open it.  I'd check that the Pico hasn't reappeared in a new COM port (use the IDE Tools menu).  In any case, the Pico can't cause a port to remain open on your computer, only the OS or other apps can.  So there's nothing in the core to do here, but do please check the port mapping after upload.

johnmholmes, 2022-08-23:
Hi Earle
Once again thank you for such a quick response to my issue.

I have just retried uploading again and checked the COM port to see if windows had changed it however it was still on the same number. I then tried the  get board in which came back with the details so I knew it was able to report back. I then tried the serial monitor and this time it opened up outputting text as a expected and when I did my input it all worked.

All I can guess is I was trying to open the serial monitor to quickly and windows was not ready

Regards
John