earlephilhower/arduino-pico #795: Deadlock on mutex_enter_blocking/recursive_mutex_enter_blocking

godario, 2022-08-24:
Hi!
Using FreeRTOS and making intensive use of multiple String objects and Serial communication I noticed a deadlock during calls to `mutex_enter_blocking()`/`recursive_mutex_enter_blocking()`
Using platformio and debugging with picoprobe I get the calls stack leading to deadlock: attached two cases as screenshots inside the archive.
I am attaching a MCVE tested both with Arduino IDE and with VSCode & platform.io using the core version 2.4.1.
Can anything be done to prevent this problem from happening? Thank you!

[mutex_enter_stuck_MCVE.zip](https://github.com/earlephilhower/arduino-pico/files/9414525/mutex_enter_stuck_MCVE.zip)

earlephilhower, 2022-08-24:
Thanks for the short-and-sweet MCVE.  I especially liked the variable names.

The 0-effort-on-my-part solution is for you to use a FreeRTOS mutex to protect any writes to the `Serial` device.  That's the general rule you should be following, anyway, as >99% of the libraries on Arduino do not understand multithreaded, let alone multicore.

That said, I implemented a simple mutex for multicore, but not multithreaded, support and that handles the non-FreeRTOS case since there's only 2 jobs in the whole system n the  Seriall device because it was so darn useful to let both cores print stuff.

I haven't run the MCVE, but thinking about it my guess is one task got the serial mutex but got swapped out and another (one or two) are trying for it and locking interrupts waiting for the mutex to free.  When IRQs are locked then the "swap task" interrupt can't reach core 1 and you're stuck.

The change-the-core method probably involves a special codepath on that `CoreMutex` object to make it use the FreeRTOS mutex (which understands priority inversion/etc.) when in FreeRTOS mode.  It's something that's been at the back of my mind for a while, but not something I've had time to look in to yet.

godario, 2022-08-25:
Thanks for the detailed answer.
I suspected that Serial was core-safe between the 2 hardware cores but not between FreeRTOS threads however, I did not imagine that the same problem also afflicted the functions for dynamic allocation.

In fact, as you can see from this other MCVE sketch, the same problem occurs if I call the malloc from multiple threads (running from both cores). To confirm this, by wrapping malloc and free with a mutex (enabling `useMutex`) the execution proceeds without any block.

While I find it more than fair to use a mutex for serial access, I find it somewhat impractical to wrap calls to any library that is supposed to use malloc/calloc/etc. In this case, what do you think would be a good solution at both the sketch and core level?

In the past, on FreeRTOS for SAMD21 ([b6f4240](https://github.com/BriscoeTech/Arduino-FreeRTOS-SAMD21/commit/b6f4240bb3b5f7195ba57670f58e0a03e5c0148f)), I had solved by wrapping the heap functions from the linker in order to use those of FreeRTOS (in that case I had also implemented calloc and realloc since FreeRTOS does not have a thread-safe version of them): do you think this could be a valid and viable path also in this case?

Thank you again!

[mutex_enter_stuck_malloc_MCVE.zip](https://github.com/earlephilhower/arduino-pico/files/9424065/mutex_enter_stuck_malloc_MCVE.zip)

earlephilhower, 2022-08-25:
Can you please give #798 a try, @godario ?  It seems to clear up hangs with the malloc MCVE (not the SerialUART one originally posted).  It is a bit of a heavy hammer but it is the most straightforward to implement.  I believe there is no mutex nesting in newlib, so it should be safe.

earlephilhower, 2022-08-25:
@godario it seems #798 also clears up your Serial1 issue.  I've tested w/o and it hangs and mixes up serial streams, and with the PR it runs forever and the lines are properly separated.

So, I think we're good on both, at least w/the MCVEs given.

godario, 2022-08-26:
Well, I tried to do more in-depth tests, I confirm that with the two previous MCVEs everything seems to work, but with the sketch with which for the first time I got this problem I still have the same problem (it's a stress test for a library which makes intensive use of Serial and String). For this reason I have written a new MCVE which, in addition to `malloc` and `free`, also calls the `realloc` (widely used by the `String` object): unfortunately, this MCVE shows that the problem persists.

**A few details about this MCVE:**
The call to `realloc` can be enabled or disabled through the `tryRealloc` flag.

The acquisition of the mutex near each of the heap functions can be enabled or disabled through the `useMutexOnMalloc`, `useMutexOnRealloc` and `useMutexOnFree` flags.

I have implemented the possibility of using a separate mutex for each of the functions that use the heap, in this `malloc`, `realloc` and `free`. Since the bug in question occurs when any two of these functions are executed at the same time, at the moment I have proceeded to use a single mutex for all of them (in a not exactly elegant way).

To replicate the problem with this new MCVE it is necessary to activate `tryMutex` and disable the use of the mutex on all functions of the heap (the sketch I am attaching already respects this _configuration_). Note that activating the mutex only for the `realloc` does not solve the problem because clearly this would not guarantee that a call to the realloc is not executed at the same time as a `free` or a `malloc`.

Thanks!

[mutex_enter_stuck_realloc_MCVE.ino.zip](https://github.com/earlephilhower/arduino-pico/files/9432319/mutex_enter_stuck_realloc_MCVE.ino.zip)

earlephilhower, 2022-08-26:
We're getting there, if slowly.  I can repro w/your MCVE, thx.

A quick trace shows that it's stuck in your setup() call with the malloc **recursive** mutex.

My thinking is that FreeRTOS's do-not-preempt calls are not really counters and I need to track my own.  The only way I can see malloc getting into that state is to have a recursive release (where count is still > 0) which sends the enable-preempt.

I'll see what I put in.  Might simply be checking the count of the recursive mutex before re-enabling preemption.

earlephilhower, 2022-08-26:
Err, and that recursive part is the issue, I think.  REcursive mutexes in the SDK identify the owner by the Core ID....but w/multithread you need to use the thread ID or two threads on the same core can succeed in the recursive call.  Looks like I need real FreeRTOS recursive semaphores, ugh.

earlephilhower, 2022-08-26:
Please try the latest push to the PR.  You will need to `git submodule update` because there was a FreeRTOS internal change (no need to suspend the task switcher while in malloc/free).

I got all 3 of your MCVEs running OK as far as I can tell (I see all tasks running continually).

godario, 2022-08-29:
Good news: everything seems to work! For the moment I would say that it is solved.
Thank you very much!

commit 92a14fa7 (2022-08-26): Use FreeRTOS muxes for Newlib when enabled

Silently replace non-thread-aware mutexes with their FreeRTOS equivalent
and adjust all calls when building with FreeRTOS

Fixes #795
PR #798 (2022-08-29): Protect core/Newlib mutexes from task preemption under FreeRTOS
Replace core and Newlib mutexes with FreeRTOS mutexes when in FreeRTOS mode.

Fixes #795
commit bde4da2b (2022-08-29): Protect core/Newlib mutexes from task preemption under FreeRTOS (#798)

Fixes #795 

Replace all CoreMutex and Newlib mutex accesses with FreeRTOS calls
when in FreeRTOS mode.  Avoid issues with hange/etc. due to priority
inversion.

No changes to normal operating mode.

Add a FreeRTOS stress test that caught the issue fixed here.
closed 2022-08-29 by earlephilhower