earlephilhower/arduino-pico #796: USB Serial not working with Keyboard

DaleMitchell, 2022-08-24:
If Keyboard is used with Serial, the COM port doesn't show up, at least not on Windows 11. Consequently, the auto-reset for uploading a new sketch doesn't work: to be able to upload code again, the Pi Pico has to be unplugged from USB and power, then the BOOTSEL button has to be held down while plugging in USB. The COM port shows up and works just fine if Keyboard is not included.

Example:

https://github.com/earlephilhower/Keyboard/blob/a6498d6228347ff8337a91b9c1ce934350214dd6/examples/KeyboardPassword/KeyboardPassword.ino
```
/* Released into the public domain */
/* Earle F. Philhower, III <earlephilhower@yahoo.com> */

#include <Keyboard.h>

void setup() {
  Serial.begin(115200);
  Keyboard.begin();
  delay(5000);
  Serial.printf("Arduino USB Password Typer\n");
  Serial.printf("Press BOOTSEL to enter your super-secure(not!) password\n\n");
}

void loop() {
  if (BOOTSEL) {
    Serial.println("Typing password for you...shhhh....");
    Keyboard.print("ThisPasswordIsWeakLikeABaby");
    while (BOOTSEL);
  }
}
```

Andy2No, 2022-08-24:
What happens if you move Keyboard.begin(); to after the delay(5000); ?

DaleMitchell, 2022-08-24:
Moving the delay makes no difference. Merely including the Keyboard header
on a blank sketch causes the Serial to not work.

earlephilhower, 2022-08-24:
I believe this is just a Windows limitation/bug about the way the 2 devices are presented.  The Serial + Keyboard + Mouse + Joystick all are presented as a single unit, not as separate USB devices (I'm getting the terminology wrong, but that's the idea).  Linux and MacOS handle this legal case fine, but Windows IIRC just can't grok it.

@hathach , is this correct?  I think I read other descriptions of the same thing under Windows but my Google-fu is weak today and I can't find them.

dhalbert, 2022-08-24:
It should theoretically work in Windows -- we do this in CircuitPython all the time. We have a composite device. The Keyboard HID device may not be set up properly to be in a composite device? (I have not looked at this code.)

dhalbert, 2022-08-24:
If the keyboard is a boot keyboard it may not like being part of a composite device.

earlephilhower, 2022-08-24:
Ah, here I think it's not the BIOS/UEFI since he's uploaded and only one HID is being created in the OS.  Do you have any pointers to the CircuitPython code I could scrape for USB descriptor init?

dhalbert, 2022-08-24:
Does this help?

https://github.com/adafruit/circuitpython/tree/main/shared-module/usb_cdc
https://github.com/adafruit/circuitpython/tree/main/shared-module/usb_hid
https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/usb/usb_desc.c

This is all pretty complicated because we allow user-provided HID report descriptors, and allow parts of the composite device to be enabled and disabled.

Would dumps of the composite device descriptor and HID report descriptors help? I thought I had some saved but I cannot find them. The same info is in the files above, but in a paramterized form.

earlephilhower, 2022-08-24:
> Would dumps of the composite device descriptor and HID report descriptors help? I thought I had some saved but I cannot find them. The same info is in the files above, but in a parameterized form.

That would help very much, but if it means lots of work then please don't.  I should be able to either work it out manually or actually run a CircuitPython script and dump directly.

dhalbert, 2022-08-24:
This is what an Adafruit QT Py RP2040 running CircuitPython 7.3.2 tells the host when plugged in. The USB composite device includes two CDC devices, MSC, MIDI (as "Audio"), and HID devices.

I feed these hex streams into http://eleccelerator.com/usbdescreqparser/ to parse. Note that the same info is in the CircuitPython source files above, in a different form.

---
### Device descriptor

#### Raw bytes of USB device descriptor (from wireshark):
```
0902da000601008032080b000202020000090400000102020004052400100105240101010424020205240600010705810340001009040100020a00000507050202400000070582024000000904020002080650060705830240000007050302400000090403000203000007092111010001229c0007058403400008070504034000080904040000010100090924010001090001050904050002010300080724010001250006240201010a06240202020009240301030102010b092403020401010100070505024000000525010101070585024000000525010103
```
#### Parsed by eleccelerator:
[eleccelator-usb.txt](https://github.com/earlephilhower/arduino-pico/files/9418313/eleccelator-usb.txt)

#### As reported by `sudo lsusb -vd 239a:80f8`:
[lsusb.txt](https://github.com/earlephilhower/arduino-pico/files/9418282/lsusb.txt)

---
### HID Report descriptors:

#### as reported by `sudo usbhid-dump -d  239a:80f8`
```
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 03
 75 01 05 08 19 01 29 05 91 02 95 01 75 05 91 01
 95 06 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00
 81 00 C0 05 01 09 02 A1 01 09 01 A1 00 85 02 05
 09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
 01 75 03 81 01 05 01 09 30 09 31 15 81 25 7F 75
 08 95 02 81 06 09 38 15 81 25 7F 75 08 95 01 81
 06 C0 C0 05 0C 09 01 A1 01 85 03 75 10 95 01 15
 01 26 8C 02 19 01 2A 8C 02 81 00 C0
```
#### As parsed by eleccelerator:
[eleccelator-hid.txt](https://github.com/earlephilhower/arduino-pico/files/9418326/eleccelator-hid.txt)

earlephilhower, 2022-08-25:
I just tested the password example in the Keyboard library which types the following "secret" password: 
<clicking bootsel now....>ThisPasswordIsWeakLikeABaby

I also see a COM port and the HID just fine.  The serial output is as expected (using putty):
[image]

I think you just need to select the correct port.  For me, with the composite device, Windows can give a different port number, and it will not appear as a "Pico" because the ID is different thanks to being composite.

closed 2022-08-25 by earlephilhower

DaleMitchell, 2022-08-28:
Thank you for getting to the bottom of this. I was confused because the Arduino IDE normally displays `(Raspberry Pi Pico)` at the end of each port number and didn't think to check all the COM parts. I often have more than what is shown in the following screenshot. It would be nice if there was some sort of fix for displaying `(Raspberry Pi Pico)` when Keyboard is used. 
[image]

earlephilhower, 2022-08-28:
That's a good idea.  The device needs a new pid when in composite mode, but I should be able to just the new ones as well in the boards file...