earlephilhower/arduino-pico #797: [Add Board] RC2040 (a RPi Pico RC2014 Emulation)

guidol70, 2022-08-25:
For compiling RunCPM on a RC2040-Board with SD-Support I did create a content for boards.txt and a directory rc2040 for the variants directory (as attachment)

Maybe you will add this board?

RC2040 from @ExtremeElectronics at github:
https://github.com/ExtremeElectronics/RC2040
https://www.extremeelectronics.co.uk/the-rc2040/

earlephilhower, 2022-08-25:
Sorry, but I'm not sure that a new board is needed, or where your boards.txt and variant files are.  It looks like you're using the castellated edges to solder a standard RPi Pico to a carrier, which is a smart way to do it, but if that's the case then there is no need for any boards.txt changes since the board and GPIO mapping are already the Rpi Pico standard, no?

guidol70, 2022-08-25:
> but if that's the case then there is no need for any boards.txt changes since the board and GPIO mapping are already the Rpi Pico standard, no?

No ( I think) - because the maker of the board does use other GPIOs for SPI0 and used mostly all other GPIOs for other purposes (jumper, audio, 8bit pio)

```
// GPIO  0 UART_TX_PIN 0
// GPIO  1 UART_RX_PIN 1
// GPIO  2 SPI0 SCK
// GOIO  3 SPI0 MOSI
// GPIO  4 SPI0 MISO
// GPIO  5 SPI0 CS/SS

// GPIO  6 RC2040 RED User-LED

// GPIO  7 RESET_BUTTON

// GPIO  8 Free(?)

// GPIO  9 DUMP_BUTTON
// GPIO 10 ROM_A13
// GPIO 11 ROM_A14
// GPIO 12 ROM_A15
// GPIO 13 SELSEL
// GPIO 14 SOUNDIO_1
// GPIO 15 SOUNDIO_2
// GPIO 22 HAS_SWITCHES_IO (former SD Card Detect?)

// GPIO 25 onboard LED

// GPIO 28 Free(?)

// PIO:
// GPIO 16 PIO_0
// GPIO 17 PIO_1
// GPIO 18 PIO_2
// GPIO 19 PIO_3
// GPIO 20 PIO_4
// GPIO 21 PIO_5
// GPIO 26 PIO_6
// GPIO 27 PIO_7
```

earlephilhower, 2022-08-25:
I'm still not quite sure what a variants would look like for this.  The variants just define the Dxxx->GPIOxxx mapping and defaults for SPI, UART, etc. ports.  As well as any special startup code (ie. initting an oboard WiFi chip, etc).

Did you make one you'd like to share?

earlephilhower, 2022-08-25:
@guidol70 it looks like you have almost everything done!  Want to do the PR yourself to get credit in the `git log`?

One minor tweak, and this will simplify your life:  Instead of editing `boards.txt` you need to edit `tools/makeboards.py`.  At the bottom it's pretty self-explanatory how to add a new line to build the boards.txt for your board.  Just make sure to please put it in alphabetical order by the vendor.  Rerun your modded .py file, commit the boards.txt update, the variant directory, and the updated makeboards.  And there's the pull request all ready...

If making a PR is a problem, I can do it myself, just let me know.

guidol70, 2022-08-25:
> @guidol70 it looks like you have almost everything done! Want to do the PR yourself to get credit in the `git log`?
> 
> If making a PR is a problem, I can do it myself, just let me know.

Yes I know. I use it already and wouldnt you make much work - so I pre-created the files I know.
Because I didnt know how to add a part to the boards.txt (and wouldnt supply the file as whole thing) I would kindly prefer that you do add this to a PR - because Iam only a github-beginner :(

earlephilhower, 2022-08-25:
Just added a PR.  If @ExtremeElectronics or @guidol70  could eyeball it and verify we can merge easily.

guidol70, 2022-08-25:
> Just added a PR. If @ExtremeElectronics or @guidol70 could eyeball it and verify we can merge easily.

For me all changes look fine/OK ;)

PR #799 (2022-08-25): Add ExtremeElectronics RC2040
Fixes #797
commit 9ff31b91 (2022-08-25): Add ExtremeElectronics RC2040 (#799)

Fixes #797
closed 2022-08-25 by earlephilhower