earlephilhower/arduino-pico #800: Mapping internal flash as a mass storage device

irishgreencitrus, 2022-08-26:
Apologies if I haven't looked into this correctly, but i'm wondering if there's a way to map the RP2040 internal storage to a drive (like how the micro python uf2 does it)?

earlephilhower, 2022-08-26:
You can look at some of the examples in the Adafruit TinyUSB, but there is not something that directly pretends the flash is the world's smallest USB stick, sorry.  If you want to add a PR, that would be great!

If you're talking about uploading programs via UF2, that's what we do already (but this is not the same as making it look like a real USB stick).

closed 2022-08-26 by earlephilhower

dhalbert, 2022-08-27:
I think this may do it for you: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MassStorage/msc_external_flash/msc_external_flash.ino