earlephilhower/arduino-pico #805: Turn on second SerialUSB, need some help

s-t-e-p-a-n-o-v, 2022-08-28:
Hi!
I would like to enable second virtual COM port in arduino-pico core.

So new USB CDC descriptor was added to `cores\rp2040\RP2040USB.cpp`
```
#define USBD_ITF_CDC0 (0) // needs 2 interfaces
#define USBD_ITF_CDC1 (2)
#define USBD_ITF_MAX (4)

#define USBD_CDC_EP_CMD0 (0x81)
#define USBD_CDC_EP_CMD1 (0x84)
#define USBD_CDC_EP_OUT0 (0x02)
#define USBD_CDC_EP_OUT1 (0x05)
#define USBD_CDC_EP_IN0 (0x82)
#define USBD_CDC_EP_IN1 (0x85)
#define USBD_CDC_CMD_MAX_SIZE (8)
#define USBD_CDC_IN_OUT_MAX_SIZE (64)
```
EP values are from @Noltari project [pico-uart-bridge](https://github.com/Noltari/pico-uart-bridge/blob/master/usb-descriptors.c)

and CFG_TUD_CDC was changed to 2 in `include\tusb_config.h` and `tools\libpico\tusb_config.h`
`#define CFG_TUD_CDC              (2)`

After flashing only one port is available in system, what can be wrong ? Could you point to any ideas/guidelines how to do it ?

Many thanks!

earlephilhower, 2022-08-28:
That's not going to work, as you found out.  The Pico SDK was configured and pre-built for the common use case and editing the header will (if you're lucky) do nothing, or crash since internal definitions may be different between the SDK and the core.

For something like this, the Adafruit_TinyUSB library that's included is the way to go.  Check out their examples, they should get you going.  You'll need to manually make all CDC serial ports, etc. of course, but will have the flexibility to change the # exposed.

Good luck!

closed 2022-08-28 by earlephilhower

s-t-e-p-a-n-o-v, 2022-08-28:
Thanks!

Sure, Adafruit_TinyUSB is an option, but it has some issue with [arduino-pico](https://github.com/earlephilhower/arduino-pico/issues/720) and isn't compatible with FreeRTOS.
That's why I tried to configure Pico SDK USB stack.

earlephilhower, 2022-08-28:
I don't recommend it, and I'm not sure how the core will handle it, but you can edit `tools/libpico/tusb_config.h` and rebuild `libpico.a` using the script in there.  Then, it's up to you to figure out how to hack the core SerialUSB to work with different CDCs.  The TinyUSB calls used in the object don't actually support identifying which CDC object is being used, but there may be other ones you can swap in.