earlephilhower/arduino-pico #822: error: 'SDFS' does not name a type

KenwoodFox, 2022-09-01:
I see SDFS in my dependancy manager but im unable to specify an included type? What did i configure wrong or is this a bug?

```
pio run
********************************************************************************************************************************************************************************************************************
Obsolete PIO Core v6.1.4 is used (previous was 6.1.5a1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
********************************************************************************************************************************************************************************************************************
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board: pico)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0+sha.de1a5ef) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.20303.0+sha.be18c76 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.100300.220714 (10.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- FreeRTOS @ 1.0.0
|-- SDFS @ 0.1.0
|   |-- SPI @ 1.0
|   |-- ESP8266SdFat @ 2.1.1
|   |   |-- SPI @ 1.0
Building in release mode
Compiling .pio/build/pico/src/main.cpp.o
src/main.cpp:18:1: error: 'c2' does not name a type
   18 | c2.setCSPin(4);
      | ^~
src/main.cpp:19:1: error: 'SDFS' does not name a type
   19 | SDFS.setConfig(c2);
      | ^~~~
*** [.pio/build/pico/src/main.cpp.o] Error 1
============================================================================================ [FAILED] Took 0.62 seconds ============================================================================================
```

```cpp
...
#include <Arduino.h>
#include <FreeRTOS.h>

// Filesystems and SD
#include <SDFS.h>

// SDFS Settings
SDFSConfig c2;
c2.setCSPin(4);
SDFS.setConfig(c2);

void setup()
{
    ...
```

earlephilhower, 2022-09-01:
````
// SDFS Settings
SDFSConfig c2;
c2.setCSPin(4);
SDFS.setConfig(c2);
````

That's not legal, you're calling member functions in global scope.  Move the setcspin/setrconfig into `setup()`

KenwoodFox, 2022-09-01:
That compiles! Thanks for the help, im not sure the [docs](https://arduino-pico.readthedocs.io/en/latest/fs.html#setconfig) made this super clear but it makes sense now. Closing this issue.

closed 2022-09-01 by KenwoodFox