earlephilhower/arduino-pico #826: Reboots 2 seconds after loading (Feather RP2040)

tloose, 2022-09-01:
I am trying to set up my Arduino IDE to program the Feather RP2040 using the Philhower core. I don't get a PORT in the IDE, but my WIN10 PC shows COM3 created. I CAN upload sketches using the IDE. When I do, they run for about two seconds then stop (reset?).

If I power the board from a wall supply after uploading the sketches run correctly.

If connected to the PC a reset OR and double-tap reset causes the same behavior - run for a couple of seconds then stop.

Any ideas what's going on??

I had a similar issue when trying to use CircuitPython, except it would only show the CircuitPython drive for 2 seconds then revert to RP drive.

Andy2No, 2022-09-01:
@tloose Windows can power down USB ports that it thinks aren't in use, but it's not normally that aggressive. It can cause problems with large transfers to USB thumb drives, for example.

I'm much more familiar with Windows 7, but the place to look is the same, AFAIK.  You go into Device Manager, on the control panel and expand the list of USB devices.  On that list, you'll see some USB hubs and root hubs.  If you right click one and choose Properties, one of the tabs is called Power Management, and has a tick box to enable automatic power management.  In Windows 7, it says "**Allow the computer to turn off this device to save power**".  If you see any that are ticked, untick them, then press Okay.

It's not always obvious which one is connected to what, but if that seems to cure it, you can go back and enable them one by one, if you want to, to find which is the culprit.  I just leave it disabled for all of them.

earlephilhower, 2022-09-01:
This looks like a HW problem, most likely the USB cable.  We had tons of issues like this on the ESP8266.

Try a different USB cable or a different USB port (ones on the motherboard I/O panel vs. the front panel) as well.  Some cables have wires so thin and are made of such poor electrical materials that the voltage droops at the board below safe levels.

There's also the possibility you have a dud board (but powering via external USB seems to work so this seems unlikely).

There's not much else we can do.  Good luck!

closed 2022-09-01 by earlephilhower

tloose, 2022-09-02:
> @tloose Windows can power down USB ports that it thinks aren't in use, but it's not normally that aggressive. It can cause problems with large transfers to USB thumb drives, for example.
>
> I'm much [...]

tloose, 2022-09-02:
Hmm, I tried disabling all the automatic power management - had no effect.  Under "Other Devices" I see "RP2 Boot" with an error "The drivers for this device are not installed" and "there are no compatible drivers for this device".
Also, this board had MicroPython installed before; I attempted to run the flash_nuke.uf2 before.

Andy2No, 2022-09-02:
> Under "Other Devices" I see "RP2 Boot" with an error "The drivers for this device are not installed" and "there are no compatible drivers for this device".

Try the Zadig method, in #96 - it seems to apply to Windows 10 too.

tloose, 2022-09-02:
OK, I tried the Zadig method, and sucessfully loaded a driver for the RP2 Boot device.
Maybe I am looking at this the wrong way; There are .UF2 files to install for MicroPython and CircuitPython; is there a corresponding file for the Arduino IDE?  Or, do I "nuke" the bootloader?

Andy2No, 2022-09-02:
I've never used MicroPython, so I'm not clear if it's a different boot loader.

One thing you can do is set an output foler that's easier to find, in the preferences.txt file (File->Preferences, click the link to open it, exit Arduino then edit and save it).

Add or edit this line, near the top of that file:

```
build.path=C:\arduino-output
```

When you compile a program, including by pressing the tick icon to compile without uploading, the .uf2 file should get placed in that folder.  You should then be able to upload it by booting the Pico, or similar, with the Boot button held down (IIRC), to get a removable drive, then dragging it to that drive.

I've done that just to try it, in the past, but I'm not sure if there's any benefit to doing it, other than to save the time needed to compile it again.  Having to press the Boot button is a bit awkward.

earlephilhower, 2022-09-02:
The bootloader is in ROM on the Pico.  In that mode it makes a USB device with a) a fake mass storage device for uploading UF2s and b) a (serial I think) port  for doing the same thing, only using the `picotool` uploader.

(a) is used by the IDE and should no drivers at all.  It looks like a USB stick to the OS.
(b) is not used by the IDE but is used by platform.io.

There are no special portions of the flash, so no need to ever "nuke" it.  Uploading a new UF2 is all you'll ever need.

tloose, 2022-09-02:
I just tried setting the board up for CircuitPython, with the same result: the circuitpython drive shows up in file explorer for 2 seconds then gets replaced by the RPI-RP2 drive.  I then tried the same process on my wife's laptop (never had any SW tools installed) and the circuitpython drive stays running.  Something on my PC is the issue!  Maybe drivers?  Maybe wipe Arduino and reload a clean copy??

tloose, 2022-09-02:
Thank for keeping in touch with me on this.  There's a general lack of
support/documentation/help.

I uninstalled the Arduino IDE and reinstalled it, then reran the zadig
procedure.  The board started to run the code without hanging up, but no
way to load any .UF2 files.  I tried some more changes using Zadig, and now
the code does not run on the board, and I get no bootloader directory in
Windows either - stepped backwards.
My board shows up in Device Manager as RP2 Boot (Interface 0) and RP2
(Interface 1).  What do you know about that?  Any ideas??

earlephilhower, 2022-09-02:
I have never had a reason to run Zadig on my own W10 box, nor has anyone else AFAIK.  You might have busted the associations for the VID/PID.  Can you undo whatever changes you did?

The USB stick mode (UF2 upload) needs no drivers at all, under any OS.  It's really just a USB stick as far as the OS knows.

The CDC serial port needs no drivers on Linux, Win10+, or MacOS.  It's as standard and boring as you can get.

That said, Win 7 (EOL for >1yr now?) did not have a driver for a generic CDC serial port, though, and does need Zadig to make it work as a COM port.

Andy2No, 2022-09-02:
@earlephilhower I appeared to need Zadig to get my first RP2040-Zero to work, on Win10 (on my "new" second hand PC), on a fresh install of Arduino 1.8.19, with the version of arduino-pico that was current at the time... maybe a couple of weeks ago.

Perhaps that was a misunderstanding on my part, but I wasn't able to get it to work until I did that.

tloose, 2022-09-02:
ANdy2NO:  What exactly did you need to do with Zadig?  Did you find drivers
like "RP2 Boot"??

tloose, 2022-09-03:
GOT IT WORKING!

The trick was to configure RP2 Boot interface 0 as mass storage, and
interface 0 as COM.

Thanks for pointing me in a good direction - I really appreciate it!

tloose, 2022-10-11:
Hmm, spoke too soon.  The system has reverted to similar behavior.  I can
start the board in BOOTSEL mode and upload code.  However, if I keep the
USB cable attached and reset the board only runs for a few seconds and
returns to BOOTSEL mode (although with RPI-RP2 available now).