earlephilhower/arduino-pico #832: USB Serial (Adafruit TinyUSB CDC) isn't always reliable if I use the 2nd Pico CPU

Jasoroony, 2022-09-03:
Hi Earle, nice work on this repo!  I have a question though...

RE: post number 2 at: https://github.com/hathach/tinyusb/issues/1618

And my slightly more complex new Arduino Pico test (this one uses both CPUs, in that I think CPU0 handles the timer task and interrupt including rx, while CPU1 initiates the USB tx operation[s]):

```cpp
#include <Adafruit_TinyUSB.h>

//#define SZ 16
#define SZ 3

static unsigned char buf [64];

void setup () {
    delay (2500);
    Serial.begin (115200);
}

void loop () {
}

void setup1 () {
    delay (2600);
}

void loop1 () {
    int i;
    int c;

    if (Serial.available () >= 64) {
        Serial.read (buf, 64);
        i = 0;
        while (i < SZ) {
            c = 64;
            while (c) {
                c -= Serial.write (buf, c);
            }
            i++;
        }
    }
}
```
(Original test was sending 16 times the amount of received data - like my actual failing use-case, but 3 times actually tests things better.)

I'm speculating that a lot of my issues I see with my stress-test above (both here and in many other devices I'm trying to get working in my project that are failing - see post 1 back at TinyUSB) is due to race conditions (in RTOS tasks or the interrupt and a 2nd CPU).

For a Python host side script (may need some editing, sorry but I use a C host-side exe for testing but they prefer Python, and I'm not entirely sure theirs is as simple as mine - which is sort of a good thing - but I don't care what data comes back, only that it's the right amount otherwise the write(buf) would be more complex here) see this post where the ESP32-S2 guys have repeated my test:
https://github.com/espressif/arduino-esp32/issues/6221#issuecomment-1090188325

Would you care to have a ponder and let me know what you think?

earlephilhower, 2022-09-03:
I really haven't looked at the Adafruit TinyUSB at all.  Can you repeat your test w/o it, just to level set (make sure to use the Pico SDK usb settings in the menus)?  Your code right now is basically running completely out of the core (the Adafruit version handles all IRQ, timer, etc.).

Jasoroony, 2022-09-03:
There doesn't appear to be a read (buf, len) method, and adding one didn't make it any faster than a 64 iteration loop in the sketch building buf up character by character either.  Both appear to run at about 50KB/S and I normally would see a fairly big improvement by using the right function.  As it stands, this stack doesn't seem to be very racy to me, and I'm sort of feeling that's by some other design now.  The other stack gets about 650KB/S which is approaching the full-speed you might expect.  I'll let it run for a while, but even SuGlider had suggested in the ESP32 thread that slowing it down using a delay also worked for him.

We'll see.

Jasoroony, 2022-09-03:
While it has been running (slowly, but without issue) I did some edits and restarted.  I moved the tud_task() out of the 1ms timer and into my now empty (CPU0's) loop along with the mutex_try_enter() etc.  It's now runs at ~630KB/S (varies, but all the devices I've been using do that).

It's up to 1.5GB transferred - and still going strong.  I'll leave it run for a while yet though, but yours appears to work much better with the mutex logic than the other one without.  Have you considered moving tud_task() into the loop handler instead of on a slow(ish) timer?  I don't think it uses many cycles if there are no messages to process?

Jasoroony, 2022-09-03:
It got to 4GB transferred without any trouble, where my counter wrapped for the first time ever.  I've added the "Big, global USB mutex" :) to my Raspberry Pi 4B version (a userspace UIO driver I configured to attach to the spare USB controller on its power port, using an overlay to name & enable it) and it seems to be much happier now too.  It only gets ~400KB/S but I think that's due to the time it takes for the high-priority thread reading the UIO interrupt signal file handle to wake...  If I run that in a busy loop (and burn a CPU) it does over 800KB/S in FS mode and 1600KB/S in HS mode.

I think I'm good now, as that was my end goal - so thank you.  You had all the missing bits in place already, in your version.

earlephilhower, 2022-09-03:
@Jasoroony thanks for the update.  That's why I asked to use the core version, I spent a lot of time getting mutexes around dangerous things.  Speed came a distant second to correctness, because debugging concurrency issues is a massive headache.

I think your core0 `loop()` could actually be removed just by adding
````
tud_task()
````
to all the `SerialUSB` methods after acquiring and checking the `CoreMutex`.

For example:
````
int SerialUSB::read() {
    CoreMutex m(&__usb_mutex, false);
    if (!_running || !m) {
        return -1;
    }

    tud_task();
    if (tud_cdc_available()) {
        return tud_cdc_read_char();
    }
    return -1;
}

int SerialUSB::available() {
    CoreMutex m(&__usb_mutex, false);
    if (!_running || !m) {
        return 0;
    }

    tud_task();
    return tud_cdc_available();
}
````
Would you be able to try that w/o the Core0 loop and report back on speeds?

earlephilhower, 2022-09-03:
You also might want to open an issue w/the Adafruit_TinyUSB upstream library.  They may have a good idea how to implement mutex protecting on the 10s of MCUs they support.

Jasoroony, 2022-09-04:
> Would you be able to try that w/o the Core0 loop and report back on speeds?

~630KB/s - So... Very good...

But you need to be careful how you implement this, before the device is mounted on the host it needs the tud_task() to respond to all the setup requests - and I've got large delays in my setup to make sure it's mounted before I write the first "Hello" message etc.  Leave the timer code in place too maybe?

Also, I had to wrap the DWC2 version's interrupt code in the big global mutex (which in my case is a thread blocked on a file descriptor, so "not hard") because I still got the odd failure without that.  I can see the device's endpoint interrupt enable register (the single one for all endpoints) is being updated (read-modify-write) by rx/tx initial setup, and the interrupt, without atomic/spinlock code.  Bad, bad, not good.

The whole thing's a bit of a mess, at least for DWC2.  I've not checked what registers are updated in each place on the Pico's USB device.  But I'm now fighting the Raspberry Pi 4B.  Thanks.

earlephilhower, 2022-09-04:
Thx.  I'll see about adding that kick to the USB stack...

closed 2022-09-04 by earlephilhower

Jasoroony, 2022-09-04:
FYI (Had to look, the Pi 4B stuff was going well, for a change...)

https://github.com/hathach/tinyusb/blob/master/src/portable/raspberrypi/rp2040/rp2040_usb.c#L41-L45

TU_ATTR_ALWAYS_INLINE static inline void _hw_endpoint_lock_update(__unused struct hw_endpoint * ep, __unused int delta) {
    // todo add critsec as necessary to prevent issues between worker and IRQ...
    //  note that this is perhaps as simple as disabling IRQs because it would make
    //  sense to have worker and IRQ on same core, however I think using critsec is about equivalent.
}

earlephilhower, 2022-09-04:
Good debug.  The core protects against itself doing anything re-entrant to the USB stack, but this is fully TinytUSB internal.

Therefore, **every RPi Pico program out there**, not just this core, would be affected.  I'm not really hearing or seeing any reports of issues here or the Pico-SDK repo.  So, maybe it's not really needed, or it's just an insanely rare occurrence where the lack of locking causes an issue.

@hathach what's your take on this?  Should this be brought up to the Rpi Pico folks?

For this core I think using the global user-level TUSB mutex + adding disable/enable interrupts will take care of things.  For FreeRTOS we also need to stop any SVC exception driven task switching (like was needed here to safely write flash in FreeRTOS) as well as disable IRQs.

hathach, 2022-10-10:
@earlephilhower thank for the mention, sorry for late response (was busy). This mutex place holder function is initially added by rpi team. I think they intended to implement it but lack an reproducible scenario to test and verify the code. Like most other things, it got forgotten until discovered. Is the issue still reproducible (or at which core version I could reproduce this), I will try to fix it whenever I could.