earlephilhower/arduino-pico #834: Re-assign UART pins?

ElDuderino, 2022-09-04:
Is it possible to re-assign the UART pins during runtime to connect to another peripheral? For example:

//connect to primary external peripheral device
Serial1.setRX(1);
Serial1.setTX(0);
Serial1.begin(9600);
//talk to primary peripheral device
//end and connect to secondary external peripheral
Serial1.end()
Serial1.setRX(13);
Serial1.setTX(12);
Serial1.begin(9600);
//talk to 2nd peripheral device

Is something like this supposed to work? 

(BTW Thank you for the excellent arduino port!!)

earlephilhower, 2022-09-04:
It should work, just make sure the pins you are assigning are legal.  If you have a failing case, please make an MCVE so it can be reproduced and debugged here.

Only thing I can think of that might be an issue is the unassigned pins might not be reset to INput.  You can use the usual pinMode call in that case.

ElDuderino, 2022-09-04:
I have been unable to make it work. (I resorted to using SerialPIO for the second peripheral instead for now). I do not have an MCVE yet, My code is a port from a device with several external library dependencies that previously used SoftwareSerial on AVR. I ported those to use HardwareSerial, but could not get the code working using UART0 and pin re-assignments during run-time. (Although I did confirm each device worked as long as I did not re-assign the pins during run time). 

I will try the pinMode calls first.

If that doesn't work, can you suggest the best way to provide an MCVE for this case?

earlephilhower, 2022-09-04:
What exactly doesn't work?  Original pins transmit + nothing on the new pins, original pins + new pins do nothing, original  + new send copied data, etc?

earlephilhower, 2022-09-04:
@ElDuderino can you give #836a try?  The GPIO pins could have been left connected to the original pins after Serial1/2.end(), and this patch saves and restores the original mapping (GPIO normally) on ::end.

ElDuderino, 2022-09-05:
Hello! #836 has resolved the issue for me now. 

My sincere thanks!!

commit 8a6a602e (2022-09-04): Restore GPIO functions on SerialUART::end

Fix #834
PR #836 (2022-09-05): Restore GPIO functions on SerialUART::end
Fix #834
commit 85d39cf2 (2022-09-05): Restore GPIO functions on SerialUART::end (#836)

Fix #834
closed 2022-09-05 by earlephilhower