earlephilhower/arduino-pico #837: Bricked with Blink

Spartacus1066, 2022-09-05:
**Arduino Nano 2040 Connect**
I uploaded Blink with settings as _TinyUSB_ & _16mb Flash._

**Blink** works, so it isn't bricked, but I cannot re-enter the **_programming mode_**. No drive shows up, no COM port. I hold down **BOOTSEL** while attaching USB and get nothing. I even attached an **FTDI board** hoping it would talk to **_Serial1_**. Nope. It just keeps Blinking.

The **BOOTSEL** button does pause the bling, so it is resetting.

earlephilhower, 2022-09-05:
There is no software way of bricking the RP2040.  The bootloader, which checks the BOOTSEL button and goes to MSD mode (not serial, only USB stick) is in ROM and runs before everything.  You might have a busted BOOTSEL button or USB cable issue.  You can try a different cable and port, and failing that your next option would be to solder on a SWT connector and use PicoProbe to do uploads/etc.  Good luck!

closed 2022-09-05 by earlephilhower

Spartacus1066, 2022-09-05:
Thanks for the reply. I appreciate your work. But I tried 2 micro USB cables that I use for Trinket m0 and Nano 33 iOT without issue including the cable I used to load Blink using this core onto the Nano Connect 2040. And the BOOTSEL button makes the board pause the "Blink" sketch and restart immediately. 
It isn't "bricked" but cannot be accessed via USB or Serial1, I will check out the SWT option - that's Greek to me right now :)

earlephilhower, 2022-09-05:
Does the Arduino board have both BOOTSEL and RESET buttons?  If the BOOTSEL button causes it to do anything at all while running, then you have a hardware problem on the board.  Basically BOOTSEL just shorts the CS line of the SPI flash to enabled....which is exactly what it is supposed to be in normal operation.  If it causes a reboot while running something is wrong with the board.  The microswitches on the Rpi Pico boards seem to get a lot of complaints as they're not the most sturdy things out there.  Maybe the same kind is used on the Arduino board?

Also, I meant to say "SWD" (single wire debug) which is the standard Arm debugger interface.  3-pins, lets you run HW-assisted GDB, load flash, etc.

LinusHeu, 2023-09-18:
The Arduino Nano RP2040 connect does not have a Bootloader button. The button is only "reset". Double tapping it enters the bootloader. (Not sure if that also works with this core.)
The pad "REC" works like a bootloader button on other RP2040 boards.
See https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-01-technical-reference#forcing-bootloader