earlephilhower/arduino-pico #839: [Feature Request] Enabling static allocation in FreeRTOS

godario, 2022-09-06:
Hi,
I was trying to create a task using static allocation ([`xTaskCreateStatic`](https://www.freertos.org/xTaskCreateStatic.html)) but I saw that this feature is currently disabled: Is there any particulare reason why this feature is currently disabled?
Taking into account what is written [here](https://www.freertos.org/a00110.html#configSUPPORT_STATIC_ALLOCATION) and looking around into your FreeRTOS implementation, currently everything should be in order to enable static allocation without further changes.
I tried to enable `configSUPPORT_STATIC_ALLOCATION` in `FreeRTOSConfig.h` and write a small example of statically created task and everything seems to work.

Is there anything that escapes me that can be implemented in order to fully support static allocation?
Thank you.

[static_task_example.zip](https://github.com/earlephilhower/arduino-pico/files/9495973/static_task_example.zip)

earlephilhower, 2022-09-06:
@godario would you like to make the PR with this change and your example?  I had it disabled only because I thought that it made it *mandatory* to statically allocate, not *optional.*  So, I'm fine with enabling the add'l support if it doesn't break anything.

earlephilhower, 2022-09-08:
Thanks for the PR!

closed 2022-09-08 by earlephilhower