earlephilhower/arduino-pico #850: wifi/OTA not useable on Arduino RP2040 Connect

AcThPaU, 2022-09-11:
When using WiFi.h from the core:
(using example: "WebUpdater.ino")

```
C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved2022811-16204-iif12u.tkv3r\WebUpdater/WebUpdater.ino:29: undefined reference to `_ZN9WiFiClass5beginEPKcS1_'
```

When using WiFiNINA.h with ArduinoOTA from the core:
```
#include <WiFiNINA.h>
#include <ArduinoOTA.h>
```

```
C:\Users\xxx\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.5.2\libraries\ArduinoOTA\src\ArduinoOTA.cpp:30:10: fatal error: include/UdpContext.h: No such file or directory
   30 | #include "include/UdpContext.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Hardware:
Arduino RP2040 Connect

To reproduce:
1. try compile anything that includes WiFi.h
2. try compile anything that includes ArduinoOTA.h

earlephilhower, 2022-09-12:
I get your point, but I'm a little not sure what the right thing to do is here.  The Arduino Nano boards have their own proprietary OTA, no?  While I can probably get things running here w/the PicoW infrastructure, I'm not sure if that's the right thing to do.  That is, the OTA setup here is from scratch and not part of the Arduino online IOT stuff.

So, does it make sense to make the Nano work w/OTA, too?  Even if it's different than the Arduino IOT stuff?  Anyone have any thoughts?

JAndrassy, 2022-09-12:
@AcThPaU delete the bundled ArduinoOTA library

AcThPaU, 2022-09-16:
@JAndrassy  I deleted ArduinoOTA library from the pico core and using your library, but MCU hangs when WiFi.localIP() is called.

JAndrassy, 2022-09-16:
WiFiNINA's WiFi.localIP()?

AcThPaU, 2022-09-19:
> 

Yes, it works in Nano 33 IoT, but not in RP2040 Connect.

AcThPaU, 2022-09-19:
> WiFiNINA's WiFi.localIP()?
Yes, it works in Nano 33 IoT, but not in RP2040 Connect.

lapfelix, 2022-09-22:
I have a Raspberry Pico W and I get this same compilation error when I have `#include <WiFi.h>` in my code. The WifiClient example doesn't compile, for instance.

```
[arduino_path]/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: [sketch_tmp_folder]/libraries/WiFi/WiFi.a(WiFiClass.cpp.o): in function `_ZN9WiFiClass5beginEPKc':
[arduino_path]/packages/rp2040/hardware/rp2040/2.5.4/libraries/WiFi/src/WiFiClass.cpp:68: undefined reference to `_ZN9WiFiClass5beginEPKcS1_'
[arduino_path]/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: [sketch_tmp_folder]/libraries/WiFi/WiFi.a(WiFiMulti.cpp.o): in function `_ZN9WiFiMulti3runEm':
[arduino_path]/packages/rp2040/hardware/rp2040/2.5.4/libraries/WiFi/src/WiFiMulti.cpp:98: undefined reference to `_ZN9WiFiClass5beginEPKcS1_'
```

AcThPaU, 2022-09-24:
> I have a Raspberry Pico W and I get this same compilation error when I have `#include <WiFi.h>` in my code. The WifiClient example doesn't compile, for instance.
> 
> ```
> [arduino_path]/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: [sketch_tmp_folder]/libraries/WiFi/WiFi.a(WiFiClass.cpp.o): in function `_ZN9WiFiClass5beginEPKc':
> [arduino_path]/packages/rp2040/hardware/rp2040/2.5.4/libraries/WiFi/src/WiFiClass.cpp:68: undefined reference to `_ZN9WiFiClass5beginEPKcS1_'
> [arduino_path]/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: [sketch_tmp_folder]/libraries/WiFi/WiFi.a(WiFiMulti.cpp.o): in function `_ZN9WiFiMulti3runEm':
> [arduino_path]/packages/rp2040/hardware/rp2040/2.5.4/libraries/WiFi/src/WiFiMulti.cpp:98: undefined reference to `_ZN9WiFiClass5beginEPKcS1_'
> ```
> 
> EDIT: agh it compiles fine with the Arduino IDE v1 (not with v2) so it's probably something on my end

I use Arduino IDE v2, and I have the same problem with RP2040 Connect, which has WiFiNINA(ESP32), your Pico W has CYW43439. I think the library only supports ESP8266.

JAndrassy, 2022-09-25:
`#include <WiFiNINA.h>` for WiFi on RP2040 Connect

AcThPaU, 2022-09-26:
> `#include <WiFiNINA.h>` for WiFi on RP2040 Connect

WiFiNINA.h with your library make RP2040 Connect hang at WiFi.localIP().

WiFiNINA.h with your library works fine on Nano 33 IoT.

closed 2022-11-30 by earlephilhower

AcThPaU, 2022-12-21:
using @JAndrassy 's OTA library: #include <WiFiNINA.h>

MCU still hung at WiFi.localIP()