earlephilhower/arduino-pico #853: [Q] Pico W connects to ifi only every second boot.

mionut22, 2022-09-12:
Hello Earl,

First let me congratulate you for the great job!

I have a question, this is not an  issue, I can live with it, I've built a workaround in my code for it.

Connection to WiFi is made only every second boot of the PicoW. If Wifi is connected and the device reboots (hard or soft), after the boot it does not connect to Wifi. After a second boot the connection is succesfull and it's working without problems. Then again, after a reboot, Wifi does not connect, but after another boot the connection is again established. Is this a known problem with the HW Wifi driver for the PicoW?

Thank you, and keepup the great job!

earlephilhower, 2022-09-13:
I've not seen this happen and I don't think it's common.  Can you give an MCVE?

earlephilhower, 2022-09-13:
I think you're just hitting a timeout on the WiFi connection, which is going to be a function of the Pico W and the router,  Some just seem kind of slow to react and you can timeout.

Try adding `WiFi.setTimeout(15000);` before the `WiFi.begin` and you might just be fine.  You could also attempt to reconnect w/o a reboot which might succeed quickly enough.