earlephilhower/arduino-pico #854: attachInterrupt does not work on GPIO27

AcThPaU, 2022-09-12:
Hardware: Arduino RP2040 Connect

Note that this works fine with Mbed core

To reproduce:

Attach a pulse source on GPIO27

```
volatile uint32_t CNT;

void setup(){
    Serial.begin(115200);
    attachInterrupt(digitalPinToInterrupt(27), ISR_func, RISING);
    //or attachInterrupt(27, ISR_func, RISING);
}

void ISR_func(){
    CNT++;
}

void loop(){
    Serial.println(CNT);
}
```

earlephilhower, 2022-09-12:
I’m traveling, but I think your problem is simply the pin you’ve selected.  PIN numbers are the GPIO coming off the chip.  Try D27 instead of just 27.  Also make sure you set the pin to INPUT.

AcThPaU, 2022-09-16:
Found the problem. For attachInterrupt to work, pinMode must be called.

However, this is not the case in Arduino official cores.
#-----------------------------------------------------------------------------------------------------------------
About pin name in pico core, numbers with no prefix means GPIO number; number with D prefix means Arduino board digital pin number.

In Mbed core, number with no prefix means Arduino digital pin number; number with D prefix is not defined.

earlephilhower, 2022-09-16:
Per the example on the Arduino site for `attachInterrupt` (and other examples and cores I've checked) you do need the pinMode to set to an INPUT of sime kind:
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

So this looks good as-is in the core.  Good debug!

closed 2022-09-16 by earlephilhower