earlephilhower/arduino-pico #856: Can I burn the program via UART, SPI, IIC or other methods other than via USB?

Arya11111, 2022-09-14:
hello，I want control others chip's interface(UART, SPI) to update rp2040's User program. What can I do?

earlephilhower, 2022-09-14:
Sure.  You need to save the new BIN to the filesystem (say, using ZMODEM or something in your app on `Serial1`.  Then, use the `PicoOTA` library to tell the OTA loader to program the new image,  See https://github.com/earlephilhower/arduino-pico/tree/master/libraries/PicoOTA/examples/OTAfromFile

closed 2022-09-14 by earlephilhower

Arya11111, 2022-09-14:
We have a project, we need it to support U disk to burn Arduino program, but also to update its program through other MCU, we can't modify its user program. We want to handle the update in its boot, rather than actively telling it to update in the user application. 
what I mean is that the rp2040 is in the U disk mode without burning any app program for the rp2040. In this case, does it support operating its UART through an external controller, or SPI and other direct connection methods to update application without affecting its USB flash drive mode.
What can I do?

earlephilhower, 2022-09-14:
I would say no.  In BOOTSEL mode it only accepts data over USB (using the Picotool protocol or the MSD mode https://github.com/raspberrypi/pico-bootrom .

You could try the RPI forums proper, since your need is unrelated to Arduino side: https://forums.raspberrypi.com/

Thinking out of the box, you could possibly hold the Pico in reset and directly program its SPI flash by tapping into the QSPI bus.  Youd need to make sure the SPI bus is held tristate while in reset, but the RP2040 datasheet should have this info.

You could also use SWD to stop the cores, upload a new program to flash, and reset the Pico.

Good luck!