earlephilhower/arduino-pico #857: Pico problem - s/w reset does not work

bwims, 2022-09-14:
This is essentially a restatement of https://github.com/earlephilhower/arduino-pico/issues/5 , which was closed without solution (that worked for me, at any rate).

```
Sketch uses 50676 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 7268 bytes (2%) of dynamic memory, leaving 254876 bytes for local variables. Maximum is 262144 bytes.
Resetting /dev/ttyACM0
Converting to uf2, output size: 131584, start address: 0x2000
No drive to deploy.
An error occurred while uploading the sketch
```

Clearly the software reset is the problem, because when I hold the button, remove and replace the pico, the disk is remounted.

The only way I can get this to work is  use sketch - export binary, then reset the pico and copy the uf2 across.  Hardly satisfactory.

The blink sketch **does** recreate /TTY/ACM0, so that bit works... we just need the s/w reset to work.

I'm running on a pi4 with running  Arduino prior to 1.8.19

I have udisks2 mounting the pico into /media/$USER/RPI-RP2
When I do a manual reset, udiskie automatically remounts it.

Can you confirm that s/w reset works correctly under those circumstances?

Thanks!

earlephilhower, 2022-09-14:
Sorry, but I don't think I could find a RPI 4 for love nor money today.  That said, the reset bit is trivial and not seen to be an issue with Windows, Linux x64 or Mac generally.

One possible thing, are you using the Adafruit USB stack?  You need to manually enable the Serial port or it won't export anything to reset (via a 1200bps toggle).

The other note is that if the chip is crashed, the SW reset can't work either (since the cores run the USB stack internally, not a separate chip).

bwims, 2022-09-14:
Hi, thanks for replying!

I wasn't aware of the Adafruit USB stack!  I'd been waiting for the Arduino support as I have no interest in micropython, so not up to speed with those developments.  I will investigate and report back.

Thanks!

bwims, 2022-09-14:
Hi, 

sadly the adafruit stack does not help.  I have used this command:
`sudo stty -F /dev/ttyACM0 1200`
 to reset from the command line, so I don't know why it isn't working for your software.

Is there anything else we could try? 

Thanks!

[image]

bwims, 2022-09-14:
Oops!

I forgot to try a reset before this experiment.

It works!

Thanks so much!

Suggestion:   How about putting that important detail about the Adafruit TinyUSB in BIG block letters in the Readme?
Question  How do I remove the Pico-SDK option from the USB Stack, or at least prevent it being the default option?

Thanks again!

[image]

B.

earlephilhower, 2022-09-14:
Thanks for the update!

closed 2022-09-14 by earlephilhower

limitedAtonement, 2025-07-25:
I think I'm experiencing this problem.

> One possible thing, are you using the Adafruit USB stack?

I installed the arduino-ide, then 

1. Selected Adafruit TinyUSB for USB Stack
1. Unplugged the device
1. Held the button and plugged in the device
1. Released the button
1. Programmed the device (this succeeded)
1. Programmed the device again

The last step failed with "Scanning for RP2040 devices   No drive to deploy." I tried again using "Adafruit TinyUSB Host (native)" with the same problem.

By the way, when I `sudo journalctl --follow` then plug in the device, no logs show up at all! When I hold the button then plug it in, I get a lot of logs from adding the USB Mass Storage Device. I feel like there is no udev rule incantation that could fix this problem, right? It's like the device isn't running the USB code at all? My code is   `setup() {} loop(){}`.

earlephilhower, 2025-07-25:
For Adafruit USB, like mentioned above, you need to start a serial port or there's nothing to reset because the Pico won't actually enumerate any USB devices.

limitedAtonement, 2025-07-25:
Thank you very much! `Serial.begin(115200);` was indeed enough to get the ball rolling. Now, any time I plug in the device, it establishes the Serial over USB communication channel which allows the system to identify the device and set up the `/dev/ttyACM0` device file. I also had to add a udev rule so that I don't have to be root (or be in that `uucp` group):

```
SUBSYSTEM=="usb", ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="000a", MODE:="0666"
```

and now I can run

```shell
arduino-cli compile --upload --fqbn rp2040:rp2040:rpipico --port /dev/ttyACM0 sketch.ino;
```

over and over! Thanks again for your time and patience!