earlephilhower/arduino-pico #860: Pi PICO-W WifiServer example code failure

revell1, 2022-09-19:
I have been trying to resolve an issue with Wifi on PI PICO-W, originally posted over at:
github.com/khoih-prog/AsyncWebServer_RP2040W : Under Issue 3

It would be nice to resolve this issue, but can see that there is little to go on to identify either the failure mode or a solution, so am raising this issue in the hope that other users may be able to add to the available knowledge.

The history so far:
+++++++++++++++++++++

But have now concluded that the issue is not directly related to that Library, but must be an underlying issue with the
RP2040 PI Pico-W wifi functionality.

My hardware/software used:
Arduino IDE : 1.8.19
github: earlephilhower/arduino-pico library version : 2.5.2
Windows 10 Laptop, (Also later tested with old Linux Mint 19.2 Laptop)
PI PICO-W x3off (all have same problem)

Original issue reported to:
https://github.com/khoih-prog/AsyncWebServer_RP2040W/
Under ISSUE 3
As was having problems with that library.

Later switched to your AdvancedWebServer.ino (File-Examples-Examples For Raspberry Pi Pico-W-Wifo-WifiServer)

as this was nearly identical application code, just using your non-async server, hoping I could eliminate the RP2040
Arduino support as possible cause.

This also exhibited same lost response/connection issues.

Dug out 15 Year old laptop with Linux Mint 19.2 after updates and installing Arduino IDE (1.8.19) [installed first time with no issues unlike Windows10] built same code.

Compared uf2, map and a few other files against Windows 10, not quite the same, Windows u2f file
was 56 Bytes larger in Prog, and 4 bytes larger in Data, but probably not significant, so probably eliminates Windows 10 build system.

Then stepped back to Micropython (rp2-pico-w-20220914-unstable-v1.19.1-409-g0e8c2204d.uf2) and created a similar app to repeatedly request web page, but used AJAX and Java to request data packets, and also identical SVG file transfers on 3+4 second basis. [Note your example uses <meta http-equiv='refresh' content='5'/> to get frequent updates].
This Micropython version has been running for several hours with no drops or lost responses.

I am starting to think there must be an issue with the Arduino/RP2040 CPP support of the PI PICO,
though do not understand how Khoi Hoang has managed to get his setup to run for 38+ Hours using the same
code and target hardware.
The only difference have to be with real world WiFi and Router, but so far Micropython has shown no issues!

On one occasion with Arduino code running, and with Wireshark monitoring network managed to catch a failure.
Packet capture indicated that failure occured during SVG image file response. Response that was recieved consisted of
two packets, the first packet was the [200, "image/svg+xml"] response packet,
while the second packet was ONLY the first line of data from the code in the drawGraph() function:

  out += "<svg xmlns=\"http://www.w3.org/2000/svg\" version=\"1.1\" width=\"310\" height=\"150\">\n";
  out += "<rect width=\"310\" height=\"150\" fill=\"rgb(250, 230, 210)\" stroke-width=\"2\" stroke=\"rgb(0, 0, 0)\" />\n";
  out += "<g stroke=\"blue\">\n";
 
No other data was sent and no further responses for HTTP data could be obtained, but the device still responded to PING's,
though I am guessing that PING replies may well be handled internally buy the firmware in the CYW43439, also as I had modified
code to provide LED flash when Wifi requests were received by the code, the LED indicated that no further requests or responses were
being received or sent in the software.

Also the heartbeat serial(USB) port debug was still occuring, so the code had not crashed.

The fact that the start of the reply occured must indicate that drawGraph() must have completed, also as a minimum there
should have been the 3 blocks of data from the (out+=) part of the code, so how has only one been sent, is that an internal
failure between the RP2040 processor and the CYW43439 that somehow chunked the data at a "\n" character, or something
internal to the CYW43439.

I am at a loss to understand, or see a way forward. I am currently forced to use Arduino IDE and CPP due to lack of suitable Python library support for other hardware that I am intending to use such as MPU6050 (all Python Libraries are incomplete and do not use the MPU processor).

earlephilhower, 2022-09-21:
The issue you referenced had his async code running on this core for >1 day, and I've done my own "testing" (well, just starting it up and checking this afternoon) which also seems fine after 8hrs of runtime w/the non-async version here.

I think you may just have some interaction between the WiFi blob/chip here and your router's WiFi chipset.  While I understand your pain (ask me about the ESP8266 and random WiFi things!), there's really nothing we can do here, sorry.

closed 2022-09-21 by earlephilhower

revell1, 2022-09-22:
OK. Was suspecting something like that. Tends to make the PI PICO somewhat useless as a wifi device if it is so unstable with off the shelf routers (TP-LINK Archer VR600, Firmware 0.2.0 0.9.1 v0074.0 Build 170504 Rel.68210n [Latest]).

But I do not understand why code using Micropython (my own test code, so functionally different to the CPP example code, but tries to present a simple web page with live data) appear to run without Wifi issues. I assume that the FIRMWARE on the PI PICO-W wifi chip is different between the Micropython uf2 file and my CPP compiled program, otherwise my  issue would have to be with the application code, i.e. the example code which works for other users!!!

I also question possible Geographic location. I am in UK, but the example programs do not appear to care about location, so I would take a guess that all things default to USA as the initial location, are there any settings, function calls that really should be added to the test code to correctly configure WiFi parameters?

revell1, 2022-09-22:
I add a brief note here, in response to khoih-prog's reply.
I need I bit of time to digest that post you linked too.

But:

Today I have had two PI PICO's running two different MICROPYTHON programs, they have both been running for over 8 hours now, without issues.
Both are based on example code I found that use "microdot_asyncio". One is using the "5 second refresh" html meta method (like your example code), while the other is using an AJAX timer event request method.

I do see in my MICROPYTHON code that both programs I have the following startup:

=-===
    ssid = secrets['ssid']
    password = secrets['password']
    country('GB')
    wlan = network.WLAN(network.STA_IF) 
=-===

Note the specific selection of UK for the country.

The (your) example CPP code is using "WiFi.h" and "WiFiServer", but I have not seen any country selection.
Am I correct in assuming that the default is most likely going to be "USA".

Also what do I need to add to set the country code? As I think that would be a good next step, though I have also just updated the Arduino IDE to the latest 2.5.4 release of the RP2040 codebase, so probably should re-run my test code unchanged just to confirm the code still has problems.

=-========
As for the question about 2.4GHz crowding, it appears to be the other way, everyone is using 5GHz and there are few if any other routers in range that are using 2.4GHz, but a good point, though it should impact MICROPYTHON the same, but at present I see no problems with WIFI when using MICROPYTHON, so is it a country code missmatch?

=-======
As for MICROPYTHON can anyone explain if there is any common CPP code in the MICROPYTHON code image, as I am unclear what is underneath the micropython, is this built using the same RP2040 codebase of the Arduino IDE builds, or is it different?
I keep getting lost in github when trying to follow how the Micropython images are built.

khoih-prog, 2022-09-22:
Check [**latest answer**](https://github.com/khoih-prog/AsyncWebServer_RP2040W/issues/3#issuecomment-1255634912) about how to set country code to UK for RP2040W, and verify if it's any better in your case.

khoih-prog, 2022-09-23:
Please test the new [**AsyncWebServer_RP2040W** v1.0.3](https://github.com/khoih-prog/AsyncWebServer_RP2040W/releases/tag/v1.0.3) with way to modify and display `country-code` and verify if correct country-code can fix your issue.

---

### Release v1.0.3

1. Modify examples to display `country-code`
2. Add tempo method to modify `arduino-pico` core to change `country-code`
3. Add example [**Async_AdvancedWebServer_Country**](https://github.com/khoih-prog/AsyncWebServer_RP2040W/tree/main/examples/Async_AdvancedWebServer_Country)

revell1, 2022-09-25:
Hi, @earlephilhower , I started a thread over on raspberrypi.com, the following link,

Title: Pico-W web server stability issues when using CPP code VS Micropython
forums.raspberrypi.com/viewtopic.php?t=340529

I have now had confirmation from another UK user, who is also seeing the same problems with loss of access to Pico when using the RP2040 core example code : AdvancedWebServer.ino (File-Examples-Examples For Raspberry Pi Pico-W-Wifo-WifiServer).

So far, the only common points between my setup and the other is,
we are both in UK, and both our routers are using channel 11.
I don't think either points are significant, but it is the only common ground.

So there must be some issue that is being triggered, though what and how to isolate it is the big question.

Are there any other lighter libraries that could be used in place of WifiServer/WifiClient that would exercise the lower level driver and firmware code?
Possibly HTTP, and maybe even TCP type load testing, to see if the Wifi link is fully stable in other usage cases?

khoih-prog, 2022-09-25:
Hi @revell1 

Just released the [**AsyncWebServer_RP2040W** v1.1.0](https://github.com/khoih-prog/AsyncWebServer_RP2040W/releases/tag/v1.1.0). Be sure to use with the latest [**AsyncTCP_RP2040W** v1.1.0+](https://github.com/khoih-prog/AsyncTCP_RP2040W/releases/tag/v1.1.0).

I've been testing using **Firefox** (in **Ubuntu**,not Windows 10) for many hours and still OK.

[image]

Please test there to see if there still more issues.

---

### Release v1.1.0

1. Fix issue with slow browsers or network. Check [Target stops responding after variable time when using Firefox on Windows 10 #3](https://github.com/khoih-prog/AsyncWebServer_RP2040W/issues/3)