earlephilhower/arduino-pico #866: Startup time, time to first edge

xdxita, 2022-09-21:
Objective: 
Set GPIO2 logic level to 1 before 25ms after power up

Pure SDK behavior:
A simple `int main() {
    gpio_init(2);
    gpio_set_dir(2, GPIO_OUT);
    gpio_put(2,false);
}` takes about 20ms to toggle high GPIO2

Arduino behavior:
The same simple code takes around 100ms to toggle high GPIO2. I've tried to add the code to the framework main() function in cores/rp2040/main.cpp but I didn't noticed any difference.

Any ideas why it's taking this long and how to reduce this time, or at least how to toggle this pin in less then 25ms?

earlephilhower, 2022-09-21:
The SDK app has nothing else going on, while the Arduino core has all the global object constructors which run serially, followed by USB initialization.

Your best bet is to enable `TinyUSB` as the USB stack in the menus to effectively disable USB startup if you don't actually start a `Serial.begin()`.  OTW, you'll need to stick w/plain SDK code as there;s no way of specifying object initialization order (assuming you placed the function in question in the constructors link segment).

Good luck!

closed 2022-09-21 by earlephilhower

xdxita, 2022-09-21:
Ok, I'm already using the NO_USB define to disable USB at all.
I thought that the main() function in cores/rp2040/main.cpp held all Arduino core initialization, that's why I've placed the code there before everything for testing and got the same result.

Thanks!

earlephilhower, 2022-09-21:
It's part of the C++ runtime setup.  All global object constructors need to be run, one after the other, before `main()` is ever called.  There's also the copying from ROM to RAM for the `.data` segment and any in-RAM functions that's run, too, before that.

One other thing I forgot:  OTA.  On chip powerup it runs ROM, BOOT2, and then the OTA code (which simply exits since there is no work to do).  But that OTA code may take 10s of ms to start up due to the same things as the main Arduino code mentioned above.

xdxita, 2022-09-21:
Thanks, Earle!
I'll try to find another way and keep using the Arduino framework.
BTW, thanks for this port!
Cheers!

earlephilhower, 2022-09-21:
This is very hacky and not something I'd suggest as a long term solution, but why not move the GPIO write to the OTA main().  That might still be fast enough to meet your timing needs.  You'll need to have a working CMake build environment, but you can edit `ota/ota.c` and run the `make-ota.sh` script and then rebuild your Arduino app.  This will obviously break on every core update, but if you can find no other way it's possible.