earlephilhower/arduino-pico #867: Error using WiFi.h with TinyUSB

brentru, 2022-09-21:
@earlephilhower (cc @hathach)  I'm experiencing an error when using both `WiFi.h` and Adafruit TinyUSB together. I believe the error is due to WiFi.h bringing in `Updater.h`/`PicoOTA.h` and further brining in `LittleFS` within PicoOTA/Updater->`BearSSLHelpers.h`. 

Is there a way to use WiFi on the Pico without including these extra libraries (Updater/PicoOta/etc) that require LittleFS?

 
**Output of compiling the `msc_external_flash.ino` with additional `#include "LittleFS.h"`  at the top of the sketch:**
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/brentrubell/Library/Arduino15/packages -hardware /Users/brentrubell/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/brentrubell/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/brentrubell/Documents/Arduino/libraries -fqbn=rp2040:rp2040:rpipicow:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Enabled,dbgport=Disabled,dbglvl=None,usbstack=tinyusb,ipstack=ipv4only -ide-version=10819 -build-path /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_832771 -warnings=none -build-cache /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_cache_305629 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-gcc.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-elf2uf2/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-elf2uf2/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-pioasm/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-pioasm/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-openocd/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-openocd/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-python3.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-mklittlefs/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-mklittlefs-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-mklittlefs/1.4.0-c-0196c06 -verbose /Users/brentrubell/Documents/Arduino/msc_external_flash/msc_external_flash.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/brentrubell/Library/Arduino15/packages -hardware /Users/brentrubell/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/brentrubell/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/brentrubell/Documents/Arduino/libraries -fqbn=rp2040:rp2040:rpipicow:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Enabled,dbgport=Disabled,dbglvl=None,usbstack=tinyusb,ipstack=ipv4only -ide-version=10819 -build-path /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_832771 -warnings=none -build-cache /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_cache_305629 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-gcc.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-elf2uf2/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-elf2uf2/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-pioasm/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-pioasm/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-openocd/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-openocd/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-python3.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-mklittlefs/1.4.0-c-0196c06 -prefs=runtime.tools.pqt-mklittlefs-1.4.0-c-0196c06.path=/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-mklittlefs/1.4.0-c-0196c06 -verbose /Users/brentrubell/Documents/Arduino/msc_external_flash/msc_external_flash.ino
Using board 'rpipicow' from platform in folder: /Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2
Using core 'rp2040' from platform in folder: /Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2
Detecting libraries used...
/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -I /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_832771/core -c -Werror=return-type -DUSE_TINYUSB -I/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/libraries/Adafruit_TinyUSB_Arduino/src/arduino -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0xf00a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipicow\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fexceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefix/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/ @/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/lib/platform_inc.txt -I/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DSERIALUSB_PID=0xf00a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10819 -DARDUINO_RASPBERRY_PI_PICO_W "-DBOARD_NAME=\"RASPBERRY_PI_PICO_W\"" -DARDUINO_ARCH_RP2040 -Os -I/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/cores/rp2040 -I/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/variants/rpipicow /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_832771/sketch/msc_external_flash.ino.cpp -o /dev/null
Alternatives for LittleFS.h: [LittleFS@0.1.0]
ResolveLibrary(LittleFS.h)
  -> candidates: [LittleFS@0.1.0]
/Users/brentrubell/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -I /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_832771/core -c -Werror=return-type -DUSE_TINYUSB -I/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/libraries/Adafruit_TinyUSB_Arduino/src/arduino -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0xf00a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipicow\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-se [... cut ...]
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper_Networking.h:37,
                 from /Users/brentrubell/Documents/Arduino/Wippersnapper_demo/Wippersnapper_demo.ino:16:
/Users/brentrubell/Library/Arduino15/packages/rp2040/hardware/rp2040/2.5.2/libraries/WiFi/src/BearSSLHelpers.h:27:10: fatal error: Updater.h: No such file or directory
   27 | #include <Updater.h>
      |          ^~~~~~~~~~~
compilation terminated.
```

Full Error output is attached:
[full_output.txt](https://github.com/earlephilhower/arduino-pico/files/9620554/full_output.txt)

earlephilhower, 2022-09-21:
````
Using library SdFat_-_Adafruit_Fork at version 1.5.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork 
Using library Adafruit_TinyUSB_Library at version 1.14.3 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library 
````
The Adafruit SDFat library is not compatible with this core (`File` is redefined as shown in your logs`.

You also should use the `Adafruit_TinyUSB` library that's distributed with this core, not another version.

brentru, 2022-09-21:
@earlephilhower  Is there a version of `SDFat` which is compatible with both this library and the `Adafruit_TinyUSB` library bundled within this core? `SDFS`?

earlephilhower, 2022-09-21:
There's not much we can do here, sorry. 

The problem is that ESP8266SDFat is v2.x (maybe 3+ yrs old now?) and taken from the upstream original SdFat library by grieman.  The TinyUSb example depends on a 1.x internal API which is no longer there, and the Adafruit SDFat library is a fork of that 1.x version that's not been updated to 2.x.

Your only option is to port the example from the Adafruit TinyUSB to the 2.x version of the library, or rewrite the SD sector access routines to mimic the older API.

closed 2022-09-21 by earlephilhower

hathach, 2022-09-22:
@brentru we have a plan to upgrade SD Fork to v2 version to match latest SdFat upstream. Will do it when possible.

brentru, 2022-09-26:
@hathach  Thank you!

hathach, 2022-10-17:
@brentru both SPIFlash and SdFat (adafruit's fork) are upgraded to support v2. It is now possible to compile LFS with tinyusb. Please try it out. Note File32 should be used to replace File when using SdFat v2 (which is the main file type in v2).

brentru, 2022-10-18:
@hathach  Will try it out later today, thank you for the work!

brentru, 2022-11-03:
@hathach Sorry for delay, was working on Wippersnapper features.

The MQTT example I included above now compiles, thank you for your work on updating spiflash+sdfat!
[image]