earlephilhower/arduino-pico #868: cyw43_arch.c:79: undefined reference to `cyw43_tcpip_link_status'

lu7did, 2022-09-21:
When trying to compile the program ntp-client of pico examples with the Arduino IDE and after few adaptations to allow for that (comment stdio_init_all() and a couple of castings as well as introduce setup() and loop()) the compilation fails at linking time with the following message

...test\ntp_client/cyw43_arch.c:79: undefined reference to `cyw43_tcpip_link_status'

The prototype for that function is at cyw43.h but including it won't solve the problem
The function itself is at cyw43_lwip.c but it should be in the library path, including it explicitly at the program folder won't solve the problem and actually starts to generate an explosion of dependency issues.

Having search for a similar problem elsewhere but to no avail. It seems a similar issue might prevent the usage of any example on the pico_w path.

I've been working successfully with the IDE, including adaptations of several other examples on the pico-examples path (specially dma, multicore and PIO).

¿Any help or suggestion?

Thanks and best regards, Pedro.

earlephilhower, 2022-09-21:
I think you're going to need to post an MCVE to get any help here.  The core included examples are built as part of  CI, of course.  Don't attempt to access the WiFi stuff directly through the CYWxxx calls, that's not going to work due to LWIP integrations.

The function call is defined in the `lwIP_CYW43` library instantiated as part of the WiFi class.

lu7did, 2022-09-21:
Hi, thanks for the answer. I'm not sure what a MCVE is or where to post it.

Meanwhile I'm just trying to compile the ntp_client example on the pico_w directory of the pico examples, to the best extend of my understanding the calls issues are:

    cyw43_arch_lwip_begin();
..
    udp_sendto(state->ntp_pcb, p, &state->ntp_server_address, NTP_PORT);
..
    cyw43_arch_lwip_end();

Actually I'm trying to compile the example  at pico_w/ntp_client/picow_ntp_client.c

as I did with other examples (successfully) I removed the stdio_init_all() and split the one time into a setup() proc and the cycling part as loop() to comply with what the IDE expect.

Also needed to strong cast

NTP_T *state = calloc(1, sizeof(NTP_T)); into NTP_T *state = (NTP_T *)calloc(1, sizeof(NTP_T)); for it to compile.

Not sure what else I can try at this point.

Appreciate any help you can provide.

Regards, Pedro.

earlephilhower, 2022-09-22:
That's not going to work with the core. Changes were needed to support all the additional functionality of the WiFi and assorted libs.  The LWIP stack and the CYW driver overrides are very different from the plain SDK to the core.  Also, those SDK examples are .C, not .CPP, so you'll run into type issues as you saw (but that's orthogonal and most of the cast issues are not real problems).

For NTP, that's already implemented.  See the docs for the call.

closed 2022-09-22 by earlephilhower