earlephilhower/arduino-pico #870: Missing library `ArduinoCore-API` => compile error for new core v2.5.3

khoih-prog, 2022-09-22:
The `ArduinoCore-API` was not automatically installed with the new core v2.5.3, leading to compile error when using even the simplest code

```
void setup() 
{
}

void loop() 
{
}
```
---

```
In file included from /home/kh/.arduino15/packages/rp2040/hardware/rp2040/2.5.3/cores/rp2040/Arduino.h:28,
                 from sketch/Client.ino.cpp:1:
/home/kh/.arduino15/packages/rp2040/hardware/rp2040/2.5.3/cores/rp2040/api/ArduinoAPI.h:2:10: fatal error: ../../../ArduinoCore-API/api/ArduinoAPI.h: No such file or directory
    2 | #include "../../../ArduinoCore-API/api/ArduinoAPI.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Raspberry Pi Pico W.
```

Reason

File [**ArduinoAPI.h**](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/api/ArduinoAPI.h) has been modified in v2.5.3 to

```
#pragma once
#include "../../../ArduinoCore-API/api/ArduinoAPI.h"
```

---

The tempo fix is to install manually the [**ArduinoCore-API**](https://github.com/earlephilhower/ArduinoCore-API) to 

`.packages\rp2040\hardware\rp2040\2.5.3`

earlephilhower, 2022-09-22:
Oops, sorry bout that.  2.5.4 just pushed and the ZIP does have the directory included.

closed 2022-09-22 by earlephilhower

khoih-prog, 2022-09-22:
v2.5.4 fixes the v2.5.3 issue. OK now.