earlephilhower/arduino-pico #873: Feature request please. multiple virtual com ports with Pico stack.

Captainslarty, 2022-09-22:
Hello and good day,
I wonder if you would consider adding CDC VCP - virtual com ports - over USB, similar to what is possible with the Adafruit TinyUSB stack ?
It would be really good to have the ability natively.
As in being able to add VCPs USBSer1, USBSer2.
It is excellent to be able to run one per core and one for debug.
Regards,
Joe

earlephilhower, 2022-09-22:
This is unfortunately not possible using the SDK and the way it's built here.  The # of serial ports is a `#define` in a header that's used to build the usb library, and it can't be changed after the fact in userland.  If you define it to be 2, the USB code will *always* expose 2 CDC ports which would be a mess in the general case.

IIRC, in fact, you need to hack the Adafruit_tinyusb `tusb_config.h` file to make the dual CDC example work, so it's quite a challenge even if you compile your own USB stack,

closed 2022-09-22 by earlephilhower

Captainslarty, 2022-09-23:
Hello Earle and thank you for the reply,
This is what I am aware of so far - with tusb and multi vcp,With tusb, the file tusb_config_rp2040.h needs editing as below (either 2 or 3)#define CFG_TUD_CDC 3       // was 1 allows use of USBSer1 and USBSer2After the edit and save then the only com ports exposed are ones that are specifically declared as in Adafruit_USBD_CDC USBSer1; 
// and  in setup ...  USBSer1.begin(115200);
Serial is as normal.
If you don't specifically add the Adafruit_USBD_CDC USBSerx; and the USBSerx.begin() then they are not created and don't appear in device manager or the IDE so editing the  #define CFG_TUD_CDC 3 still needs the ports initialising but, yes, slightly more resources are used.It's a shame it can't be more controllable because it is extremely useful and I use it a lot.Best regards and thank you for the amazing work.Joe

s-light, 2025-07-16:
just for users that come across this:
i found that at least in my `tusb_config_rp2040.h` file it was already set to 2.
(maybe i could not remember that i already did this long time ago.. or it is so default today..)