earlephilhower/arduino-pico #874: [Feature Request] Permit modifying WiFi country-code for RP2040W

khoih-prog, 2022-09-23:
It's great if we can provide a way for users to change WiFi `country-code`, if necessary. Check [**Pi PICO-W WifiServer example code failure** #860](https://github.com/earlephilhower/arduino-pico/issues/860#issuecomment-1255470103) for possible reason to use correct `country-code` as MICROPYTHON can do.

Currently, I can see a tempo way to modify is to change [**picow_init.cpp**](https://github.com/earlephilhower/arduino-pico/blob/master/variants/rpipicow/picow_init.cpp) of `RP2040W` to, e.g `CYW43_COUNTRY_UK`

```
#include <pico/cyw43_arch.h>

extern "C" void initVariant() {
  //cyw43_arch_init();
  // Select country code, if necessary
  // For example: CYW43_COUNTRY_AUSTRALIA, CYW43_COUNTRY_CANADA, CYW43_COUNTRY_CHINA, CYW43_COUNTRY_FRANCE, CYW43_COUNTRY_GERMANY, 
  // CYW43_COUNTRY_INDIA, CYW43_COUNTRY_ITALY, CYW43_COUNTRY_JAPAN, CYW43_COUNTRY_NETHERLANDS, CYW43_COUNTRY_SOUTH_KOREA, 
  // CYW43_COUNTRY_SWEDEN, CYW43_COUNTRY_UK, CYW43_COUNTRY_USA, etc.
  cyw43_arch_init_with_country(CYW43_COUNTRY_UK);
}
```

from

```
#include <pico/cyw43_arch.h>

extern "C" void initVariant() {
    cyw43_arch_init();
}
```

---

I'm confident @earlephilhower will know the better way to let users to select, without manually changing the core's file. The best, IMO, is to select via the board selection /  [boards.txt](https://github.com/earlephilhower/arduino-pico/blob/master/boards.txt), similar to Flash Size, CPU Speed, etc.

earlephilhower, 2022-09-23:
Simple enough to do as a menu item.  Would be possible to do at runtime but requires add'l work to make the CYW43 initialization lazy (i.e. store any overridden WIFI regions and only call cyw3_init() when doing WiFi.begin/beginAP or a write to the LED pin) so that's not supported for now.

closed 2022-09-23 by earlephilhower

khoih-prog, 2022-09-23:
Great. You're amazingly quick and efficient. Have been learning quite a lot from your works.

khoih-prog, 2022-09-23:
Working great in menu and compiling.

[image]

However, I can't upload using either Arduino IDE v1.8.16 or v1.8.19. The error is it's trying to use non-existing `2.5.4/system/python3/python3`

```
Sketch uses 347132 bytes (22%) of program storage space. Maximum is 1568768 bytes.
Global variables use 68876 bytes (26%) of dynamic memory, leaving 193268 bytes for local variables. Maximum is 262144 bytes.
/home/kh/.arduino15/packages/rp2040/hardware/rp2040/2.5.4/system/python3/python3 -I /home/kh/.arduino15/packages/rp2040/hardware/rp2040/2.5.4/tools/uf2conv.py --serial /dev/ttyACM0 --family RP2040 --deploy /tmp/arduino_build_309113/Async_AdvancedWebServer_Country.ino.uf2 
java.io.IOException: Cannot run program "/home/kh/.arduino15/packages/rp2040/hardware/rp2040/2.5.4/system/python3/python3": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

The problem is created by these new lines

 https://github.com/earlephilhower/arduino-pico/blob/7a85c3917f9e931202c38d5cb5292f69af7abef0/platform.txt#L25-L29

earlephilhower, 2022-09-23:
Not sure what your setup is, but you can't just extract the changes here onto a Board Manager installation, you need to use a GIT copy.  The lines you have listed are needed when using GIT copies, and are removed as part of the packaging process.
https://github.com/earlephilhower/arduino-pico/blob/7a85c3917f9e931202c38d5cb5292f69af7abef0/package/build_boards_manager_package.sh#L104-L118

khoih-prog, 2022-09-23:
Sorry, I followed the sequence according to [**Installing via GIT**](https://github.com/earlephilhower/arduino-pico#installing-via-git)

```
mkdir -p ~/Arduino/hardware/pico
git clone https://github.com/earlephilhower/arduino-pico.git ~/Arduino/hardware/pico/rp2040
cd ~/Arduino/hardware/pico/rp2040
git submodule update --init
cd pico-sdk
git submodule update --init
cd ../tools
python3 ./get.py
```

but the `platform.txt` still contains

```
runtime.tools.pqt-gcc.path={runtime.platform.path}/system/arm-none-eabi
runtime.tools.pqt-python3.path={runtime.platform.path}/system/python3
runtime.tools.pqt-mklittlefs.path={runtime.platform.path}/system/mklittlefs
runtime.tools.pqt-pioasm.path={runtime.platform.path}/system/pioasm
runtime.tools.pqt-elf2uf2.path={runtime.platform.path}/system/elf2uf2
```

earlephilhower, 2022-09-23:
They're suppose to be there in the GIT version.  The get.py will DL and install the toolchain in ~/Arduino/hardware/pico/rp2040/system and those lines will reference it.  This can't be done in the ~/.arduino15/... board manager dirs.

Also if you have 2 copies of the repo, make sure you select the one that says `(from sketchbook)` (or similar...don't remember exact wording).

PR #875 (2022-09-23): Allow setting the WiFi region for PicoW
Fixes #874
commit 7a85c391 (2022-09-23): Allow setting the WiFi region for PicoW (#875)

Fixes #874