earlephilhower/arduino-pico #878: Using FreeRTOS attachInterrupt() does hang the core.

beegee-tokyo, 2022-09-25:
I am using FreeRTOS (used to it from ESP32 and nRF52), I can use **`attachInterrupt()`** there without any problems.

With this BSP I have a strange problem that I do not understand. As soon as I add **`#include <FreeRTOS.h>`**, the core that calls attachInterrupt() hangs.

Example IRQ code that works:
```cpp
volatile uint32_t CNT;

void setup()
{
	Serial.begin(115200);
	while (!Serial){};
	Serial.println("Attach IRQ");
	attachInterrupt(digitalPinToInterrupt(27), ISR_func, RISING);
	// or attachInterrupt(27, ISR_func, RISING);
	Serial.println("Attached IRQ");
}

void ISR_func()
{
	CNT++;
}

void loop()
{
	Serial.println(CNT);
	delay(500);
}
```

Adding only the **`#include <FreeRTOS.h>`** in the first line and the core that calls **`attachInterrupt()`** hangs.
```cpp
#include <FreeRTOS.h>

volatile uint32_t CNT;

void setup()
{
	Serial.begin(115200);
	while (!Serial){};
	Serial.println("Attach IRQ");
	attachInterrupt(digitalPinToInterrupt(27), ISR_func, RISING);
	// or attachInterrupt(27, ISR_func, RISING);
	Serial.println("Attached IRQ");
}

void ISR_func()
{
	CNT++;
}

void loop()
{
	Serial.println(CNT);
	delay(500);
}
```

Not sure what I am doing wrong.

beegee-tokyo, 2022-09-25:
Little digging and if I remove the **`detachInterrupt(pin)`** from **`attachInterrupt()`** function it seems to work.
```cpp
extern "C" void attachInterrupt(pin_size_t pin, voidFuncPtr callback, PinStatus mode) {
    CoreMutex m(&_irqMutex);
    if (!m) {
        return;
    }

    uint32_t events;
    switch (mode) {
    case LOW:     events = 1; break;
    case HIGH:    events = 2; break;
    case FALLING: events = 4; break;
    case RISING:  events = 8; break;
    case CHANGE:  events = 4 | 8; break;
    default:      return;  // ERROR
    }
    noInterrupts();
    // detachInterrupt(pin); // <<<<<<<<<<<<<<<
    CBInfo cb(callback);
    _map.insert({pin, cb});
    gpio_set_irq_enabled_with_callback(pin, events, true, _gpioInterruptDispatcher);
    interrupts();
}
```

earlephilhower, 2022-09-25:
Thanks for the detailed MCVE and debug!  This was a simple deadlock while in FreeRTOS mode (mutexes work differently there).  Your MCVE works fine with the PR #879 .

closed 2022-09-25 by earlephilhower

beegee-tokyo, 2022-09-26:
@earlephilhower 
Glad I could help.
Great work for the RP2040. I love it!

imwoo90, 2022-11-03:
attachInterruptPram() should have the same fix.

```c++
void attachInterruptParam(pin_size_t pin, voidFuncPtrParam callback, PinStatus mode, void *param) {
    CoreMutex m(&_irqMutex);
    if (!m) {
        return;
    }

    uint32_t events;
    switch (mode) {
    case LOW:     events = 1; break;
    case HIGH:    events = 2; break;
    case FALLING: events = 4; break;
    case RISING:  events = 8; break;
    case CHANGE:  events = 4 | 8; break;
    default:      return;  // ERROR
    }
    noInterrupts();
    detachInterrupt(pin);
    CBInfo cb(callback, param);
    _map.insert({pin, cb});
    gpio_set_irq_enabled_with_callback(pin, events, true, _gpioInterruptDispatcher);
    interrupts();
}
```

imwoo90, 2022-11-04:
I think the CoreMutex does't work recursive mutex on freeRTOS
and this code does not use the FromISR serise in ISR

deadlock occur when fooB() called
i change to recursive mutex serise and it work fine
```
mutex

fooA() {
 CoreMutex m(&mutex)
}

fooB() {
 CoreMutex m(&mutex)
 fooA()
}
```

This code work in ISR but the freeRTOS mutex is not FromISR serise
I modified the related code a little and when calling the attachInterrupt function, it works as an existing code routine, and in isr, I tested it to be called with the FromISR series, and it was confirmed that it works well.
```c
void _gpioInterruptDispatcher(uint gpio, uint32_t events) {
    (void) events;
    // Only need to lock around the std::map check, not the whole IRQ callback
    CoreMutex m(&_irqMutex);
    if (m) {
        auto irq = _map.find(gpio);
        if (irq != _map.end()) {
            auto cb = irq->second;
            cb.callback();
        }
    }
}
```

PR #879 (2022-09-25): Fix deadlock during attachInterrupt
Fixes #878
commit 939c8312 (2022-09-25): Fix deadlock during attachInterrupt (#879)

Fixes #878