earlephilhower/arduino-pico #884: If WiFi.softAPConfig executed after WiFi.begin, DNS not works, and softAP ip is not correct.

MrGreensWorkshop, 2022-09-28:
Hi, First of all, Thank you for this project.

When I run WiFi.softAPConfig after running WiFi.begin, DNS not works, and softAP ip is not correct.
If you need more information or want me to debug please let me know.

use case 1: Application tries to connect wifi, but it couldn't find proper wifi, so starts config manager.
  - With non existing access point credentials. (WIFI_SSID, WIFI_PASS)
result: IP is not set correctly. sometimes when trying to connect softap from android, couldn't get an ip address.

use case 2: Application tries to connect wifi and success, User starts config manager for setting change.
  - With existing access point credentials. (WIFI_SSID, WIFI_PASS)
result: IP is not set correctly. sometimes when trying to connect softap from android, couldn't get an ip address.

This is the code I test with Raspberry Pi Pico W
 
```C++

// based on 
//https://github.com/earlephilhower/arduino-pico/blob/master/libraries/DNSServer/examples/DNSServer/DNSServer.ino
#include <WiFi.h>
#include <DNSServer.h>
#include <WebServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(172, 217, 28, 1);
DNSServer dnsServer;
WebServer webServer(80);

const char* WIFI_SSID = "xxxxxxx";
const char* WIFI_PASS = "12345678";  

bool serverOn = false;
bool WifiConnected = false;
int WifiDisconTime = 10; //5sec.

bool wifiSetup(const char* ssid, const char* pass) {
  bool connected = false;
  int tryCount = 10;
  // Set WIFI module to STA mode
  WiFi.mode(WIFI_STA);

  // Connect
  Serial.printf("[WIFI] Connecting to %s ", ssid);
  WiFi.begin(ssid, pass);

  // Wait for wifi connection
  while (1) {
    //check wifi connection
    static unsigned long last = millis();
    if (millis() - last > 500) {
      last = millis();
      if (WiFi.status() == WL_CONNECTED) {
        // Connected!
          Serial.printf("\n[WIFI] STATION Mode, SSID: %s, IP address: %s\n", \
                          WiFi.SSID(), WiFi.localIP().toString().c_str());
        connected = true;
        break;
      }
      Serial.print(".");
      tryCount--;
    }
    
    if (tryCount <= 0) break;
    delay(100);
  }
  Serial.println("");
  return connected;
}

void setup() {
  Serial.begin(115200);
  Serial.println("");
  delay(1000);

  // to disable soft AP after restart
  WiFi.mode(WIFI_OFF);
  delay(100);

  WifiConnected = wifiSetup(WIFI_SSID, WIFI_PASS);
  if (!WifiConnected) setServer();
}

void setServer(){
  WiFi.disconnect();
  WiFi.end();
  WifiConnected = false;

  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("picow", "12345678");
  delay(500);
  Serial.print("AP IP address: ");
  Serial.println(WiFi.softAPIP());

  // modify TTL associated  with the domain name (in seconds)
  // default is 60 seconds
  dnsServer.setTTL(300);
  // set which return code will be used for all other domains (e.g. sending
  // ServerFailure instead of NonExistentDomain will reduce number of queries
  // sent by clients)
  // default is DNSReplyCode::NonExistentDomain
  dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);

  // start DNS server for a specific domain name
  dnsServer.start(DNS_PORT, "*", apIP);

  // simple HTTP server to see that DNS server is working
  webServer.onNotFound([]() {
    String message = "Hello World!\n\n";
    message += "URI: ";
    message += webServer.uri();

    webServer.send(200, "text/plain", message);
  });
  webServer.begin();

  serverOn = true;
  Serial.println("Server setup done");
}

void loop() {
  if (WifiConnected) {
    static unsigned long last = millis();
    if (millis() - last > 500) {
      last = millis();
      Serial.print(".");
      WifiDisconTime--;
    }
    if (WifiDisconTime <= 0) {
      setServer();
      WifiConnected = false;
    }
  }
  delay(100);
  if (serverOn){
    dnsServer.processNextRequest();
    webServer.handleClient();
  }
}
```

MrGreensWorkshop, 2022-09-29:
After further test, I realize that, DNS works if the ip address is set to 192.168.42.1. 
But still setting IP address in softAPConfig not works If WiFi.softAPConfig called after WiFi.begin.

khoih-prog, 2022-09-29:
AFAIK, the current version of `cyw43 WiFi` library has some limitations

1. AP IP is fixed at 192.168.42.1
2. AP WiFi Channel is fixed to 10. Can't be changed by `beginAP(ssid, pass, channel)`

MrGreensWorkshop, 2022-09-29:
Thank you for you insights. What I'm saying is, its not impossible to change it. If you fist start WiFi.softAP yes you can change the IP. But if you call WiFi.begin then WiFi.softAP then it doesn't work. That's why I open this issue.

earlephilhower, 2022-10-06:
You have an interesting (and useful) use case.  This look to be a complicated debug, so it's been thrown behind the IDE 2.0 stuff.  Now that's out of the way I'll give this a go...

I'm thinking a lot of this might be to some leftover sanity checks (i.e. since you started WiFi.begin already, the code might not clear a flag somewhere and ignore your 2nd calls to WiFi.beginAP.)

FWIW, the AP mode *should* use the `WiFi.softAPconfig()` options, but there might me some assumptions or call ordering issues.  The IP should not need to fixed to 192.168.1.42, but that is what will happen if things are not set properly in the the sketch or the WiFiclass backend.

MrGreensWorkshop, 2022-10-07:
Thank you! I appreciate your time and effort.

earlephilhower, 2022-10-09:
@MrGreensWorkshop can you give the PR a try?

MrGreensWorkshop, 2022-10-09:
I tested and It it working. It is working on both cases. Thank you so much.

PR #905 (2022-10-10): Clear LWIP started flag on LWIPIntfDev::end
When moving between different modes or even WiFi.begin/ends, any setting of the IPs will fail because the internal flag _started was not cleared. Clear _started on a ::end call.

Fixes #884
commit 25ceb08f (2022-10-10): Clear LWIP started flag on LWIPIntfDev::end (#905)

When moving between different modes or even WiFi.begin/ends, any setting
of the IPs will fail because the internal flag _started was not cleared.
Clear _started on a ::end call.

Fixes #884
closed 2022-10-09 by MrGreensWorkshop