earlephilhower/arduino-pico #888: frequency glitch ?

nirbkm, 2022-10-03:
Hi, 
I tried some code i found online running simple task to check speed & accuracy while using the Arduino-pico library,
After 1 hour of running the loop (forgot it was running while watching Netflix :P ) i noticed sudden change in the behaviour,
from ~1,197,050 loops per second, the value went down to ~822,990 loops per second.

This behaviour repeat 3/3 times, and each time after approximately 1 hour running.

Could you please advise?
Hopefully it's something that i did wrong and not a bug in your library.

`
int Power = 11;
long lastMillis = 0;
long loops = 0;

void setup() {
  pinMode(Power, OUTPUT);
  digitalWrite(Power, HIGH);
  Serial.begin(57600);
  unsigned int currentMillis = millis();
  while (true) {
    currentMillis = millis();
    loops++;
    if (currentMillis - lastMillis >= 1000) {
      Serial.print("ms:");
      Serial.print(currentMillis - lastMillis);
      Serial.print(" loops: ");
      Serial.print(loops);
      Serial.print(" CPU_FREQ: ");
      Serial.print(rp2040.f_cpu());
      Serial.print(" millis: ");
      Serial.println(millis());

      

      lastMillis = currentMillis;
      loops = 0;
    }
  }
}

void loop(void) {
}
`

and the output:
11:04:49.456 -> ms:1000 loops: 1197047 CPU_FREQ: 133000000 millis: 4291000
11:04:50.456 -> ms:1000 loops: 1197057 CPU_FREQ: 133000000 millis: 4292000
11:04:51.446 -> ms:1000 loops: 1197052 CPU_FREQ: 133000000 millis: 4293000
11:04:52.436 -> ms:1000 loops: 1197056 CPU_FREQ: 133000000 millis: 4294000
11:04:53.456 -> ms:1000 loops: 1184809 CPU_FREQ: 133000000 millis: 4295000
11:04:54.454 -> ms:1000 loops: 822973 CPU_FREQ: 133000000 millis: 4296000   ## sudden change in loop count
11:04:55.446 -> ms:1000 loops: 822974 CPU_FREQ: 133000000 millis: 4297000
11:04:56.436 -> ms:1000 loops: 822973 CPU_FREQ: 133000000 millis: 4298000
11:04:57.456 -> ms:1000 loops: 822975 CPU_FREQ: 133000000 millis: 4299000
11:04:58.456 -> ms:1000 loops: 822971 CPU_FREQ: 133000000 millis: 4300000
11:04:59.446 -> ms:1000 loops: 822972 CPU_FREQ: 133000000 millis: 4301000
11:05:00.436 -> ms:1000 loops: 822975 CPU_FREQ: 133000000 millis: 4302000
11:05:01.456 -> ms:1000 loops: 822973 CPU_FREQ: 133000000 millis: 4303000
11:05:02.461 -> ms:1000 loops: 822975 CPU_FREQ: 133000000 millis: 4304000

Please advise,
Thank you,
Nir.

revell1, 2022-10-03:
Could it be that you laptop slowed the USB port while watching Netflix, so slowed the serial.print() commands?
When in your log did you start watching Netflix?
Also where you highlight, the previous log line appears to show something starting to happen, as the "loops:" value has reduced a bit already.

nirbkm, 2022-10-03:
> Could it be that you laptop slowed the USB port while watching Netflix, so slowed the serial.print() commands? When in your log did you start watching Netflix? Also where you highlight, the previous [...]

Actually i was watching Netflix on TV,  so the PC only ran the mentioned code.
But anyway, i ran the same code 3 times, and the exact thing happen again in all 3 times including the single row that is in between the two states.

revell1, 2022-10-03:
OK, next thought, has the PC got a screen saver that kicked in at the same time each of the three tests, assuming you started the test then walked away, leaving PC to enter screen save state after period of inactivity.

nirbkm, 2022-10-03:
Thanks for the fast feedback,
no screen savers or any other unwanted activity..
it just switches to the slower state and stays like that, only restarting the pico will retrieve it to the initial state until the next run.

If there is nothing wrong with the code maybe someone with a pico can upload the same code trying to reproduce this behaviour.

earlephilhower, 2022-10-03:
@revell1 is correct, thqanks.  The USB is interrupt driven and runs code on the core to communicate.  If it has to do work or wait for an attached computer, your main loop will be a little slower.

If you want fully stable performance, you need to disconnect the USB completely and use a UART serial port.  OTW, there is nothing in particular running on the core other than user code (i.e. no OS).

closed 2022-10-03 by earlephilhower

nirbkm, 2022-10-04:
Hi, 
Thanks for the feedback,
I tried powering up the pico using VSYS pin with external 5V PSU,
writing the same lines from the pico to a UNO board using UART0, 
from there the UNO sent the lines to the PC with USB serial..

Unfortunately i got the same behaviour..
It's not meant to use in any specific project right now, just a thing i ran into and thought it will be interesting to understand.

Thank you,