earlephilhower/arduino-pico #889: ILabs Challenger 2040 WiFi

ITstreet1, 2022-10-03:
While trying to compile a simple Scan example from the lib, I get this error:
C:\Users\:usr:\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.3.2\libraries\lwIP_Ethernet\src/LwipIntfDev.h:59:29: error: 'SS' was not declared in this scope
   59 |     LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
      |                             ^~
exit status 1

This trows for any WiFi example from the lib

revell1, 2022-10-03:
Looking at your path directory, it looks like you are using an old release of 2.3.2. If so, try the latest and see if that makes any difference.

ITstreet1, 2022-10-03:
Updated lib

C:\Users\usr\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.5.4\libraries\lwIP_Ethernet\src/LwipIntfDev.h:61:29: error: 'SS' was not declared in this scope
   61 |     LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
      |                             ^~

JAndrassy, 2022-10-03:
lwIP_Ethernet should work with a W5500 on any RP2040, but I am not sure if the library is ready

ITstreet1, 2022-10-03:
> 

I just try to verify. Can someone try to verify any sketch from board examples about WiFi?

PontusO, 2022-10-03:
> Hi Pontus,
> Yes, I have read Getting Started on your pages.
> Still, examples from the package for this board do not work. And I am
> addressing the examples from the package. If examples are not compatible
> then they should be fixed or removed.
> Some other examples, such as Fade, forks fine. :)Are there any other
> examples for Arduino IDE?
> CircuitPython works just fine.—

Dejan, you should be able to run any example from the wifiespat library 
package, we've tested all examples without any issues.

Having the wifi library (for the rpi w) integrate into the board support 
package automatically makes it available for all rp2040 boards. That is 
just an Arduino thing.

JAndrassy, 2022-10-03:
> @JAndrassy
> 
> I just try to verify. Can someone try to verify any sketch from board examples about WiFi?

the WiFi library bundled with the boards package is only for Raspberry's Pico W board with a CYW43439 WiFi module on board

ITstreet1, 2022-10-03:
@JAndrassy 
This one has ESP8285 but I get the point
@PontusO 

This clears a lot. Thank you, guys!

JAndrassy, 2022-10-03:
off topic: @PontusO I think for the ESP8285 AT 1.7.5 would be a better option than AT2. AT 2 for esp82xx stopped at 2.2.0 and has many problems

ITstreet1, 2022-10-03:
Just in case others stumble upon this:

```
#include <ChallengerWiFi.h>
#include <WiFiEspAT.h>
void setup() {
  Serial.begin(115200);
  while (!Serial2);
  Serial2.begin(115200);
  WiFi.init(Serial2);
  WiFi.begin("ssid", "pass");
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();
  Serial.println("Connected to WiFi network.");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

This is a basic sketch for a WiFi connection. Works like a charm.

earlephilhower, 2022-10-03:
Looks like this was cleared up, thanks guys.  If the Challenger WiFi board needs a different `pins_arduino.h` please do submit a PR.

closed 2022-10-03 by earlephilhower