earlephilhower/arduino-pico #890: First Upload: Port not detected

Phoenix1747, 2022-10-03:
According to [this section of the README](https://github.com/earlephilhower/arduino-pico#uploading-sketches) the following should work:
> To upload your first sketch, you will need to hold the BOOTSEL button down while plugging in the Pico to your computer. Then hit the upload button and the sketch should be transferred and start to run.

In the past this always worked for me either by pressing the BOOTSEL button when plugging in or as a default for new Pico boards. It seems that with the new Arduino IDE 2.0 this doesn't work anymore -- at least for me. I tried several times today with brand-new boards and existing ones and they always pop up as a drive in Windows like they should. However, nothing will detect the ports from the inside of the Arduino IDE. Tried hitting `UPLOAD` multiple times and it just fails giving `Upload error: Failed uploading: no upload port provided` with still no ports showing.

Is this just a problem with my setup here or is anybody able to replicate this behaviour? Don't know how to program new Pico boards now, without looking at all the command line options.

earlephilhower, 2022-10-03:
Would you be able to try editing one line in boards.txt?  Change the line
````
rpipico.upload.wait_for_upload_port=true
````
to
````
rpipico.upload.wait_for_upload_port=false
````

If that works, I can change the repo and do a new release for 2.0 compat...

Phoenix1747, 2022-10-03:
Doesn't work, unfortunately. Still returns `Failed uploading: no upload port provided` and does nothing 😕

earlephilhower, 2022-10-04:
To get you going, the easiest thing is to take the built UF2 (the path is shown in the IDE) and manually copy it to the drive.  After that, a serial port will be in the IDE and it should be possible to upload normally.

In the meantime, I'll see what can be done here.  There's a lot of rough edges on the 2.0 IDE (i.e. no plugin support whatsoever) still...

Phoenix1747, 2022-10-04:
> To get you going, the easiest thing is to take the built UF2 (the path is shown in the IDE) and manually copy it to the drive. After that, a serial port will be in the IDE and it should be possible to upload normally.

Thank you, I was already looking at how to get the UF2 files manually, but enabling verbose output in the IDE showed me the path!

> There's a lot of rough edges on the 2.0 IDE (i.e. no plugin support whatsoever) still...

True, I'm also getting update notifications for libraries that are already up-to-date and stuff like that. Also think that the board config options under `Tools` like `Flash Size` or `CPU Speed` are much more unclear now. But hey, it's the version that new users will be downloading from now on so it is what it is, hope there will be some updates in the coming weeks...

earlephilhower, 2022-10-04:
It turns out that I had some bit rot in my UF2 detector script.  I've fixed it and now get a UF2 port on a completely new install of the IDE (after deleting `~/arduino*`).  That can be selected and it'll do the normal upload and you're set.  Works in 1.8.9 and 2.0 now.

closed 2022-10-04 by earlephilhower

Phoenix1747, 2022-10-05:
Awesome!

PR #897 (2022-10-04): Support IDE2, detect UF2 volumes
Allow the IDE to detect UF2 volumes (i.e. when you hold BOOTDEL and plug in the board).

Allows the IDE2 to properly upload using OTA and serial.

Fixes #890 and others
commit b249811e (2022-10-04): Support IDE2, detect UF2 volumes (#897)

Allow the IDE to detect UF2 volumes (i.e. when you hold BOOTDEL and
plug in the board).

Allows the IDE2 to properly upload using OTA and serial.

Fixes #890 and others