earlephilhower/arduino-pico #917: Increase frequency limit of AnalogWrite() ?

fjansson, 2022-10-15:
Would it be possible to increase the upper frequency limit of AnalogWrite()?

Here
https://github.com/earlephilhower/arduino-pico/blob/69ab736cf8fbbba7bf2345accd7a5b03699eefaa/cores/rp2040/wiring_analog.cpp#L45
the frequency is capped at 1 MHz. I think there's no reason for the limit except that the PWM resolution becomes low at high frequencies. 

I want to use AnalogWrite as an easy way to set up PWM as a pulse generator, and I'd like it to reach a few MHz at least. For a project for just myself I would use the Pico SDK's pwm functions, but this time I want to publish the code and keep it as simple as possible.

earlephilhower, 2022-10-15:
Can you up the frequency limit on that line to 8MHz or 16MHz and give it a test and report back/give a PR?

I don't mind increasing the limit, but I am a little worried about some of the math later over/underflowing and winding up with division by 0s or something causing it to fail when actually setting the parameters.

fjansson, 2022-10-15:
Testing that, I discovered two problems in analogWrite() as it currently is:
* `pwm_config_set_wrap(&c, analogScale);` should be `pwm_config_set_wrap(&c, analogScale-1);`
The wrap value is not the counter period but the highest counter value before wrapping. Adding '-1' here gives better frequency accuracy, and also makes the output stay high all the time when outputting the full-scale value.

* `(float)(analogScale * analogFreq)` can overflow, since the multiplication is done in uint32. Then the frequency set is wrong.
`((float)analogScale * analogFreq)` works better.

I'll try to make a pull-request soon.

Btw, I found that analogWrite to one pin changes the PWM settings for *all* PWM blocks, which I didn't expect. I wanted to use another PWM block as a frequency counter for testing, and found I had to set up that after the analogWrite.

earlephilhower, 2022-10-16:
Closed via #918

closed 2022-10-16 by earlephilhower