earlephilhower/arduino-pico #919: PWM glitching/optimization

earlephilhower, 2022-10-16:
IMO, this section is a little bit extreme, and disturbing the PWM operation of remaining pins

https://github.com/earlephilhower/arduino-pico/blob/10090b60a9458f4a495e88f8df989401eded7c02/cores/rp2040/wiring_analog.cpp#L100-L102

It's better to just use

```cpp
pwm_init(pwm_gpio_to_slice_num(pin), &c, true);
```

Moreover, to avoid glitch when changing, it's better to use

```cpp
pwm_set_wrap(pwm_gpio_to_slice_num(pin), analogScale);
```

instead of

```cpp
pwm_config_set_wrap(&c, analogScale);
```

Check [**Changing Duty Cycle Dynamically Creates Runt PWM pulse** #10](https://github.com/khoih-prog/RP2040_PWM/issues/10) for more info.

_Originally posted by @khoih-prog in https://github.com/earlephilhower/arduino-pico/issues/918#issuecomment-1280019684_

earlephilhower, 2022-10-16:
(moved to a new issue for tracking)

re: the for(all GPIO) loop:  This is only run on when the PWM is not initted, so basically once when the PWM config is set, no?  I'm not sure how it could affect other running pins since at that time nothing else would be running...

I was unaware of the glitching, so thanks for the notice.  I think that's a no-brainer change.

khoih-prog, 2022-10-16:
That's called every time you run `analogWrite(pin_size_t pin, int val)`

https://github.com/earlephilhower/arduino-pico/blob/10090b60a9458f4a495e88f8df989401eded7c02/cores/rp2040/wiring_analog.cpp#L74

The case is that 

- You've run `analogWrite()` or use PWM  on pin x 
- Now you call `analogWrite()` on pin y
- The `for` loop, which set the new data for every pin, will destroy all previous PWM settings

khoih-prog, 2022-10-16:
The code in mbed_rp2040 just change for the related pin

https://github.com/arduino/ArduinoCore-mbed/blob/e9dc61751360aaa11f95c011e790e1db305abfc0/cores/arduino/wiring_analog.cpp#L69-L81

```cpp
  float percent = (float)val/(float)((1 << write_resolution)-1);
  mbed::PwmOut* pwm = digitalPinToPwm(pin);
  ...
  if (percent < 0) {
    delete pwm;
    digitalPinToPwm(pin) = NULL;
  } else {
    pwm->write(percent);
  }
```

khoih-prog, 2022-10-16:
This is simple code to demo the issue

```cpp
#define _PWM_LOGLEVEL_        3
#include "RP2040_PWM.h"

//creates pwm instance
RP2040_PWM* PWM_Instance;

float frequency = 100;
float dutyCycle;

#define pinToUse            25
#define pinAnalogWrite      10

void setup()
{
  pinMode(pinToUse, OUTPUT);
  
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  delay(200);

  Serial.print(F("\nStarting analogWrite_PWM on ")); Serial.println(BOARD_NAME);
  Serial.println(RP2040_PWM_VERSION);
  
  //assigns pin 25 (built in LED), with frequency of 100 Hz and a duty cycle of 0%
  PWM_Instance = new RP2040_PWM(pinToUse, frequency, 0);
}

void loop()
{
  frequency = 20000;
  dutyCycle = 10;

  PWM_Instance->setPWM(pinToUse, frequency, dutyCycle);

  delay(2000);
  dutyCycle = 90;
  
  PWM_Instance->setPWM(pinToUse, frequency, dutyCycle);

  delay(2000);

  // After this analogWrite, the PWM of other pins will stop
  // without it, PWM still running OK
  Serial.println(F("analogWrite will stop PWM"));
  analogWrite(pinAnalogWrite, 0);

  while (true)  { }
}
```

fjansson, 2022-10-18:
I like the setup where only requested pins are initialized. Perhaps add to the documentation that
the values set by analogFrequency() and analogRange() only affect subsequent analogWrite() calls.
Pins already doing PWM keep their old frequency and range.

earlephilhower, 2022-10-20:
> ...Pins already doing PWM keep their old frequency and range.

Not always.  The PWM HW has 8 configurable slices and each one drives 2 pins.  I'd have to dig up the datasheet to give mappings, but this means that in general any one change could affect at least one more pin.  That was the reason I initially just set every slice up identically.  This change would allow you the possibility of setting different configs, but you as the user need to be cognizant of how any pin updates would affect other pins running in your app.

PR #926 (2022-10-20): Only init PWM on requested pin
Avoid glitches on other pins, fix #919
commit 4fdcc6e5 (2022-10-20): Only init PWM on requested pin (#926)

Avoid glitches on other pins, fix #919
closed 2022-10-20 by earlephilhower