earlephilhower/arduino-pico #927: Pimoroni PGA2040 wont accept Arduino-Core

Dans-Droids, 2022-10-18:
Hi, 
Thank you for making such a great resource! 

I am having issues getting the core to properly install on a PGA2040, 
Plugging in the PGA2040 as I connect BS to GND, Ardunio IDE uploads the initial sketch fine, reboots the PGA2040 but then the board is not seen by windows or the IDE, unplugging and plugging back in (to expect the IDE to see a port and windows to play "Device Connected Sound") but nothing happens. 
I have to repeat the BOOTSEL - GND connection and replug the board, taking me back to square one of uploading the initial sketch.
I have tried numerous sketches including the RP2040 examples and just a blank one but none seem to install the core after the initial upload..

I have been using the Generic RP2040 board from the board-list, with the 2MB Flash setting, although the board has 8MB..

Would it be worth trying to make a dedicated Variant for the PGA2040?

Thank you for your help
Kind Regards

maxgerhardt, 2022-10-18:
Have you iterated through all possible Boot2 possibilities at flash size = 8MByte? 

If I look at https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/pimoroni_pga2040/mpconfigboard.mk, it has W25Q64JVxQ as its flash chip. In circuit-python it goes through a [gen_stage2.py](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/gen_stage2.py) script that pulls info from https://github.com/adafruit/nvm.toml/blob/2d292ad4e67890d4b85b027431ba9fef7bf561fd/flash/winbond/W25Q64JVxQ.toml, but I'm not exactly sure what comes out of it then.

maxgerhardt, 2022-10-18:
Another quick sanity check is that, if you copy-paste the circuitpython .uf2 file on it, does it still do nothing afterwards? https://circuitpython.org/board/pimoroni_pga2040/

Dans-Droids, 2022-10-18:
> Another quick sanity check is that, if you copy-paste the circuitpython .uf2 file on its boot drive, does it still do nothing afterwards? https://circuitpython.org/board/pimoroni_pga2040/

Hey, thanks for the replies!
Yes, I have tried copying the uf2 file to the board and it does the same thing, plays the windows "Device Disconnected Sound" but never seems to properly reboot the PGA2040 into a usable state..

Dans-Droids, 2022-10-18:
> I managed to compile circuitpython for the PGA2040 and obtain the used boot2 source: [boot2_w25q64j_padded_checksummed.zip](https://github.com/earlephilhower/arduino-pico/files/9812193/boot2_w25q64j_padded_checksummed.zip)
> 
> Let me compile a test Arduino firmware for you to see if it helps..

Interesting, I am able to use Thony to reflash microPython and it works fine on that so i might just stick with microPython for my project..

maxgerhardt, 2022-10-18:
I've custom-built a firmware firmware for you with the boot2 source for the flash chip. Can copy-paste the .uf2 file on your board and check if it works @Dans-Droids?

[firmware.zip](https://github.com/earlephilhower/arduino-pico/files/9813984/firmware.zip)

Source behind it:
```cpp
#include <Arduino.h>

#define LED_PIN 3

void setup() {
  Serial.begin(115200);
  pinMode(LED_PIN, OUTPUT);
}
int i = 0;
void loop() {
  Serial.println("Inside loop(), iteration: " + String(i++));
  digitalWrite(LED_PIN, HIGH);
  delay(500);
  digitalWrite(LED_PIN, LOW);
  delay(500);  
}
```

Dans-Droids, 2022-10-18:
Hey, so the firmware flashes correctly and is seen by windows as PicoArduno. However upon modifying the source sketch to the following: 
`
#define LED_PIN 3

void setup() {
  Serial.begin(115200);
  pinMode(LED_PIN, OUTPUT);
}
int i = 0;
void loop() {
  Serial.println("Inside loop(), iteration: " + String(i++));
  digitalWrite(LED_PIN, HIGH);
  Serial.println("LED HIGH");
  delay(500);
  digitalWrite(LED_PIN, LOW);
  Serial.println("LED LOW");
  delay(500);  
}`
to add some verbal indication of the LED state change, the sketch uploads fine and reboots the PGA2040 but upon reboot the same issue where it is no recognised or seen at all.. unplugging and replugging has no efffect - it is still not seen, i have flashed the firmware and uploaded the sketch twice with the same result...

earlephilhower, 2022-10-18:
I would suggest trying the generic configuration and just go with "Generic SPI / 4" which is the most compatible 1-bit, unidirectional, plain SPI interface running at 1/4 clock rate.  If that's not running then the issue is not the flash boot2.S but something else,

maxgerhardt, 2022-10-18:
>Hey, so the firmware flashes correctly and is seen by windows as PicoArduno

That was the main point I wanted to check, without the proper modifications to the `platform.txt` and the additional boot2.S file, you won't be able to produce your own sketches *yet*. I will file a PR to enable the usage of this boot2 file.

(If you wanted a 'dirty hack', you could overwrite your local installation's  e.g. `boot2_generic_03h_4_padded_checksum.S` file with the `.S` file I've zipped above, then choose 8MB flash and the Generic SPI /4 flash configuration, and it will use your moded content)

maxgerhardt, 2022-10-18:
Okay, poster only said
>Yes, I have tried all the flash size variations available 
But not all *Boot2* options.

@Dans-Droids can you explicitly confirm that selecting Boot2 -> Generic SPI /4  at 8MByte flash also *does not work*?

earlephilhower, 2022-10-18:
> But doesn't the fact that with the modified boot2.S that I compiled the firmware with works, and all other boot2 selections do not, confirm that it's a boot2 issue?

If we are 1000% sure that the generic SPI / 4 doesn't work, but the modified boot2 works then yes.

But, logically, that setting is as generic and compatible as possible and it would mean the attached SPI flash doesn't really speak SPI at all...whick would make me wonder how the boot2 would be read out of it at all by the ROM.

Having an add'l, customized boot2 w/QSPI/etc. would be better, of course.  It should also wind up upstream in the Pico SDK at some point (I'd say that's Pimoroni's job, not anyone here's)...

Dans-Droids, 2022-10-18:
> 

thank you for going to all this trouble!
I have just amended the IDE settings as per Earl's advise, changing to "Generic SPI/4" instead of "SPI/2" with 8MB flash and reflashed the firmware & uploaded the sketch again, the board rebooted and was seen again by windows as "Raspberry Pi Pico W" which made me laugh! It changed the COM port it was on but same physical USB connection? and is now sending the updated sketch back in the Serial Monitor

maxgerhardt, 2022-10-18:
Kinda sad it was an easy fix with only a wrong selection now.

Will PR new boot2 for reference anyways -- can't hurt to have it there I guess.

>was seen again by windows as "Raspberry Pi Pico W" which made me laugh

Yes, the core sets the USB product + manufacturer name, when I compiled the test firmware with a custom PlatformIO board definition, I gave it the "Pimoroni PG2040" name, but since you compilew ith the generic board definition, that is actually correct.

Dans-Droids, 2022-10-18:
Thank you so much for your help!

closed 2022-10-20 by earlephilhower

maxgerhardt, 2022-10-21:
For completeness sake: I created https://github.com/maxgerhardt/pico-boot2-generator that has pre-generated boot2 files for all flash chips known in https://github.com/adafruit/nvm.toml/, building it the same way CircuitPython does. With CI :)

PR #929 (2022-10-20): Add W25Q64JV QSPI /4 Boot2 file
A simple addition. Expands only the "Generic" variant by a "W25Q64JV QSPI /4" Boot2 option.

I was very tempted to also just add
```py
# Pimoroni
MakeBoard("pimoroni_pga2040", "Pimoroni", "PGA2040", "0x2e8a", "0x1008", 250, "PIMORONI_PGA2040", 8, "boot2_w25q64jv_4_padded_checksum")
```
but the `pimoroni_pga2040` would have just been a copy of the `generic` variant folder -- felt weird.

Resolves #927 more cleanly by giving them the possibility to use their flash chip on Quad-SPI mode ([clock div 4](https://github.com/adafruit/circuitpython/blob/main/ports/raspbe [... cut]
commit b82336c8 (2022-10-20): Add W25Q64JV QSPI /4 Boot2 file (#929)

Fixes #927