earlephilhower/arduino-pico #928: Keyboard Garbled Output/Race Conditions

brxken128, 2022-10-18:
I've noticed an issue that occurs on multiple Pico boards, and all seem to produce the same issue. I have tried the official Pi Pico, and an Adafruit Trinkey QT2040.

While using `Keyboard.println()`, and while even printing manually, the text will become mangled at some point until the device is unplugged and re-plugged into the PC.

Here's an example sample of output:

```
hello world hello world hello world
hello world hello world hello world
hello world hello world hello world
hello world hello world hello world
hello world hello world hlelo world
hello world hello wrold hello wrold
hello world hello world hello world
hello world helol world hello world
hlelo world hello world ehllo world
```

I have tried adding delays with code such as this (to no avail):

```cpp
for(int i = 0; i < text.length(); i++) {
  Keyboard.write(text[i]);
  delay(2);
}
Keyboard.write(0xB0);
```

I've also tried setting the delay between keypresses to something absurdly high (50ms), and I've tried calling `Keyboard.end()` and then `Keyboard.begin()` before the next entry. Nothing seems to help.

I assumed it was a race condition at first, but now I'm not sure. I recently bought some more boards, and tested them, and the issue persists (the Adafruit Trinkey QT2040). It seems to be even worse for some reason - re-plugging the device does not restore functionality (nor does re-flashing, which is a little scary). This happens to both boards.

I ensure that I select the correct board within the Arduino IDE before uploading, and I have tried the standard Pi Pico board too.

I have also tried loading the password from a file stored within a LittleFS partition (to try and single out memory issues), and this had no effect.

My main use of this Pico is for password entry, and here is a sketch that I know does not work. Just click bootsel repeatedly and eventually the sequence of `abcde` will break.
```cpp
#include <Keyboard.h>
#include "OneButton.h"

const String pw1 = "abcdeabcdeabcde"; // password for single click

OneButton *button;

void enterPw(String password) {
  Keyboard.println(password);
}

void singleClick() {
  enterPw(pw1);
}

void setup() {
  Keyboard.begin();
  button = new OneButton();
  button->attachClick(singleClick);
  delay(1000);
}

void loop() {
  button->tick(BOOTSEL);
}
```

earlephilhower, 2022-10-20:
I have never seen the `OneButton` library.  Do you have a failing complete sketch without that dependency?

FWIW, all the RP2040 boards are basically identical in relations to the USB....just a couple passives (maybe not even that) connected to pins on the RP2040 chip from RPI, so any behavior would be expected to happen no matter what board.

brxken128, 2022-10-20:
The sketch below also produces the same results on all Pico boards I've tested it on. It seems when the text is over a certain length, the output text is almost guaranteed to be misplaced. Even on smaller text it happens, although not as often.

Sometimes it will print normally (usually the first couple of iterations after flashing), but more often than not it fails.

It's worth noting that viewing what was supposed to be typed over serial shows the correct text, it just always seems to differ from what was actually typed.

```cpp
#include <Keyboard.h>

const String password1 = "abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc";

void enterPw(String password) {
  Keyboard.println(password);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(150);
  digitalWrite(LED_BUILTIN, LOW);
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Keyboard.begin();
  delay(1000);
}

void loop() {
  if (BOOTSEL) {
    enterPw(password1);
  }
}
```

earlephilhower, 2022-10-20:
Thanks for the update, I'll take a look at it.

earlephilhower, 2022-10-22:
I'm unable to reproduce any failure on my system with your example.  What OS/version/CPU/etc. are you running?  I'm under Ubuntu 22.04 (close to latest kernel) w/a Ryzen 5600X and I always am getting the proper output....
````
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
abcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabcabc
````

Looking at the code I can see something that might be fishy, but w/o being able to reproduce a failure it's shooting in the dark...

brxken128, 2022-10-22:
I'm not too sure what's going on here, it could very well just be something specific to my setup. I'm running a Ryzen 9 5900x on Fedora 37 with Linux 5.19.14.

All boards I tested exhibited the same results, and I tried a few cables with each just to be sure.

I apologise for this, I assumed it would be reproducible as I couldn't find a way to make it work at all.

I'll close this for now and update if I find anything new, and I'll test on a different machine too in order to see if it's just my main workstation.

Thank you so much for the help anyway, and apologies for the hassle :)

closed 2022-10-22 by brxken128

earlephilhower, 2022-10-22:
No worries.  I also just tried on a Win10 21H2 laptop w/8-th gen Intel chip under Notepad++ and Word and didn't see any issues.  Any progress, update here or open a new issue.

brxken128, 2022-10-23:
> No worries. I also just tried on a Win10 21H2 laptop w/8-th gen Intel chip under Notepad++ and Word and didn't see any issues. Any progress, update here or open a new issue.

Good afternoon, it appears as I have the answer the answer - it's specific to my machine. I tested this on my laptop and it works flawlessly. I wrote and tested same kind of tool in Rust for the RP2040, and again it worked great on my laptop but not at all on my PC.

No worries if not, but do you have any pointers for diagnosing this issue? I'm going to see if it's reproducible on Windows too, but I'm out of ideas aside from that. It happens on USB ports both directly from the motherboard/connected to the front panel.

earlephilhower, 2022-10-23:
It's our was dropping keys, I could understand it. But reordering keys is plain weird.

There are tools to capture USB packets under Linux and Windows.  You could try them out and see what the difference is on the wire, if any, between working and non working computers.