earlephilhower/arduino-pico #931: HttpClient struggles to load big payloads

lapfelix, 2022-10-23:
I'm not talking huge payloads. It will often stop loading after 1000 or 10,000 characters (or earlier).

I made a simple arduino program to reproduce the issue:
```
#include <WiFi.h>
#include <HTTPClient.h>

WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);

  // put your setup code here, to run once:
  WiFiMulti.addAP("FElix", "Austin-Didier Tran");
}

void loop() {
  if (WiFiMulti.run() != WL_CONNECTED) {
    //Not connected to Wi-Fi yet
    delay(1000);
    printf("Connecting to Wi-Fi\n");
    return;
  }

  HTTPClient http;

  Serial.print("[HTTP] begin...\n");
  if (http.begin("http://felix.lapal.me/payloadtest.txt")) {
      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0)
      {
          // HTTP header has been send and Server response header has been handled
          Serial.printf("[HTTP] GET... code: %d\n", httpCode);

          // file found at server
          if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY)
          {
              auto payload = http.getString();
              auto cPayload = payload.c_str();
              if (strlen(cPayload) > 4 && !strcmp(cPayload + strlen(cPayload) - 4, "#end"))
              {
                Serial.printf("successfully read the payload\n");
              }
              else {
                Serial.printf("bad payload\n");
              }
              Serial.printf("payload ends with %s\n",cPayload + strlen(cPayload) - 10);
          }
      }
      else
      {
          Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
  }
  else
  {
      Serial.print("[HTTP} Unable to connect\n");
  }

  delay(2000);
}
```

This program loads the payload at http://felix.lapal.me/payloadtest.txt which is a string that contains numbers that indicate the string length where they begin (so the `1` in `1080` is the 1081th character).

Running the program on a Raspberry Pico W, I get this output:
```
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 5-1080-108
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 876-10882-
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 13678-1368
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 876-10882-
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 876-10882-
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 5640-#end

[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
bad payload
payload ends with 876-10882-
```

So it stops around 1080 once, then 10882, then 13678, then back at 10882 twice, it succeeds (that's ~15640 characters!) then fails again around 10882.

Am I doing something wrong? Is that an issue with the Pico's Wi-Fi module or something in `arduino-pico` that can be fixed?

earlephilhower, 2022-10-23:
Looks like your app is doing a `String` result and a 1-shot read which is really awful in terms of memory usage.  Even though the total size may be small, there are lots of memory re-allocations involved as the string is generated leaving it susceptible to things like OOM because of not being able find a contiguous memory space to fit things.

For large payloads, rewrite to use something like the `HTTPUpdate` which easily pulls down ~1MB size payloads for update.  You'll need to use the `Stream` pointer and be smart about allocation and/or storing to temporary filesystems.

lapfelix, 2022-10-23:
Oh okay thanks a lot! I'll look into it!

closed 2022-10-23 by lapfelix