earlephilhower/arduino-pico #932: Error during Upload: Property 'upload.tool.serial' is undefined

stiliajohny, 2022-10-23:
Hi

I am using `arduino-cli` and I have installed all the libraries needed by the projects etc 

when I am issuing the command 
`arduino-cli upload --config-file ./arduino_config.yaml --port $(PORT) -b rp2040:rp2040:rpipico $(FILE)` I am getting the project compiled but I cant upload it due to the the following error 
```log
Sketch uses 52148 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 7328 bytes (2%) of dynamic memory, leaving 254816 bytes for local variables. Maximum is 262144 bytes.

Used library  Version Path
Rtc by Makuna 2.3.5   /Users/johnstilia/Documents/GitHub/hardware-gps-tracker/hardware/gps_unit/.arduino/user/libraries/Rtc_by_Makuna

Used platform Version Path
rp2040:rp2040 2.6.1   /Users/johnstilia/Documents/GitHub/hardware-gps-tracker/hardware/gps_unit/.arduino/data/packages/rp2040/hardware/rp2040/2.6.1

Error during Upload: Property 'upload.tool.serial' is undefined
make: *** [code-upload] Error 1  
```

That didn't seem to be the case on the 2.5.0 version. 

It could be something different too. 

Thank you in advance for any help you can provide

earlephilhower, 2022-10-23:
What version of the Arduino packagers are you using?  And what version of the core?

The 2.0 IDE introduced a different, kind-of incompatible `platform.txt` way of specifying uploaders.  I've tested the IDE 2.x and 1.x IDEs w/o issue, and I think they both use the `arduino-cli` behind the scenes to make things work.

per1234, 2022-10-23:
Hi all. I have identified the problem and am preparing a pull request right now. I will provide a full explanation at that time.

earlephilhower, 2022-10-23:
Great, thanks @per1234 !

closed 2022-10-24 by earlephilhower