earlephilhower/arduino-pico #935: Publishing framework-arduinopico on platformio

jesserockz, 2022-10-27:
Thanks for all the work you have done here.

Are you planning to resume publishing the releases to https://registry.platformio.org/tools/earlephilhower/framework-arduinopico ?

Cheers,
Jesse 
@ESPHome

jesserockz, 2022-10-27:
Ah interesting that platformio are not accepting the releases and publishing them.

ESPHome always defaults to a specific version of the framework, but allows the users to change the version, for which I build either a package version (from platformio) or now, a url to the github release which is working fine.

Thanks,
Jesse

earlephilhower, 2022-10-28:
Good news, it looks like the latest one did make it in.
````
A new version of the package framework-arduinopico (1.20602.0) was published at 2022-10-28 12:56:08 from 20.114.114.178.
The SHA-256 of this package was 872a474bccd9ee7f31ca4d901e0dd797522849aaeb5fe00138c85dd507493ea8.
````

maxgerhardt, 2022-10-31:
The *submitting* of the package to the PlatformIO registry is automatic in this repo, however, every package publish still has to be approved of manually by PlatformIO staff. Hence delays.