earlephilhower/arduino-pico #939: Add different modes for I2S Audio Output

DatanoiseTV, 2022-10-28:
The modified audio library (for example at https://github.com/DatanoiseTV/RP2040-DSP-FreeRTOS-Template/tree/main/lib/audio) can provide 24-bit and 32-bit Audio Output at 44100, 48000Hz etc and is therefore more flexible than the stock Pico Audio Lib. But the most important change is, that it has a weak callback function in which it processes a number of samples:

```
.....
audio_buffer_pool_t *ap;
.....
void i2s_callback_func()
    {
        audio_buffer_t *buffer = take_audio_buffer(ap, false);
        if (buffer == NULL)
        {
            return;
        }
        int32_t *samples = (int32_t *)buffer->buffer->bytes;

        dsp_start = to_us_since_boot(get_absolute_time());
        for (uint i = 0; i < buffer->max_sample_count; i++)
        {
            int32_t smp = Dsp_process(ctx, 0) << 16u;
            samples[i * 2 + 0] = smp; // LEFT
            samples[i * 2 + 1] = smp; // RIGHT
        }
        dsp_end = to_us_since_boot(get_absolute_time());

        buffer->sample_count = buffer->max_sample_count;
        give_audio_buffer(ap, buffer);
        return;
    }

```

Would you consider doing the same for the arduino-pico? I would love to see it to also provide Arduino support
for my upcoming PicoADK+ and also happy to send you a board.

https://github.com/DatanoiseTV/PicoDSP-Hardware#news-as-of-20-10-2022---introducing-the-picoadk

[image]

earlephilhower, 2022-10-28:
Neat board.  I'm sure the USB-C alone will win many fans!

It looks like you're using modified `pico-extra` I2S code.  The core here isn't using anything from that so there's not going to be a 1:1 mapping of calls, but I think you can get there today with even the "standard" Arduino I2S calls.

There's already support for the callback through the `I2S::onTransmit/onReceive` calls.  You get a CB on a DMA buffer interrupt (i.e. at the end of chunk).  At that point, you can do what you're doing here (but need your own buffer ) and do an `I2S::write(buff, buflen)` to stuff it in the next available ring slot.

earlephilhower, 2022-11-06:
Closing with no feedback in 9 days.

closed 2022-11-06 by earlephilhower

DatanoiseTV, 2022-11-07:
@earlephilhower Thanks! Let me know if you want one, then I can ship the new version incl. a 12-bit 8x ADC with 1M samples/s.
What would be your suggested implementation for porting my Pico-SDK code (currently using FreeRTOS) to arduino-pico?
https://github.com/DatanoiseTV/RP2040-DSP-FreeRTOS-Template/blob/main/src/main.cpp#L186

DatanoiseTV, 2022-11-19:
Could you update the example in the I2S library? 16 bit output without tx callback work so far, 24 and 32-bit not yet. Would be great to have a simple callback example.

earlephilhower, 2022-11-23:
Thanks for the offer, that would be nice!  I now manually swap between up a Google AIY microphone board and whatever DAC I can dig out of the junk drawer using clone DuPont connectors and it's quite a pain.  If you ping my email (at the bottom of the readme) could you drop me a mail and I can reply w/the shipping info?

As for 24- and 32-bit, they should "just work."  The I2S core doesn't really care as long as it is told the right # of bits to shift out before toggling LRCLK.  The thing to be aware of is that because I2S is fixed point with the binary point at bit 32, everything should be left-aligned ( i.e. 24bit data is 0x??????00).  I haven't done any testing w/a real DAC, but I did run some traces using a logic analyzer and Sigrok on Linux and it seemed OK.  Maybe a regression slipped in.  What exactly are you seeing?

As for the code, select an appropriate buffer size and register an `onReceive` callback for the I2S block.  When you get that (from IRQ context so no mallocs, as normal) you should be safe to stuff your buffer's size of data w/o any blocking.   The `I2s::write(uint8_t *buff, size_t len)`  should do it, but be aware you need to make sure the data is properly formatted there (i.e. 24-bit is really 32-bit with LSByte of 0x00, 16-bit has 2 packed samples per 32-bits, etc.).

DatanoiseTV, 2022-11-23:
@earlephilhower I am happy to send you a [PicoADK](https://github.com/DatanoiseTV/PicoADK-Hardware) if that helps debugging.

DatanoiseTV, 2022-11-23:
> If you ping my email (at the bottom of the readme) could you drop me a mail and I can reply w/the shipping info?

Done! :)

DatanoiseTV, 2022-12-01:
@earlephilhower Your PCB should arrive hopefully soon. I've been rather unsuccessful with the tryout.
The code crashes and just makes noise.

Code is at: https://github.com/DatanoiseOrg/VultArduino

```cpp
#include <I2S.h>

// Create the I2S port using a PIO state machine
I2S i2s(OUTPUT);

// GPIO pin numbers
#define pMute 25
#define pDemp 23

#include "vult.h"

Dsp_process_type ctx;

#define NUM_SAMPLES 16
int32_t *audioBuffer[NUM_SAMPLES];

void __not_in_flash_func(i2s_callback)(void) {
  for (uint8_t i = 0; i < NUM_SAMPLES; i++) {
    int32_t sample = Dsp_process(ctx, 0) << 8u;
    i2s.write(sample, 0); // L
    i2s.write(sample, 0); // R
  
}

void setup()
{
  Serial1.begin(115200);

  i2s.setBitsPerSample(32);
  i2s.setBuffers(16, 2, 0);
  if (!i2s.begin(44100)) {
    Serial1.println("Failed to initialize I2S!");
    while (1); // do nothing
  }

  pinMode(pMute, OUTPUT);
  digitalWrite(pMute, HIGH); // unmute codec

  pinMode(pDemp, OUTPUT);
  digitalWrite(pDemp, LOW);

  Dsp_process_init(ctx);
  Dsp_default_init(ctx);
  Dsp_default(ctx);

  i2s.onTransmit(i2s_callback);
}

void loop()
{
  /*
  static unsigned long lastRun = 0;

  unsigned long now = millis();
  if (now - lastRun >= 1000) {
    uint8_t randomNote = (rand() % 12) + 60;
    Serial.println("Note on.");
    Dsp_noteOn(ctx, randomNote, 127, 1);
    lastRun = now;
  }
  */
  Serial1.println("foo");
}```

DatanoiseTV, 2022-12-01:
> Thanks! I got distracted during our holidays here and completely forgot. Much appreciated.
> 
> For simplicity, can you look at the I2S sample https://github.com/earlephilhower/arduino-pico/blob/master/libraries/I2S/examples/SimpleTone/SimpleTone.ino and verify that works @ 16-bits (may need to check the I2S frequency to make sure it's legal).
> 
> If that's good, then I think you can just bump things to 24-bit with `int32_t sample`, `setBitsPerSample(24)`, and `sample = 0x40000000` and the `I2S.write(sample)` should just "work" to send 24-bits between LRCLK phases.
> 
> I don't know what's in the `DSP_xxx` calls, but remember you're in an IRQ for the callback so no `malloc()`, no long-running ops, and make sure you're not overflowing the stack (since 24bpp = 2x the storage of 16bpp).

Dsp_process() returns a signed 32-bit integer: https://github.com/vult-dsp/vult/blob/master/runtime/vultin.h#L44
The processing for a polyphonic synth currently takes 700uS for 64 samples.

earlephilhower, 2022-12-19:
Just received the board, very nice!  I'll solder up some pins and see what I can see on it.

DatanoiseTV, 2022-12-20:
> Just received the board, very nice! I'll solder up some pins and see what I can see on it.

Great. Check out these resources to get started quickly:
https://github.com/DatanoiseTV/PicoADK-Hardware
https://github.com/DatanoiseTV/PicoADK-Firmware-Template
Interactive Pinout: https://datanoise.net/picoadk/

Let me know how it goes. Also, I recommend joining the discord: https://discord.gg/BsHUEdStMt

earlephilhower, 2022-12-20:
Insomnia got me so I just tried a simple test at 32bit/44khz and I'm getting the sound I expect.  This is using the SimpleTone example with hacks to go 32b and manually making sure I set the mute/demp pins.  It's humming away on my mini speaker just fine.  I also used the world's crappiest logic analyzer to capture the I2S signal from a regular Pico and I see no issues at 24 or 32b.  Data changes on the falling edge of bclk, and the lrclk is exactly 24 or 32 clocks long.  

````
#include <I2S.h>

// Create the I2S port using a PIO state machine
I2S i2s(OUTPUT);

// GPIO pin numbers
#define pBCLK 17
#define pWS (pBCLK+1)
#define pDOUT 16

const int frequency = 440; // frequency of square wave in Hz
const int amplitude = 1 << 30; // amplitude of square wave
const int sampleRate = 44100; // minimum for UDA1334A

const int halfWavelength = (sampleRate / frequency); // half wavelength of square wave

int32_t sample = amplitude; // current sample value
int count = 0;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, 1);
  pinMode(25, OUTPUT);
  digitalWrite(25, HIGH);
  pinMode(23, OUTPUT);
  digitalWrite(23, LOW);

  Serial.begin(115200);
  Serial.println("I2S simple tone");

  i2s.setBCLK(pBCLK);
  i2s.setDATA(pDOUT);
  i2s.setBitsPerSample(32);

  // start I2S at the sample rate with 16-bits per sample
  if (!i2s.begin(sampleRate)) {
    Serial.println("Failed to initialize I2S!");
    while (1); // do nothing
  }

}

void loop() {
  if (count % halfWavelength == 0) {
    // invert the sample every half wavelength count multiple to generate square wave
    sample = -1 * sample;
  }

  // write the same sample twice, once for left and once for the right channel
  i2s.write32(sample, sample);

  // increment the counter for the next sample
  count++;
}
````

DatanoiseTV, 2022-12-20:
Great, happy it works. Could you still extend your example code in the repo with the I2S callback example which refills the buffer with n samples? I still didn't have time to look into that and after that works I can include Vult DSP support for Arduino Pico.

DatanoiseTV, 2022-12-20:
Btw, if you choose the PicoADK Board in the Arduino IDE, then we don't need to manually define the I2S pins anymore :)

earlephilhower, 2022-12-21:
Some more insomnia driven testing this (very early) morning did show a problem when writing samples from the IRQ callback which really puzzles me.  So I can reproduce your problem andthat's the first step in finding a fix!

DatanoiseTV, 2022-12-21:
> Some more insomnia driven testing this (very early) morning did show a problem when writing samples from the IRQ callback which really puzzles me. So I can reproduce your problem andthat's the first step in finding a fix!

Maybe set a GPIO high / low on the IRQ to see if there is any weird problems?