earlephilhower/arduino-pico #941: USB serial input does not behave as expected

JoeEarlam, 2022-10-28:
Hi there

I'm trying to write a small routine to allow a user to send data to a sketch on the pico, it's based upon a fairly common Arduino forums sketch (I think by the user Robin2):

```
//serial
uint8_t         newData = false;            //flagged true when new serial data received
const uint8_t   numChars = 100;             //input buffer size
char            receivedChars[numChars];    //input buffer
uint8_t         ndx = 0;                    //received char index
boolean         recvInProgress = false;     //receive in progress flag

void setup() 
{
  Serial.begin(115200);
}

void loop() 
{
  pollSerial();
}

void pollSerial()
{
  char startMarker = '<';
  char endMarker = '>';
  char rc;

  while (Serial.available() > 0 && newData == false) 
  {
    rc = Serial.read();

    if (recvInProgress == true) 
    {
      if (rc != endMarker) 
      {
          receivedChars[ndx] = rc;
          ndx++;
          if (ndx >= numChars) 
          {
              ndx = numChars - 1;
          }
        }
        else 
        {
          receivedChars[ndx] = '\0'; // terminate the string
          recvInProgress = false;
          ndx = 0;
          newData = true;
        }
    }

    else if (rc == startMarker) 
    {
        recvInProgress = true;
    }
  }

  if(newData == true) 
  {
    Serial.print("Received chars: "); Serial.println(receivedChars);
    parseSerData();
    newData = false;
  }
}

void parseSerData() 
{
  char * strtokIndx; // this is used by strtok() as an index
  const uint16_t  SER_DATA_LEN = 16;    //maximum number of integers
  uint16_t  serData[SER_DATA_LEN];      //integers from serial packet
  char  tempChars[numChars];            //temporary array for use when parsing

  strcpy(tempChars, receivedChars); // this temporary copy is necessary to protect the original data because strtok() used in parseData() replaces the commas with \0
  
  strtokIndx = strtok(tempChars,",");     // get the first part - the string
  serData[0] = atoi(strtokIndx);        // convert this part to an integer

  for(uint8_t i = 0; i < SER_DATA_LEN-1; i++)
  {
    strtokIndx = strtok(NULL, ",");         // this continues where the previous call left off
    serData[i+1] = atoi(strtokIndx);        // convert this part to an integer
  }

  Serial.print("Received integers: ");

  for(uint8_t i = 0; i < SER_DATA_LEN; i++)
  {
    Serial.print(serData[i]); Serial.print(" ");
  }
  Serial.println();
}
```

If I run it on an Adafruit Feather M0 with the following input:

```<1,2,3>
<1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16>
<1,2,3>
```
I get the following output:

```
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
Received chars: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16
Received integers: 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 

```

However, on the pico with the same input I get:
```Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
Received chars: 1,2,3,4,5,6,7,8<1,2,3
Received integers: 1 2 3 4 5 6 7 8 2 3 0 0 0 0 0 0 
```

It seems Serial.available() is either not returning the correct number of characters, or Serial.read() is not reading/clearing them from the buffer correctly. I notice the number of characters that can be read successfully is always 16 - is this a buffer size limitation somewhere?

Please let me know if there's any extra information I can supply.

Thanks!

earlephilhower, 2022-10-28:
Interesting, thanks for the sketch.  I suppose I can just open a `minitem` or `cat` the strings directly to `/dev/TTYxxx` to reproduce your run?

I did a lot of work on the UART-based serial ports with respect to FIFOs, avoiding dropped chars, etc., but never had anyone with a problem in the USB stack side.  Will take a look...

earlephilhower, 2022-10-28:
I can't reproduce your failure in the IDE w/Serial Monitor.  I cut-n-pasted each line and hit Send, and got this
````
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
Received chars: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16
Received integers: 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
````

earlephilhower, 2022-10-28:
I also just pasted them using `minicom` (a dump serial terminal emulator) and had the exact same response:
````
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
Received chars: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16
Received integers: 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 
Received chars: 1,2,3
Received integers: 1 2 3 0 0 0 0 0 0 0 0 0 0 0 0 0 
````

JoeEarlam, 2022-10-28:
Interesting, I was using moserial on Ubuntu 20.04 - maybe it's a serial
terminal or USB issue? I will continue experimenting next week.

earlephilhower, 2022-10-28:
I never heard of moserial before, but looking at the source code it looks like it's busted and not checking for success on actually sending the bytes out to a port:

https://github.com/GNOME/moserial/blob/58e34e2c18b0b41873e21067294a290fd141784e/src/SerialConnection.vala#L95-L114

If `Posix.write()` behaves like POSIX `write`, then it returns the amount of data written.  That amount **may be less than the requested size**.  In this code it looks like it doesn't do an any handling of that case...if the USB port is bottlenecked this app will just throw away bytes...which is what you're seeing.

So my $0.02 would be to toss this `moserial` app or provide some fixes that check for partial writes and re-try accordingly.

JoeEarlam, 2022-10-31:
Hi, I can confirm that the output from the Arduino serial monitor works correctly. I'll close this issue and find a different terminal programme that actually works like it should. Thank you for your assistance, and for your work on this core in general! It's a great resource and a fantastic asset for the community, I'm really enjoying working with the Pico and having a solid core is a large part of that.

closed 2022-10-31 by JoeEarlam