earlephilhower/arduino-pico #944: WIFI connection leaks on Raspberry Pi Pico W board

NykyTV, 2022-10-30:
Hello,
I see some strange issue using the WiFi connection on a Raspberry Pico W board, it seems the WiFi connection leaks every some seconds. I'm getting a ping result of -1 in such case, where in normal case I get 5. 
I have seen this issue using PubSubClient MQTT communication first but stripped down the issue to simple WIFIClient connection plus ping, see code below.  WIFI FirmwareVersion of my board is 1.4.0.

It seems also other people have the same issues:
https://www.reddit.com/r/raspberrypipico/comments/wstmdl/bad_latency_on_the_pico_w_using_arduino_ide_and/

`
#include <WiFi.h>

#ifndef STASSID
#define STASSID "*****"
#define STAPSK "******"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

WiFiMulti multi;

void setup() {
  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  multi.addAP(ssid, password);

  if (multi.run() != WL_CONNECTED) {
    Serial.println("Unable to connect to network, rebooting in 10 seconds...");
    delay(10000);
    rp2040.reboot();
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  String fv = WiFi.firmwareVersion();
  Serial.println("FirmwareVersion: ");
  Serial.println(fv);
}

void loop() {
  Serial.println(WiFi.ping("192.168.0.20", 5));
}
`

earlephilhower, 2022-10-31:
I can't reproduce this on my network,  I've only adjusted the IP being pinged (tried my Linux desktop and the pfSense router) but I get consistent results over and over.  The "5"s roll over the Serial Monitor at a high rate.  If I cup my hands around the board (to decrease RSSI) I can get an occasional `-1` every 30 seconds or so, but it's definitely a periodic thing and seems reasonable to assume it's RF-power/noise related..

You might want to output `WiFi.RSSI()` and check if that's periodic.

The RF link layer is a black box, so there's not much we can do here.  Maybe you've got something on your network that's broadcasting periodically in a way that's incompatible w/the link? (just guessing here, but B/G/N compatibility w/the double-wide channels, etc.).

deival1980, 2022-11-01:
Hello Earle,
I'm working together with @NykyTV on some little "project" and wrote the first message above also together with him.
You suggested to call `WiFi.RSSI()` but I have found this in your code (version 2.6.1 installable using Arduino IDE):
```cpp
/*
    Return the current BSSID associated with the network.
    It is the MAC address of the Access Point

    return: pointer to uint8_t array with length WL_MAC_ADDR_LENGTH
*/
uint8_t* WiFiClass::BSSID(uint8_t* bssid) {
    // TODO - driver does not return this?!
    memset(bssid, 0xee, WL_MAC_ADDR_LENGTH);
    return bssid;
}

/*
    Return the current RSSI /Received Signal Strength in dBm)
    associated with the network

    return: signed value
*/
int32_t WiFiClass::RSSI() {
    // TODO - driver does not return this?!
    return 0;
}
```
I saw as well, that 2.6.2 fixes this, but I'm not quite sure how to update to 2.6.2. I will try to do this via GIT now.

earlephilhower, 2022-11-01:
> update this in Aduino IDE

If you're using the Board Manager, it should automatically recognize the new version when you start up, I think.  Or you can go to the board manager and then it should offer the option to upgrade.

OTW, `git` head would be fine as there are no major changes since the 2.6.2 release last week.

deival1980, 2022-11-01:
No it doesn't automatically update as long as you do not create a new release I think.

earlephilhower, 2022-11-01:
D'oh, you're right.  I thought the changes were in the 2.6.2 release but they were done after release (think the PR came in 3 hours after the promotion!)
https://github.com/earlephilhower/arduino-pico/compare/2.6.2...master

So, as you suggest, you'll need to use `git` (see the README for full instructions) to access these calls, or manually hack in the small changes required to your Arduino-installed version.

deival1980, 2022-11-01:
I'm running on git head now.
As far as I see ping is much more stable using that, could be also because I'm in a different room currently.
Also I see that using the WiFimulti is not connecting to the WIFI which has the highest RSSI if you re using the same SSIDs for several APs, you can see this in this log:
`WiFi connected
IP address: 
192.168.1.57
FirmwareVersion: 
1.4.0
scan done
7 networks found
1: WIFI_A (-96)* BSSID: 25:EF:C6:FD:3A:2C
2: WIFI_B (-80)* BSSID: 27:5D:53:14:E:C8
3: WIFI_C (-82)* BSSID: 27:5D:53:14:E:CA
4: WIFI_B (-69)* BSSID: 61:A6:F:6F:39:DC
5: WIFI_C (-70)* BSSID: 61:A6:F:6F:39:DE
6: WIFI_B (-61)* BSSID: E5:B7:60:70:DF:E8
7: WIFI_C (-63)* BSSID: E5:B7:60:70:DF:EA
WiFi connected: WIFI_B -69 BSSID: 61:A6:F:6F:39:DC
5
WiFi connected: WIFI_B -69 BSSID: 61:A6:F:6F:39:DC
5`

currently I'm using this code:

```cpp
#include <WiFi.h>

#ifndef STASSID
#define STASSID "WIFI_B"
#define STAPSK "*******"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

WiFiMulti multi;

void setup() 
{
  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  multi.addAP(ssid, password);

  if (multi.run() != WL_CONNECTED) {
    Serial.println("Unable to connect to network, rebooting in 10 seconds...");
    delay(10000);
    rp2040.reboot();
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  String fv = WiFi.firmwareVersion();
  Serial.println("FirmwareVersion: ");
  Serial.println(fv);

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
  {
      Serial.println("no networks found");
  } 
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.print((WiFi.encryptionType(i) == CYW43_AUTH_OPEN)?" ":"*");
      byte bssid[6];
      WiFi.BSSID(i, bssid);    
      Serial.print(" BSSID: ");
      Serial.print(bssid[5],HEX);
      Serial.print(":");
      Serial.print(bssid[4],HEX);
      Serial.print(":");
      Serial.print(bssid[3],HEX);
      Serial.print(":");
      Serial.print(bssid[2],HEX);
      Serial.print(":");
      Serial.print(bssid[1],HEX);
      Serial.print(":");
      Serial.println(bssid[0],HEX);
      delay(10);
    }
  }
}

void loop()
{
  //if the connection to the stongest hotstop is lost, it will connect to the next network on the list
  if (multi.run(5000) == WL_CONNECTED)
  {
    Serial.print("WiFi connected: ");
    Serial.print(WiFi.SSID());
    Serial.print(" ");
    long rssi = WiFi.RSSI();
    Serial.print(rssi);
    Serial.print(" ");
    byte bssid[6];
    WiFi.BSSID(bssid);    
    Serial.print("BSSID: ");
    Serial.print(bssid[5],HEX);
    Serial.print(":");
    Serial.print(bssid[4],HEX);
    Serial.print(":");
    Serial.print(bssid[3],HEX);
    Serial.print(":");
    Serial.print(bssid[2],HEX);
    Serial.print(":");
    Serial.print(bssid[1],HEX);
    Serial.print(":");
    Serial.println(bssid[0],HEX);
  }
  else 
  {
    Serial.println("WiFi not connected!");
  }
  
  Serial.println(WiFi.ping("192.168.0.1", 5));

  delay(100);
}
```

I guess this is because the implementation compared to ESP is different:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp

ESP uses `WiFi.begin(ssid, entry.passphrase, **channel, bssid**);` which is not implemented for Pico, right?
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L334

earlephilhower, 2022-11-01:
> WiFimulti is not connecting to the WIFI which has the highest RSSI if you re using the same SSIDs for several APs

Unfortunately the Pico-SDK's CYW43 driver does not allow requesting a specific BSSID.  (And as you just found out, it doesn't even support reporting the connection RSSI!  An end user hacked it in from reverse-engineering the MicroPython driver sources).  The WiFiMulti class just picks the name of the best one, but can't control which actual BSSID the WiFi chip/blob actually pick.

deival1980, 2022-11-01:
OK forgot to tell you, that our WIFI is a 'mesh' WIFI which is managed by the main router/AP. If the main router sees that the RSSI of a device is not the best and could be better using a different repeater connected to it, it can force the reconnect of the device somehow. It seems that the Pico is almost always connecting to the first AP in the scan list with matching SSID.
My main AP is then forcing to reconnect and almost every time it picks then the one with highest RSSI.
But it also happens very often at my side, that also with very high and constant RSSI values, that the WIFI has ping issues:
```
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -44 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -44 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -44 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -42 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 -1
WiFi connected: WIFI_B -43 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -44 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -47 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -47 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -45 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -46 BSSID: E5:B7:60:70:DF:E8 5
WiFi connected: WIFI_B -47 BSSID: E5:B7:60:70:DF:E8 5
```

deival1980, 2022-11-01:
> Unfortunately the Pico-SDK's CYW43 driver does not allow requesting a specific BSSID

So as far as I understand a issue should be opened in pico-sdk, right?

earlephilhower, 2022-11-01:
> So as far as I understand a issue should be opened in pico-sdk, right?

The CYW43 driver is actually a separate repo, but you could start w/ Pico-SDK and ask for it a feature addition and see where it leads.  I'd recommend bunching up all the WiFi additions so as to keep the noise down, they're busy guys.

> OK forgot to tell you, that our WIFI is a 'mesh' WIFI which is managed by the main router/AP. If the main router sees that the RSSI of a device is not the best and could be better using a different [...]
My main AP is then forcing to reconnect and almost every time it picks then the one with highest RSSI.
But it also happens very often at my side, that also with very high and constant RSSI values, that the WIFI has ping issues:

If the mesh is dropping the PicoW when we try and send a ping, then I would imagine the driver blob just drops the packet.  Since this is an ICMP ping and not a TCP-based one there's no built-in retransmit.  Not sure there's anything to be done about that with the knobs we have.

Is the drop a periodic occurrence (i.e. every X seconds you see a drop)?  Is there any way to disable the mesh bits to ensure it's not somehow still causing grief?  My own WiFi setup is much simpler with a several independent APs w/the same PSK.

Again, though, the actual WiFi blob is not under our control and just sends or received packets from the core.  So while it's not perfect, we really can't do much about it.  Luckily it seems to be infrequent and doesn't result in permanent connectivity drops in your case.

One very unlikely possibility is that https://github.com/earlephilhower/arduino-pico/blob/9b28fc3e179e22482352db018a6fd3e3d0404fcb/libraries/lwIP_CYW43/src/utility/CYW43shim.cpp#L117-L137
is dropping the received response packet.  On a complely unloaded system like your test that seems highly unlikely, but you could add a global variable that increments in the case of the dropping and print that every `ping` to see if it increments on the failure.  If so, then I can see about reworking things.  Since I can't repro your failure I can't actually try anything on my end.

For anything more I think we need to dig into your router.  Can you get raw Ethernet packet dumps to see if the echo is lost on the way from the Pico->router or from router->pico consistently?

deival1980, 2022-11-02:
> Is the drop a periodic occurrence (i.e. every X seconds you see a drop)?
No it is very random like.

> Is there any way to disable the mesh bits to ensure it's not somehow still causing grief?
That is hard to do, as it is a very restricted user interface, not very much config possibilities. I can try other situations using a Smartphone WIFI AP or a Raspberry Pi used to simulate a AP later. But first of all I think we should follow up on the last mentioned below.

> One very unlikely possibility is that is dropping the received response packet. On a complely unloaded system like your test that seems highly unlikely, but you could add a global variable that [...]
Using that I found one issue. I added variables like this:
```cpp
// CB from the cyw43 driver
extern "C" void cyw43_cb_process_ethernet(void *cb_data, int itf, size_t len, const uint8_t *buf) {
    //cyw43_t *self = (cyw43_t *)cb_data
    (void) cb_data;
    (void) itf;
    struct netif *netif = CYW43::_netif; // &self->netif[itf];
#if CYW43_NETUTILS
    if (self->trace_flags) {
        cyw43_ethernet_trace(self, netif, len, buf, NETUTILS_TRACE_NEWLINE);
    }
#endif
    if (netif->flags & NETIF_FLAG_LINK_UP) {
        struct pbuf *p = pbuf_alloc(PBUF_RAW, len, PBUF_POOL);
        if (p != nullptr) {
            pbuf_take(p, buf, len);
            if (__inLWIP || (netif->input(p, netif) != ERR_OK)) {
                CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF++;
				pbuf_free(p);
            }
            CYW43_STAT_INC(PACKET_IN_COUNT);
        }
		else
		{
			CYW43_GLOBVAR_COUNTER_DROP_PCKTS++;
		}
    }
}
```
My loop() looks like this:
```cpp
void loop()
{
  //Serial.print("\x1b[1G\x1b[K");
  
  //if the connection to the stongest hotstop is lost, it will connect to the next network on the list
  if (multi.run(5000) == WL_CONNECTED)
  {
    Serial.print("WiFi connected: ");
    Serial.print(WiFi.SSID());
    Serial.print(" ");
    long rssi = WiFi.RSSI();
    Serial.print(rssi);
    Serial.print(" ");
    byte bssid[6];
    WiFi.BSSID(bssid);    
    Serial.print("BSSID: ");
    Serial.print(bssid[5],HEX);
    Serial.print(":");
    Serial.print(bssid[4],HEX);
    Serial.print(":");
    Serial.print(bssid[3],HEX);
    Serial.print(":");
    Serial.print(bssid[2],HEX);
    Serial.print(":");
    Serial.print(bssid[1],HEX);
    Serial.print(":");
    Serial.print(bssid[0],HEX);
  }
  else 
  {
    Serial.println("WiFi not connected!");
  }
  
  Serial.print(" ");
  Serial.print(WiFi.ping("192.168.0.1", 5));

  Serial.print(" ");
  Serial.print(CYW43_GLOBVAR_COUNTER_DROP_PCKTS);
  Serial.print(" ");
  Serial.println(CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF);

  delay(100);

  if (Serial.available())
  {
    char chr = Serial.read();
    if (chr == 'r') 
    {
      Serial.print("rebooting...");
      delay(1000);
      rp2040.reboot();
    }
  }
}
```
using that I saw something like this:
```
WiFi connected: rainer@rkieslinger.de_home -74 BSSID: 61:A6:F:6F:39:DC 5 0 0
WiFi connected: rainer@rkieslinger.de_home -74 BSSID: 61:A6:F:6F:39:DC 5 0 0
WiFi connected: rainer@rkieslinger.de_home -75 BSSID: 61:A6:F:6F:39:DC 5 0 0
WiFi connected: rainer@rkieslinger.de_home -75 BSSID: 61:A6:F:6F:39:DC -1 0 1
WiFi connected: rainer@rkieslinger.de_home -75 BSSID: 61:A6:F:6F:39:DC 5 0 1
WiFi connected: rainer@rkieslinger.de_home -74 BSSID: 61:A6:F:6F:39:DC 5 0 1
WiFi connected: rainer@rkieslinger.de_home -74 BSSID: 61:A6:F:6F:39:DC 5 0 1
WiFi connected: rainer@rkieslinger.de_home -74 BSSID: 61:A6:F:6F:39:DC 5 0 1
```
or:
```
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -68 BSSID: E5:B7:60:70:DF:E8 5 0 0
WiFi connected: rainer@rkieslinger.de_home -69 BSSID: E5:B7:60:70:DF:E8 -1 0 2
WiFi connected: rainer@rkieslinger.de_home -66 BSSID: E5:B7:60:70:DF:E8 5 0 2
WiFi connected: rainer@rkieslinger.de_home -67 BSSID: E5:B7:60:70:DF:E8 5 0 2
WiFi connected: rainer@rkieslinger.de_home -68 BSSID: E5:B7:60:70:DF:E8 5 0 2
WiFi connected: rainer@rkieslinger.de_home -68 BSSID: E5:B7:60:70:DF:E8 5 0 2
WiFi connected: rainer@rkieslinger.de_home -66 BSSID: E5:B7:60:70:DF:E8 5 0 2
```

deival1980, 2022-11-02:
Did some further investigations and changed my loop code like this:
```cpp
void loop()
{
  //Serial.print("\x1b[1G\x1b[K");
  
  //if the connection to the stongest hotstop is lost, it will connect to the next network on the list
  if (multi.run(5000) == WL_CONNECTED)
  {
    Serial.print("WiFi connected: ");
    Serial.print(WiFi.SSID());
    Serial.print(" ");
    long rssi = WiFi.RSSI();
    Serial.print(rssi);
    Serial.print(" ");
    byte bssid[6];
    WiFi.BSSID(bssid);    
    Serial.print("BSSID: ");
    Serial.print(bssid[5],HEX);
    Serial.print(":");
    Serial.print(bssid[4],HEX);
    Serial.print(":");
    Serial.print(bssid[3],HEX);
    Serial.print(":");
    Serial.print(bssid[2],HEX);
    Serial.print(":");
    Serial.print(bssid[1],HEX);
    Serial.print(":");
    Serial.print(bssid[0],HEX);
  }
  else 
  {
    Serial.println("WiFi not connected!");
  }
  
  Serial.print(" ");
  Serial.print(WiFi.ping("192.168.0.1", 5));

  Serial.print(" ");
  Serial.print(CYW43_GLOBVAR_COUNTER_DROP_PCKTS);
  Serial.print(" ");
  Serial.print(CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF);
  for(int i = 0; i< 10;i++)
  {
    Serial.print("#");
    Serial.print(CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF_STATE[i]);
    Serial.print("=");
    Serial.print(CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF_BOOL[i]);
  }
  Serial.println(" ");

  delay(100);

  if (Serial.available())
  {
    char chr = Serial.read();
    if (chr == 'r') 
    {
      Serial.print("rebooting...");
      delay(1000);
      rp2040.reboot();
    }
  }
}
```
and lwip callback function code like this:
```cpp
// CB from the cyw43 driver
extern "C" void cyw43_cb_process_ethernet(void *cb_data, int itf, size_t len, const uint8_t *buf) {
    //cyw43_t *self = (cyw43_t *)cb_data
    (void) cb_data;
    (void) itf;
    struct netif *netif = CYW43::_netif; // &self->netif[itf];
#if CYW43_NETUTILS
    if (self->trace_flags) {
        cyw43_ethernet_trace(self, netif, len, buf, NETUTILS_TRACE_NEWLINE);
    }
#endif
    if (netif->flags & NETIF_FLAG_LINK_UP) {
        struct pbuf *p = pbuf_alloc(PBUF_RAW, len, PBUF_POOL);
        if (p != nullptr) {
            pbuf_take(p, buf, len);
            if ((CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF_BOOL[CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF] = __inLWIP) || (CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF_STATE[CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF] = netif->input(p, netif) != ERR_OK)) {
                CYW43_GLOBVAR_COUNTER_DROP_PCKTS_IF++;
				pbuf_free(p);
            }
            CYW43_STAT_INC(PACKET_IN_COUNT);
        }
		else
		{
			CYW43_GLOBVAR_COUNTER_DROP_PCKTS++;
		}
    }
}
```
my logs are looking like this:
```
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 0#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 0#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 0#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 0#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 -1 0 1#0=1#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -64 BSSID: E5:B7:60:70:DF:E8 5 0 1#0=1#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 1#0=1#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 1#0=1#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1 
WiFi connected: WIFI_B -65 BSSID: E5:B7:60:70:DF:E8 5 0 1#0=1#0=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=0#-99=1
```

earlephilhower, 2022-11-02:
Can you give #944 a try and report back?  I just threw it together before leaving for work this morning, so it's not tested, but I believe it will now just not pull any packets from the CYW43 when in LWIP, instead of pulling and dropping.  On the next time slice it will try again (the CYW43 holds packets in its own memory) and eventually retrieve the packets.

deival1980, 2022-11-02:
The not supported 11k and 11v options are also shown up using MicroPython code and I can also see the interruptions in case of pinging using the following code:
```python
import network
import socket
from time import sleep
from picozero import pico_temp_sensor, pico_led
import machine

ssid = 'rainer@rkieslinger.de_home'
password = '977wng2h8o9m3iap&/%$dsc5f5s587s6fzs73d5f7d6tsdtfr5%!$?&{*gfdt})'

wlan = network.WLAN(network.STA_IF)

def checksum(data):
    if len(data) & 0x1: # Odd number of bytes
        data += b'\0'
    cs = 0
    for pos in range(0, len(data), 2):
        b1 = data[pos]
        b2 = data[pos + 1]
        cs += (b1 << 8) + b2
    while cs >= 0x10000:
        cs = (cs & 0xffff) + (cs >> 16)
    cs = ~cs & 0xffff
    return cs

def ping(host, count=4, timeout=5000, interval=10, quiet=False, size=64):
    import utime
    import uselect
    import uctypes
    import usocket
    import ustruct
    import urandom

    # prepare packet
    assert size >= 16, "pkt size too small"
    pkt = b'Q'*size
    pkt_desc = {
        "type": uctypes.UINT8 | 0,
        "code": uctypes.UINT8 | 1,
        "checksum": uctypes.UINT16 | 2,
        "id": uctypes.UINT16 | 4,
        "seq": uctypes.INT16 | 6,
        "timestamp": uctypes.UINT64 | 8,
    } # packet header descriptor
    h = uctypes.struct(uctypes.addressof(pkt), pkt_desc, uctypes.BIG_ENDIAN)
    h.type = 8 # ICMP_ECHO_REQUEST
    h.code = 0
    h.checksum = 0
    h.id = urandom.randint(0, 65535)
    h.seq = 1

    # init socket
    sock = usocket.socket(usocket.AF_INET, usocket.SOCK_RAW, 1)
    sock.setblocking(0)
    sock.settimeout(timeout/1000)
    addr = usocket.getaddrinfo(host, 1)[0][-1][0] # ip address
    sock.connect((addr, 1))
    not quiet and print("PING %s (%s): %u data bytes" % (host, addr, len(pkt)))

    seqs = list(range(1, count+1)) # [1,2,...,count]
    c = 1
    t = 0
    n_trans = 0
    n_recv = 0
    finish = False
    while t < timeout:
        if t==interval and c<=count:
            # send packet
            h.checksum = 0
            h.seq = c
            h.timestamp = utime.ticks_us()
            h.checksum = checksum(pkt)
            if sock.send(pkt) == size:
                n_trans += 1
                t = 0 # reset timeout
            else:
                seqs.remove(c)
            c += 1

        # recv packet
        while 1:
            socks, _, _ = uselect.select([sock], [], [], 0)
            if socks:
                resp = socks[0].recv(4096)
                resp_mv = memoryview(resp)
                h2 = uctypes.struct(uctypes.addressof(resp_mv[20:]), pkt_desc, uctypes.BIG_ENDIAN)
                # TODO: validate checksum (optional)
                seq = h2.seq
                if h2.type==0 and h2.id==h.id and (seq in seqs): # 0: ICMP_ECHO_REPLY
                    t_elasped = (utime.ticks_us()-h2.timestamp) / 1000
                    ttl = ustruct.unpack('!B', resp_mv[8:9])[0] # time-to-live
                    n_recv += 1
                    not quiet and print("%u bytes from %s: icmp_seq=%u, ttl=%u, time=%f ms" % (len(resp), addr, seq, ttl, t_elasped))
                    seqs.remove(seq)
                    if len(seqs) == 0:
                        finish = True
                        break
            else:
                break

        if finish:
            break

        utime.sleep_ms(1)
        t += 1

    # close
    sock.close()
    ret = (n_trans, n_recv)
    not quiet and print("%u packets transmitted, %u packets received" % (n_trans, n_recv))
    return (n_trans, n_recv)

def connect():
    #Connect to WLAN
    wlan.active(True)
    wlan.connect(ssid, password)
    while wlan.isconnected() == False:
        print('Waiting for connection...')
        sleep(1)
        print(wlan.ifconfig())

try:
    #print(wlan.scan())
    connect()
    while True:
        ping('192.168.0.1',1)
        print(wlan.status())
        #print(wlan.status('bssid'))
except KeyboardInterrupt:
    machine.reset()
```

earlephilhower, 2022-11-02:
Thanks for the update on the test.

Your Python results make sense.  The 802.x stuff is handled in the CYW43 chip and its own blob firmware.  We don't have control there from the Pico.

closed 2022-11-02 by earlephilhower

deival1980, 2022-11-09:
Hello @earlephilhower,
I just would like to inform you that the PicoW WIFI feature is still completely unstable for me, at least with my MQTT use case I see very big delays in response and repeating reconnections of the MQTT. I'm now completly annoyed by the PicoW maturity, and stepped over in my project to a already known ESP32 module which works without any delays and reconnections using the same MQTT libs.
I guess that there is still some issue with lost packets and maybe also retransmissions of packets. I would suggest to do some WIFI performace, reliability and stress tests using some functional known AP where you have full tracing abilities, e.g. Raspberry PI running some Kali Linux image.

k-madsenDK, 2024-10-31:
my pico is disconnecting from the network after several hours.
so in the start of the loop i added this ->

int connectwifi(void){
    int statuslocal = WiFi.begin((char *)ssid, (char *) password);
    return(statuslocal);
  }// end connectwifi

 if(WiFi.status() != WL_CONNECTED){
    Serial.println("Disconectet");
    status = connectwifi();// er der disconect reconect  
 }

It works the pico is joining the net again :-)

PR #951 (2022-11-02): Properly disable Ethernet IRQ, don't drop packets
Fix #944
commit 63dfd9ab (2022-11-02): Properly disable Ethernet IRQ, don't drop packets (#951)

Fix #944