earlephilhower/arduino-pico #946: rp2040 class internals functions not found by the compiler

stefat, 2022-10-31:
First of all thank you for this arduino-pico core: great job!
I find it very useful and handy. 
However I found few functions of the rp2040 helper class not found in the IDE ver. 1.8.19, e.g.:
rp2040.wdt_begin(8000);
rp2040.reboot();
rp2040.wdt_reset();
Other rp2040 functions, instead, properly work, e.g.:
rp2040.idleOtherCore();
rp2040.resumeOtherCore();
Any clue ?
Thanks, and keep up your good work.

stefat, 2022-10-31:
The compiler does not find the three functions of the rp2040 listed above and issues an error.  Where do I find the examples ?

earlephilhower, 2022-10-31:
`rp2040.reboot()` is defined here:
https://github.com/earlephilhower/arduino-pico/blob/44eeebbc6f8d94fb9dda34e339fb356f4a15dc88/cores/rp2040/RP2040Support.h#L306-L311

and used in the `WiFiClient.ino` example here (which is built as part of every CI):
https://github.com/earlephilhower/arduino-pico/blob/44eeebbc6f8d94fb9dda34e339fb356f4a15dc88/libraries/WiFi/examples/WiFiClient/WiFiClient.ino#L33-L37

Make sure you're running the latest release (this past weekend a new version came out) as these calls were added a month or so back.  Also make sure you're using this core and not the `Arduino MBED` version which is completely different and not partof the core here.

closed 2022-10-31 by earlephilhower

stefat, 2022-11-01:
Sorry having bothered you. I was using version 1.10.0 .... quite old. Thanks.