earlephilhower/arduino-pico #952: custom rp2040 board identifying itself as "Pico W" over USB

Ing-Dom, 2022-11-03:
I recently noticed that a custom RP2040 board loaded with a blinky sketch identifies itself as a "Pico W" (Wifi) when I connect it via USB to Win10.
No idea where that text exactly cames from and if it is configureable?

earlephilhower, 2022-11-03:
The board is detected by the VID:PID provided by the code running on it.  You've just ended up using one of the ones for the PicoW.

It's only used to show that string in the IDE, so it's not a real problem.  You could change your VID:PID or just ignore it.

closed 2022-11-03 by earlephilhower

Ing-Dom, 2022-11-03:
Now I see.. for the Generic Board, 0xF00A PID seems to be set, what is a Pico W...

generic.build.usbpid=-DSERIALUSB_PID=0xf00a

strange.. why is that?