earlephilhower/arduino-pico #954: Add BSSID connection possibility

earlephilhower, 2022-11-03:
> Unfortunately the Pico-SDK's CYW43 driver does not allow requesting a specific BSSID

So as far as I understand a issue should be opened in pico-sdk, right?

_Originally posted by @deival1980 in https://github.com/earlephilhower/arduino-pico/issues/944#issuecomment-1299295309_

earlephilhower, 2022-11-03:
Looking at the low-level driver, there is a BSSID input to `cyw43_wifi_join` but it's not exposed by the `pico-sdk`.  It should be almost trivial to add to the SDK.

earlephilhower, 2022-11-03:
PR/bug submitted upstream:
https://github.com/raspberrypi/pico-sdk/issues/1090 and https://github.com/raspberrypi/pico-sdk/pull/1091

deival1980, 2022-11-04:
Did you test, if it also allows to roam into other APs (with different BSSIDs) of the same SSID also in case you connected using a specific BSSID? This would be needed if the PicoW moves in the room, I think. ...first connect to the strongest, and later if moving in the room connect to the next strongest...

earlephilhower, 2022-11-04:
> Did you test, if it also allows to roam into other APs (with different BSSIDs) of the same SSID also in case you connected using a specific BSSID?

That's not how specifying the SSID+BSSID works anywhere, I would argue.  By listing both you are explicitly asking for AP X not Y, so I would not expect it to reconnect to another AP.  Your main loop would need to check for `WiFi.connected()` and if not then `WiFiMulti.run()` (or whatever...this is from memory) to have it manually specify the next.

To do what you suggest, I think you'd need the 802.11(X?V?) link-layer migration working in the CYW43 and not specify a hardcoded BSSID.

PR #1167 (2023-02-28): Update to SDK 1.5, add alpha-level BT support, use Pico-SDK CYW43 infrastructure
Fixes #1212
Fixes #1113 
Fixes #954 
Fixes #906
closed 2023-02-28 by earlephilhower