earlephilhower/arduino-pico #955: Problems with default pwm resolution

jjsch-dev, 2022-11-03:
Hello, using the default resolution:

static uint32_t analogScale = 255;
static uint32_t analogFreq = 1000;

analogWrite(10, 255); outputs about 25 percent on the output pin.

[image]

analogWrite(10, 512); outputs about 50 percent on the output pin.

[image]

analogWrite(10, 100); outputs about 99 percent at the output pin.

[image]

analog write (10, 1020); outputs 100 percent on the output pin.

[image]

Is there a way to set the resolution to be 8bit at frequencies below 1.5KHZ?

earlephilhower, 2022-11-04:
This is an interesting problem, thanks for the graphs.  Does look like a bug from the images.

There's code which (should) handles this in most cases, but maybe we've hit an overflow case here (`analogWritePseudoScale`):
https://github.com/earlephilhower/arduino-pico/blob/40855c90fb1d37a7544c1324c291a2413ff48bdc/cores/rp2040/wiring_analog.cpp#L81-L101

earlephilhower, 2022-11-04:
There is some debug info from the core to quickly verify this.  

Can you please re-run your testing with `Tools->Debug Port->Serial` and `Tools->Debug Level->Core` and capture the logs (you may need a `delay(5000);` at the head of your `setup()` to make sure the printouts don't start before the PC and the Pico negotiate all the USB stuff....otherwise you may miss the `DEBUGCORE` messages shown above.)

jjsch-dev, 2022-11-04:
Thank you for taking the time to review the report, you do a great job.

Turn on core debugging and these are the results for: analogWrite(255):

10:04:42.270 -> Adjusting analogWrite values PS=2, scale=510
10:04:42.270 -> Adjusting analogWrite values PS=3, scale=1020

earlephilhower, 2022-11-04:
That's very odd and seems to indicate it should work.
https://github.com/earlephilhower/arduino-pico/blob/40855c90fb1d37a7544c1324c291a2413ff48bdc/cores/rp2040/wiring_analog.cpp#L103-L104

`analogWritePseudoScale` = 3, `analogWriteSlowScale` = 1 (the default) so it should take your inputs and shift left (3-1 = 2) or convert it to 0...1023 automatically.

I'll need to single step through the code and see what's going on.  Can you confirm your code is like this:

````
analogWriteRange(255);
analogWriteFreq(1000);
analogWrite(10, 255);
````

fjansson, 2022-11-04:
@jjsch-dev more questions: 
* Are you using the latest version of arduino-pico, or which version?  
* What is your system clock frequency (set by "CPU Speed" in tools menu)? 
* When you measured the signal, did you by any chance use analogWrite on another pin simultaneously with other frequency or analogRange? (because the pins aren't all independent, some combinations of pins affect each other)

The debug output you got looks ok to me, as if the code is working as it should. 

Maybe there's a typo in the third test, so it should read analogWrite(10, **1000**); outputs about 99 percent at the output pin. Then your measurements look as if the range would be about 1020.

earlephilhower, 2022-11-04:
@fjansson I think you just described a bug, then.  Might not be what is happening here, need more info, but I see how the change from "all pins initted to same PWM settings" to "allow individual PWM settings per pin" has busted things.

When there was only 1 setting for all pins, then a single `analogWritePrescale` and `analogWriteSlowScale` is needed.

Now, we need one of those variable for each PWM slice, or you could set slice A to 10khz@1024, set slice B to 1khz@255, and then have a write to slice A have the wrong setting...

jjsch-dev, 2022-11-04:
@fjansson thanks for the tips.

- Yes, I use the last version from Arduino Board Manager.
[image]

- The system clock frequency is: 133 Mhz.
- Yes, I use two pin with the same PWM value.
- you're right, it's a typo, it should be analogWrite(10, 1000);

@earlephilhower to avoid polluting the sample (my fault), I wrote the following imo with the pwm functions. And I found that when I use a single analogWrite(), the output behaves as expected. But with two, as @fjansson suggests, the strange behavior appears.

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(5000);
  
  analogWriteRange(255);
  analogWriteFreq(1000);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);

  analogWrite(10, 255);
  analogWrite(11, 255);
}

These are the screenshots with the different options:
analogWrite(10, 255);
//analogWrite(11, 255);
[image]

analogWrite(10, 255);
analogWrite(11, 255);
[image]

analogWrite(10, 127);
analogWrite(11, 127);
[image]

analogWrite(10, 127);
//analogWrite(11, 127);
[image]

//analogWrite(10, 127);
analogWrite(11, 127);
[image]

earlephilhower, 2022-11-05:
I seem to have the world's flakiest noname 8-bit logic analyzer clone, so testing for me is difficult.  It's hard to know if there's a PWM issue when the pins themselves seem to read out with lots of random noise.  :sob: 

That said, I've got #962 for you all to try out on your apps and report back on.  The PWM devices are slice-based, not pin-based, so this change pretty minimally ensures the init isn't duplicated when you're hitting the same slice (as in pins 10, 11 in the failing test).

If you could give it a go and let me know if this clears things up I would much appreciate it!

earlephilhower, 2022-11-05:
Latest push on #962 seems to clear it up.  I've written 3 pins in 2 different slices and the PWM values all look good.  Please test w/your own app and report back.

jjsch-dev, 2022-11-05:
@earlephilhower  I have updated the arduinopico version to the ppp branch and compiled the code successfully. I see that you added the variable static bool `scaleInitted = false;` in the `wired_analog.cpp` file, which I assume will not re-init the pwm range when the `analogWrite` function is called more than once.

Unfortunately, I don't have the oscilloscope at home to take the measurements right now, butr I'll do them no later than Monday and send them to you.

I wonder what the world's flakiest unnamed 8-bit logic analyzer clone is? :thinking:

revell1, 2022-11-06:
This appears to have fixed (restored) my motor control WRT V2.6.1 (also linked with issues #962 and #961)

earlephilhower, 2022-11-06:
Thanks!  I'll wait for @jjsch-dev 's report for the merge.

jjsch-dev, 2022-11-07:
@earlephilhower  Good job, this push #962 fixed the problem. Below I put the captures for: `analogWriteRange(255);` and `analogWriteFreq(1000);`. Thanks for your time.

analogWrite(10, 255);
analogWrite(11, 255);
[image]

analogWrite(10, 127);
analogWrite(11, 255);
[image]

analogWrite(10, 255);
analogWrite(11, 127);
[image]

analogWrite(10, 127);
analogWrite(11, 127);
[image]

commit 23e3d63a (2022-11-05): Fix analogWrite scaling issues

The PWM internals on the RP2040 are based on 8 slices with independent
clocking.  Make sure that we init the PWM slice being used only once per
change of frequency/range.

Only init the scaling values one time, because we also adjust the input
scale/frequency values when calculating them.  If we ran this twice (i.e.
writing to two slices), it would overwrite the pseudo scale/slow scale
values with 1 causing the wrong PWM values to be set.

Fixes #955
commit 895ffced (2022-11-07): Fix analogWrite scaling issues (#962)

The PWM internals on the RP2040 are based on 8 slices with independent
clocking.  Make sure that we init the PWM slice being used only once per
change of frequency/range.

Only init the scaling values one time, because we also adjust the input
scale/frequency values when calculating them.  If we ran this twice (i.e.
writing to two slices), it would overwrite the pseudo scale/slow scale
values with 1 causing the wrong PWM values to be set.

Fixes #955
closed 2022-11-07 by earlephilhower