earlephilhower/arduino-pico #956: Re: "Add AP mode with simple DHCP server" for Pico W WiFi

kenneth558, 2022-11-04:
I'm starting with a project I wrote with Arduino IDE in Linux Mint I'm porting - going from non-wifi Arduino environment to RPi Pico W in hopes of enhancing the way continuous real-time data will be streamed from my training aid to the user/student.  The Arduino version of this project simply sent the data over USB to the Arduino IDE plotter tool, but I hope to change from USB to wifi - utilize the second core of the RPi Pico W to serve a web page (continuous real-time data will be a secondary challenge for sure).  As I study this out and make feeble attempts with this notion, I sense there is a lack of software tools to do this with the Arduino IDE in c++ that would allow me not to have to port the entire project into micropython or take it into the RPi SDK, or other frameworks/environments.  As trivial as those two+ options MIGHT or MIGHT NOT be, I certainly don't feel adequately knowledgeable to embark down either one confident that I'm accurately foreseeing the complexity either would present.  

All has gone according to plan in bringing the project's full Arduino-level functionality into the RPI Pico W environment, separating out the circuitry control workload to core 0 and the data streaming workload to core 1.  But getting the core I've designated for data streaming USB+wifi data fails at the step of the Pico W becoming an AP from which to serve a web page.  It does not appear to broadcast wifi at all - I don't see any available for me on my Android to select.  Note I am not explicitly including any wifi.h:
  
#include <WebServer.h>
#include <WiFiClient.h>
#include <LEAmDNS.h>
const char *ssid = STASSID;
const char *password = STAPSK;
uint8_t port = 80;
WebServer server( port );

void handleRoot() { <code unchanged except to disable LED; culminates in server.send(200, "text/html", temp);> }
void handleNotFound() { <code unchanged except to disable LED; culminates in server.send(404, "text/plain", message);> }
void drawGraph() { <code unchanged except to disable LED; culminates in server.send(200, "image/svg+xml", out);> }

void setup1(void) {
  WiFi.mode( WIFI_AP );
  WiFi.setHostname( "PicoW2" );
  WiFi.begin( ssid, password );
  while ( WiFi.status() != WL_CONNECTED ) {;
    delay( 500 );
    Serial.print( "." );
  }
  server.begin();  //Also executed below and without error, but unsure if this is OK ???
  if (MDNS.begin("picow")) {
    Serial.println("MDNS_responder_started");
  }
  server.on("/", handleRoot);
  server.on("/test.svg", drawGraph);
  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });
  server.onNotFound(handleNotFound);
  server.begin();
  
  Serial.println("HTTP_server_started");

}
bool firstRun = true;
void loop1()
{
  PlotThePoints();
//  delay( 1 );
  if( firstRun == true)
  {
        firstRun = false;
//        server.setNoDelay(true); //no such member.  Time to wonder if Arduino IDE will even work
        server.handleClient();
        MDNS.update();
        while( !Serial.available() ); //temporary code during troubleshooting to wait for programmer go-ahead
//shouldn't there be an AP broadcast by this point? I don't see any available for my Android to select
        Serial.read(); //temporary code during troubleshooting
  }
}

My next troubleshooting step was to investigate the reality of your note I see that says "Add AP mode with simple DHCP server" in your July 15 comments for "Add Pico W WiFi support".  I don't find the code for that capability anywhere in my Arduino IDE menus to allow me to make any troubleshooting steps in that direction I hoped to go.  

Is the Arduino IDE up to the task for a web server on the RPi Pico W, yet?  Or do I have to get born again into a different framework?  Or is some mistake in the above code my only problem?  Maybe I am mixing incompatible libraries (b/c this compilation becomes unstable within a few seconds)? Thank you for pointing me to the right path for success!!!!!

earlephilhower, 2022-11-04:
AP mode works and there is an example you can check out from the IDE.

Are you sure you want AP mode?  If you want to serve a web page that you can access anywhere on your network you don't need AP mode at all, just a `WebServer` and normal (STA) mode.  AP mode makes a new hotspot that you need to manually connect your phone or laptop to (i.e. disconnect from home Wifi, connect to Pico hotspot).

I suggest you swap your cores.  Core0 runs the USB and also really should be the one doing any WiFi operations.  Core1 is  100% unused by default.

kenneth558, 2022-11-04:
Great advice!  And Yes, I do want the devices to function autonomously without the need for any other wifi whatsoever.  At the same time, there will be other situations that your thinking is the most flexible, so BOTH will need to exist depending on the environment.

I trust you saw the line of code above:

server.setNoDelay(true); //no such member.  Time to wonder if Arduino IDE will even work

where I am led to suspect I'm mixing libraries or something.  I'll stay the course and Thank you!!!!

earlephilhower, 2022-11-04:
I didn't go over the code, no.  There are several issues as you noted.  `server.begin()` should only be called once. `setNoDelay` is only a per-TCP connection thing, not a server one, and it is only a hint to the TCP/IP stack to send data immediately (i.e. lots of little packets) instead of waiting a bit to possibly combine transmits (i.e. fewer, larger packets with less overhead).

This doesn't look like a core issue so I'm going to move to a discussion...