earlephilhower/arduino-pico #960: C++ extension configuration issues when developing in VSCode

vishesh-varma, 2022-11-05:
Hi Earle! This is not is more of a general issue that I am having when developing code for Raspberry Pi Pico in VSCode.
In the first line of my code, I get squiggle errors that mention this:
[image]

I have an auto-configured `c_cpp_properties.json` that auto configures when I compile the code, getting rid of any changes I make to the include path list.
I wanted to know if it is something that you can look into, or it's more of an "Arduino has to take care of it" thing.
Also, is there a "good" configuration file that exists?

earlephilhower, 2022-11-05:
You'll need to at least copy the full text of the output message.  It's all cut off in your image.

FWIW there is no `pico/stdio.h` dependency AFAIK in the core.  We use Newlib's `stdio.h`.

Also, have you tried Platform.IO?  Doesn't that use the same IDE?  It seems to work fine using the info given in the docs.

vishesh-varma, 2022-11-07:
Apologies, this is the exact error.
[image]
```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (D:\Projects\tactonic-pad-v2\TactonicPad2\TactonicPad2.ino).C/C++(1696)
```
---
```
cannot open source file "pico/stdio.h" (dependency of "C:\Users\guest1.RWL026\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.6.0\cores\rp2040\Arduino.h")C/C++(1696)
```
I'm not using platformIO, I'm only using the Arduino extension for development, as it was working well enough for every other board I've connected, the ones with official Arduino support (Leonardo/Mega/Nano) and otherwise (esp32/esp8266), with the default settings.
That being said, I'll switch to Platform.IO for the time being, I just think that this might be worth looking into. Thanks a lot for your work, Earle!

earlephilhower, 2022-11-07:
I'm not sure where to go with this, so I've put it as `help wanted` and hopefully there are some VSCode users out there who can give some config settings.

I double-checked and there is no `#include <stdio.h>` anywhere in the core except for one spot in the SDK-override files for the CYW43 (and in a .C not a header file)  #965 will get rid of that, but I don't think it's your problem here.

vishesh-varma, 2022-11-10:
This is only an intellisense error ,code gets compiled normally, but only inellisense features stop working, like autocomplete and error-detection.

jlarnal, 2022-11-29:
The pico-sdk include hierarchy is a hot mess for intellisense when. It has been created with the Linux ecosystem in mind, whereas Arduino ecosystem's been designed for ease-of-use, handling and obfuscating (wrongly so ?) almost every dependencies search or reference declaration for the final user.
Two diametrically opposed philosophies here. One says "know your sh..", the other one says "come along everybody, we've ironed the details for you".

It's as if a part of the programmers community is expecting each and every one of their pair to be some sort of toolchain expert. 
No wonder python (and micropython henceforth) has gained such a popularity. Just add water.

Meanwhile, we're here lacking auto-completion because we don't wanna sacrifice 2 hours of work per project by solely typing a cursed, obscure and barely human-readable CMakeList we'll have to edit again as soon as we implement a new class. 

Sorry for the rant lasses and lads... I'm utterly pissed at all those polka-dotted C/C++ SDKs. 😤

StefanKellerAC, 2022-12-10:
Yes, the Arduino Extension for VSC does not find the paths in \Pico-SDK, 
that may be a problem with \ and / when the extension parses the output, which is e.g.:

```
"-IC:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4/pico-sdk/src/rp2040/hardware_regs/include/"
```

I will post this also there:
https://github.com/microsoft/vscode-arduino/issues/1354

After disabling "Intelli Sense Auto Gen" in the Arduino extension to prevent overwriting when compiling
 you can add those paths in   "\.vscode\c_cpp_properties.json"  in your project folder.

Simply all files like this:
```
            "includePath": [

...

           "C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\include\\pico_base\\**"
```
but this may lead to some wrong includes for different files with same name, e.g. time.h

Therefore I took (most of) the paths from 
   "... rp2040\hardware\rp2040\2.6.4\lib\platform_inc.txt"
and put that with some text replacing into   
   \.vscode\c_cpp_properties.json

```
       //  do not include:       
       //  ,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\lib\\tinyusb\\src\\**"

,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\boards\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_base\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_binary_info\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_bit_ops\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_divider\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_stdlib\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_sync\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_time\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\common\\pico_util\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2040\\hardware_regs\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2040\\hardware_structs\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\cmsis\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\cmsis\\stub\\CMSIS\\Core\\Include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\cmsis\\stub\\CMSIS\\Device\\RaspberryPi\\RP2040\\Include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_adc\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_base\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_claim\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_clocks\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_divider\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_dma\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_exception\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_flash\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_gpio\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_i2c\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_interp\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_irq\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_rtc\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_pio\\include\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\src\\rp2_common\\hardware_pll\\include\\**"
[... cut ...]
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\lib\\cyw43-driver\\src\\**"
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\pico-sdk\\lib\\lwip\\src\\include\\**"

// additonally:                
,"C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4\\include\\pico_base\\**"
```

Maybe there is a variable with the platform path, 
You have to change this with every update because the version is in the path.
b.t.w. wouldn't it be better not to do that??

jlarnal, 2022-12-14:
And once again we have to give our thanks to Microsoft for choosing, a score ago,  some distinctive folder separator. "Let's use backslash instead of forward slash, because... why not ?" 😅🤪😭

earlephilhower, 2022-12-14:
@StefanKellerAC is there a non-absolute-path version of the file we could include in the core itself in an `extras` directory or something?  I can programatically swap `/` with `\\` and gen files as part of the `make-libpich.sh` script if so...

StefanKellerAC, 2022-12-29:
`/` and `\\` seems not to be the problem. with ESP 32 it works:

this line in output
`"-IC:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/newlib/platform_include"`
gives this in c_cpp_properties.json:
```
 "includePath": [
                "C:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.5\\tools\\sdk\\esp32\\include\\newlib\\platform_include",
```

bus710, 2023-03-05:
Sorry for bumping up this ticket after couple months...

I happened to prepare the environment like others (arduino-cli, arduino-pico, and vscode-arduino), so the **includePath** in the c_cpp_properties.json was little confusing part as well.

Eventually I picked these paths:
<details>
  <summary>66 lines</summary>
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/cores/rp2040",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/cores/rp2040/api",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/cores/rp2040/api/deprecated-avr-comp",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/include/pico_base",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/variants/rpipico",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/lib/tinyusb/src",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/boards/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_base/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_binary_info/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_bit_ops/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_divider/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_stdlib/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_sync/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_time/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_usb_reset_interface/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/common/pico_util/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2040/hardware_regs/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2040/hardware_structs/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/cmsis/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Core/Include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_adc/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_base/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_claim/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_clocks/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_divider/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_dma/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_exception/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_flash/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_gpio/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_i2c/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_interp/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_irq/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_rtc/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_pio/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_pll/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_pwm/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_resets/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_spi/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_sync/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_timer/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_uart/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_vreg/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_watchdog/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/hardware_xosc/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_async_context/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_bootrom/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_btstack/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_cyw43_arch/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_cyw43_driver/include",
- "~/.arduino15/packages/rp2040/hardware/rp2040/3.0.0/pico-sdk/src/rp2_common/pico_double/include",
[... cut ...]
In case of Linux (as I do this in Debian):
1. we can have an environment variable to shorten the long prefix (~/.arduino15/packages/rp2040/hardware/rp2040/).
2. the version can be checked by the ls command under the shortened path.
3. refer the platform_inc.txt file to add most of the missing headers.
4. the gathered information can be used as arguments for **jq** command to fill the includePath (I am not sure if Windows has this command...?).

Then a shell script can be written in this manner and we can be happy ever after....?

bus710, 2023-03-06:
https://github.com/bus710/Arduino-pico-development-in-linux

So I created this repository for 2 purposes:
1. To introduce how to setup the development environment with specific items.
2. A Go app to read the platform_inc.txt and appropriately translate for the includePath ([see also](https://github.com/bus710/Arduino-pico-development-in-linux/blob/main/README2.md#little-better-way)). The output works well for me. 

The app has some caveats:
- As I don't have a Windows PC, it won't work for Windows (no double back slashes!!). 
- Go SDK should exist in the system (maybe github action to generate executables??).
- Not very flexible yet as there are some fixed parameters.

Please feel free to open a ticket there.
Thanks,

oddstr13, 2023-03-06:
Forward slashes should be fine on Windows too.

bus710, 2023-03-06:
Then probably the path of .arduino15 directory in Windows might be the next key, I suspect.

bus710, 2023-03-06:
I was able to create executables for Linux/Mac/Windows (Thanks to Go's cross build capability).
Can someone download and test the Windows executable? It should be executed from cmd or Windows Terminal. Also, the platform_inc.txt should be placed under "$MAIN_DISK:/Users/$USER/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/$VERSION/lib/platform_inc.txt"

https://github.com/bus710/Arduino-pico-development-in-linux/releases/tag/v0.0.1

oddstr13, 2023-03-07:
> Also, the platform_inc.txt should be placed under "$MAIN_DISK:/Users/$USER/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/$VERSION/lib/platform_inc.txt"

There is an environment variable for Local AppData, which you might want to use ($LOCALAPPDATA or some such, I don't have a reference handy right now)

earlephilhower, 2023-09-17:
Unfortunately there's been nobody who'sa user of VSCode and Windows who has been able to help out here.  Closing for now, but if someone does have a PR it would be most welcome!

closed 2023-09-17 by earlephilhower

berman00, 2024-07-17:
I'm also dealing with this problem. I have no idea how to fix it because I'm not familiar with how the Arduino extension checks for the include directories. I have a quick fix that will reduce your pain though.

In vscode you are allowed multiple configurations in the c_cpp_properties.json file. The arduino extension only changes the one named "Arduino".

What I do is copy and paste the "Arduino" configuration inside the json list named "configurations". Then just change the name of the configuration to something like "myArduino", and add the following line to the includePath:
```
"configurations": [
        {
            "name": "myArduino",

           {...}

            "includePath": [
                
                "C:\\Other\\Include\\Paths",

                "C:\\Users\\berman00\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.9.3\\**",  // << Added path
            ],

           {...}
```

Of course, change the path to suit your particular use case.

Then, change the configuration in the bottom right corner of the IDE:

[image]

This is better than disabling the automatic code generator feature of the extension IMO, because if you change your program by adding a library, the path to the library will automatically appear in the Arduino config, and you can copy and paste it to your own config. A little annoying, but it is the best solution I found so far.

Hope this helps!