earlephilhower/arduino-pico #961: Unable to build a project in latest release V2.6.3 but fine in 2.6.2 or lower.

revell1, 2022-11-05:
I tried updating from version 2.6.1 to 2.6.3, but find I have several issues.

First issue (I am not bothered by this as I now manually unistall the current version then install the new/latest version):
If in the BOARDS MANAGER page, I try to just UPDATE to V2.6.3 (from 2.6.1 at the time), the update fails with some error, can't remember the exact wording, but I think it was failing during the update AFTER automatically removing the OLD version.
At present I find that I have to manually unistall the current RP2040 library version, quit the IDE, then restart the IDE, and INSTALL V2.6.3, then quit IDE, restart IDE, then select my board type PI PICO-W (No FS)

Second issue:
When I try building my project (which builds fine in 2.6.1 and 2.6.2), I get the following failure:

Compiling sketch...
"C:\\Users\\....\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3" -I "C:\\Users\\....\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.3/tools/signing.py" --mode header --publickey "C:\\Users\\.....\\Documents\\Coding\\PiPico_CodeTests\\PiPicoW\\_WifiBuggyCode\\WebBuggy/public.key" --out "C:\\Users\\garet\\AppData\\Local\\Temp\\arduino_build_398783/core/Updater_Signing.h"
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

It appears to be trying to use a file "public.key" from my project directory (where my INO project file and other C/H code files), but I have no such file, I guess there is some issue with your latest release of 2.6.3 that has not done something to do with signing.

This error does not occur if I use V2.6.2 or lower of your library release.

At present, I will revert back to using 2.6.2, as that appears to be building OK.

revell1, 2022-11-05:
Further info.

Having reverted to V2.6.2, I found that although the code was building, the PWM control of my two motors was not working (motors would buzz but not rotate).

So I then tried to revert to V2.6.1, but was getting the same build error as V2.6.3.
I had to uninstall V2.6.1 and go through a number of install/uninstall cycles before I finally got my code building on V2.6.1 again, and the PWM motor control was working again.

So I am not sure what the build issue is between V2.6.1, V2.6.2 and V2.6.3, but it would appear that there is some issue with the Arduino IDE install/update process that is causing this (what now appears to be possibly a random) build issue.

I also noted that when I build my projects, the IDE is building to a TEMP directory, which only lasts for as long as the IDE is open. If I close the IDE, then the TEMP directory gets purged, though I am not sure it occurs straight away, so it may be that my problem is one of timing, in that I am not leaving long enough between exiting the IDE and doing a rebuild.

I am going to try some more installs of V2.6.3 but with longer delays between IDE launches, and see if I can get my project to build for V2.6.3, though I suspect my motors will not run.

revell1, 2022-11-05:
OK. Well after a number of uninstall/reinstall cycles, with various pauses I finally got V2.6.3 to install, and for my project to compile and download to the target.

So there does appear to be some IDE issue with installing your library, which has some time sensitive issue.

So I guess there is probably no point looking into this, unless it is something that you have a suspicion about.

As for the PWM issue, that still exists in V2.6.3, so I assume it must be relate to the issue :
"Only init PWM on requested pin (#926)"
Which was addressed in V2.6.2.

earlephilhower, 2022-11-05:
Yeah, #926 looks to have busted PWM and if I can't figure out how I'm going to undo that change in the next release.

As for the python error, nobody else seems to be hitting that (or at least it's not really reported).

Even if you have your own Python3 installed somewhere in the path, the `boards.txt` explicitly uses the one installed bu the IDE/get.py and uses the `-I` isolated mode which completely ignores any environment variables, paths, Python libs, etc. that are not in the explicit python installation used.  That bit has been tested under Windows before and verified.

earlephilhower, 2022-11-05:
Can you please try #962?  The PWM issue is being tracked under #955 so closing here to keep the issue in 1 spot.

closed 2022-11-05 by earlephilhower

IAmOrion, 2022-11-13:
I too am getting the Python error

`Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
`